SpinalHDL/NaxRiscv #3: Use IDEA build NaxRiscv occurs error

djcmaster123, 2022-03-09:
Hi, when I use idea build NaxRiscv which is a wonderful project for learning cpu design, it's build.sbt need spinalHDL 1.6.5, however this repo(SpinalHDLTemplate) version isn't it and I found  content of SpinalHDL has updated. Can you help me solve it by update spinalHDLTemplate's build.sbt version. Thanks!

Dolu1990, 2022-03-09:
I think one good template is SaxonSoc build.sbt : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/build.sbt

It depend on VexRiscv dev branch which is a similar case to NaxRiscv
So basicaly, you need to have the ext folder with both SpinalHDL and NaxRiscv cloned in it, update that build.sbt to use NaxRiscv instead of VexRiscv and it should be good.

Also, in a first time, try with SBT, not intellij, as intellij, is a bit clunky sometime with "complex" sbt setups

djcmaster123, 2022-03-09:
Thanks! I'll try. I think it need  correct jar file because I found spinalHDL's lib has updated sucah as utils.scala. Another method is use spinalHDL source code directly rather than build.sbt.

djcmaster123, 2022-03-09:
> Thanks! I'll try. I think it need correct jar file because I found spinalHDL's lib has updated sucah as utils.scala. Another method is use spinalHDL source code directly rather than build.sbt.

Just as you said, put spinalHDL and NaxRiscv in one ext folder, change build.sbt. Maybe the right way.

Dolu1990, 2022-03-09:
Cool :)

> NaxRiscv which is a wonderful project for learning cpu design

If there is anything unclear / need doc, do not hesitate to open some issue / pull request ^^ 
Including the doc repo : 
https://github.com/SpinalHDL/NaxRiscv-Rtd

djcmaster123, 2022-03-10:
Cool :)). After studying, compile it & reboot Linux successfully on high performance server. Some environment like "VexRiscv" also need. I'll learn NaxRiscv/VexRiscv which are fun & challenge ,meaningful. Complete documention about this project will help learner study , dive into it and make contributions. Thanks again! (forgive my poor english)