SpinalHDL/NaxRiscv #4: Do you have any maximum frequency numbers for Series-7 FPGAs?

teknoman117, 2022-04-07:
I came across this project today. I was wondering if you have any maximum frequency numbers for the Series-7 FPGAs? You state that one of your goals is to achieve higher single threaded throughput compared to your VexRiscv project, so I was wondering how much the increased complexity of the core drags down the maximum frequency.

Dolu1990, 2022-04-07:
Hi,
Grosso modo, for Linux config with similar cache settings, Nax frequency is at 145 Mhz, while Vex is at 185 Mhz, (nax having double the IPC)

Here is some perf / synthesis numbers : 
https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/performance/index.html#rv32

Note that the 64 bits support has also very little impact on FMax so far

teknoman117, 2022-04-07:
Ah! I missed that section. Closing.

closed 2022-04-07 by teknoman117

Dolu1990, 2022-04-07:
no worries ^^