SpinalHDL/NaxRiscv #6: [ help ] ExecuteUnitDemo and custom operator

dreamflyings, 2022-04-12:
Dear Dolu1990
The Nax project is really great!
Its software architecture is very helpful for learning and research, and it also has great application prospects in engineering implementation. I sincerely hope that this architecture can become popular among developers like LazyModule in chisel.

So far, I have been learning Nax for almost half a year, but I have only touched a little bit, and it is still a little far from engineering applications.
I hope to be familiar with Nax's development process as soon as possible and use it in product development.

The requirements of my project are as follows. Input instructions through the Fifo interface or FetchCachePlugin_mem interface (no branch prediction is required), and then through instruction decoding, instruction dispatch, instruction execution, instruction reordering, instruction commit, instruction writeback and other operations, the instruction result will be Storage FIFO or RAM.

I would like to be able to reuse most of the Nax code and be able to implement a user-defined instruction set.

I have been fumbling with ExecuteUnitBase and ExecuteUnitDemo for a few days, and probably understand the relevant code, but it feels very difficult to modify ExecuteUnit according to the needs, and, so far, I can't integrate ExecuteUnitDemo into Gen (it seems that I need to write an extended subclass) .

[https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/execute/ExecuteUnitDemo.scala](url)
[https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/Gen.scala](url)

@Dolu1990 
Can you have a simple demo code of the whole process (including: decoding, encoding, integration and bare metal testing of custom operators), which can enable us beginners to apply engineering as soon as possible.

Best regards！

Dolu1990, 2022-04-12:
> So far, I have been learning Nax for almost half a year

Hoo, so, NaxRiscv only started in october 2021, things are still very fresh XD

>  Input instructions through the Fifo interface or FetchCachePlugin_mem 

One thing, i recently added JTAG support, which added bypass support in the FetchCachePlugin plugin to provide debug instruction stream.

Also about instruction stream, be aware that NaxRiscv LSU / branches / .. may make restart instruction which were already fetched

> The requirements of my project are as follows. Input instructions through the Fifo interface or FetchCachePlugin_mem interface (no branch prediction is required), and then through instruction [...]

Ahhh so extending RISC-V or replacing it with something else ?

> ExecuteUnitDemo

ExecuteUnitDemo was the very early execution unit made for putting things in place and debugging the framework, i haven't used it since a long long while, and it isn't realy made to be composable / reusable, nor it is updated to last changes.

ExecuteUnitBase is the reusable and extendable execution unit design. Alternative to it could be made

> Can you have a simple demo code of the whole process (including: decoding, encoding, integration and bare metal testing of custom operators), which can enable us beginners to apply engineering as soon as possible.

So far, i think the easiest aproache to add custom instruction is to look at : 
- IntAlu : https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/execute/IntAluPlugin.scala
- BarrelShifter : https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/execute/ShiftPlugin.scala
- Div : https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/execute/DivPlugin.scala
- Mul : https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/execute/MulPlugin.scala

They all rely being on being coupled to a ExecuteUnitBase, but some are mostly designed to be in fixed latency pipeline (int alu / barrel shifter) and some are design to be only in variable latency pipeline (div)

This plugins are self contained, nothing is hardcoded outside of them

But overall, it would be good i think that i add some doc about the full process of implementing, debugging a custom instruction. I will do that ^^

> Also, do we have plans to implement riscv vector instruction extensions? Or just implement a framework, what do you think?

Currently, the plan is to finish the JTAG debug support, and then implement FPU, which would enable debian 64 bits to run <3
Then, maybe multicore / memory coherency.

There is no plan for vector instructions so far, but maybe in future ?

Best Regards
Charles

Dolu1990, 2022-04-23:
added https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/execution_units/index.html#hardcore-way