SpinalHDL/NaxRiscv #12: Different behavior compared to spike when accessing an invalid address

mmxsrup, 2022-09-23:
Hi, The NaxRiscv project is really great!
I am learning a lot about testing hardware designs from this project.

An error like `*** MISSMATCH PC DUT=8000000c REF=0 ***` occurred when a program containing an illegal memory access was executed by verilator. Is this an expected behavior? If the post-processing due to illegal memory accesses is matched with spike, I would expect the RTL and spike behavior to match. What could be the cause?

Reproduction steps:
1. Prepare the following assembly file in NaxSoftware's bare metal environment. The file path is `/NaxRiscv/ext/NaxSoftware/baremetal/test/crt.S`.
  ```asm
  .globl _start
  _start:
      addi x1, x1, 64
      addi x2, x2, 64
      sw x3, 0(x0)
      ret
  ```
2. Compile with command `make -C $NAXRISCV/ext/NaxSoftware/baremetal/test RISCV_PATH=$RISCV MARCH=rv32im MABI=ilp32`.
3. Execute with command `cd $NAXRISCV/src/test/cpp/naxriscv; ./obj_dir/VNaxRiscv --load-elf ../../../../ext/NaxSoftware/baremetal/test/build/test.elf`.
4. The following output is generated.
```txt
*** MISSMATCH PC DUT=8000000c REF=0 ***

TIME=262
LAST PC COMMIT=8000000c
INCOMING SPIKE PC=0
ROB_ID=x5
FAILURE ???
```

Dolu1990, 2022-09-23:
Hi,

> I am learning a lot about testing hardware designs from this project.

Great, thanks :)

So, i just run the sim, and remember that it in that binary, both Spike and Nax are doing what they are expected to do.
The main thing is that NaxRiscv, for stores, will only execute the store after commit, and so, if a access fault appear there, it is too late to revert the state of the CPU, and so, the access fault is just ignored.
This is done like this mostly for performance reasons, not having to wait for the data cache to wait the store cache line before commit (it speed up things).

I have no idea how other CPU in general handle this kind of case. As it is mostly a baremetal / no mmu case.

Notice that store access fault are checked for IO accesses.

I think, the spike fork could be patched to never catch regular memory store fault in order to match nax

So, if you want, another way to produce an exception on store is to have a missaligned address : 

```asm
_start:
    la x1, pass
    csrw mtvec, x1
    addi x1, x1, 64
    addi x2, x2, 64
    sw x3, 0(x0)
    j fail
pass:
  j pass
fail:
  j fail
```

./obj_dir/VNaxRiscv --load-elf ../../../../ext/NaxSoftware/baremetal/play/build/rv32ima/play.elf  --trace-ref --spike-debug --pass-symbol pass --fail-symbol fail

```
core   0: 0x80000000 (0x00000097) auipc   ra, 0x0
core   0: 3 0x80000000 (0x00000097) x 1 0x80000000
core   0: 0x80000004 (0x01c08093) addi    ra, ra, 28
core   0: 3 0x80000004 (0x01c08093) x 1 0x8000001c
core   0: 0x80000008 (0x30509073) csrw    mtvec, ra
core   0: 3 0x80000008 (0x30509073) c773_mtvec 0x8000001c
core   0: 0x8000000c (0x04008093) addi    ra, ra, 64
core   0: 3 0x8000000c (0x04008093) x 1 0x8000005c
core   0: 0x80000010 (0x04010113) addi    sp, sp, 64
core   0: 3 0x80000010 (0x04010113) x 2 0x00000040
core   0: 0x80000014 (0x003020a3) sw      gp, 1(zero)
core   0: exception trap_store_address_misaligned, epc 0x80000014
core   0:           tval 0x00000001
core   0: 0x8000001c (0x00000013) nop   <- that nop is a trick added by the simulation environnement, but that's the pass symbole
core   0: 3 0x8000001c (0x00000013)
```

Also, seems you are looking into security stuff ?
Have you seen https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/misc/index.html#side-channel-attack ? 
It is recent :)

mmxsrup, 2022-09-24:
Thanks for your prompt reply!

It seems that by setting the `pass` symbol to a trap vector, the behavior matches that of the spike. This approach allows for proper differential testing even with input containing illegal instructions.

> Also, seems you are looking into security stuff ?

Partially yes. More specifically, I would like to test this core by generating random inputs. I am also interested in side channels, so this will help me a lot. Thank you.

Dolu1990, 2022-09-24:
> , I would like to test this core by generating random inputs

Ahhh nice :)
Let's me know how it goes :D