SpinalHDL/NaxRiscv #13: An assertion is triggered by self-modifying code

mmxsrup, 2022-09-25:
I found that a self-modifying code trigger the assertion `*** INTEGER WRITE DATA DUT=40 REF=0 ***`. Nax does not seem to support rewriting code sections, but spike seems to support rewriting code sections. Is this expected behavior?

The input that triggers this assertion was found by a random generator for hardware designs.

Reproduction steps:
1. Prepare the following assembly file in NaxSoftware's bare metal environment. The file path is `/NaxRiscv/ext/NaxSoftware/baremetal/test/crt.S`.
  ```asm
.globl _start
_start:
    la x6, target
    sb x0, 3(x6) // rewrite addi t2, t2, 64
    addi t1, t1, 64
target:
    addi t2, t2, 64
    nop
    nop
    nop
    nop
    ret
  ```
2. Compile with command `make -C $NAXRISCV/ext/NaxSoftware/baremetal/test RISCV_PATH=$RISCV MARCH=rv32im MABI=ilp32`.
3. Execute with command `cd $NAXRISCV/src/test/cpp/naxriscv; ./obj_dir/VNaxRiscv --load-elf ../../../../ext/NaxSoftware/baremetal/test/build/test.elf --trace-ref --spike-debug`.
4. The following output is generated.
```
*** INTEGER WRITE DATA DUT=40 REF=0 ***

TIME=264
LAST PC COMMIT=80000010
INCOMING SPIKE PC=ffffffff80000014
ROB_ID=x6
FAILURE ???
```

```
core   0: 0x80000000 (0x00000317) auipc   t1, 0x0
core   0: 3 0x80000000 (0x00000317) x 6 0x80000000
core   0: 0x80000004 (0x01030313) addi    t1, t1, 16
core   0: 3 0x80000004 (0x01030313) x 6 0x80000010
core   0: 0x80000008 (0x000301a3) sb      zero, 3(t1)
core   0: 3 0x80000008 (0x000301a3) mem 0x80000013 0x0
core   0: 0x8000000c (0x04030313) addi    t1, t1, 64
core   0: 3 0x8000000c (0x04030313) x 6 0x80000050
core   0: 0x80000010 (0x00038393) mv      t2, t2 <- addi t2, t2, 64 is dynamically rewritten
core   0: 3 0x80000010 (0x00038393) x 7 0x00000000
```

Dolu1990, 2022-09-25:
Hi,

>  Is this expected behavior?

Yes it is.

So, there is two things : 
- RISC-V spec do not require memory coherency between data cache and instruction cache, and instead, you have to do a fence.i instruction to encorce all previous store to be visible by the instruction cache.
- I don't know if spike itself implement memory coherency between data and instruction, but likely, it only fetch instruction at the moment they are executed, while NaxRiscv fetch instruction 64 byte at the time, creating the memory coherency issue.

```asm
.globl _start
_start:
  la x6, target
  sb x0, 3(x6) // rewrite addi t2, t2, 64
  fence.i //added this
  addi t1, t1, 64
target:
  addi t2, t2, 64
  nop
  nop
  nop
  nop
  ret
  ```
  
  Should work i think.

mmxsrup, 2022-09-26:
Thanks for the quick reply!

> RISC-V spec do not require memory coherency between data cache and instruction cache, and instead, you have to do a fence.i instruction to encorce all previous store to be visible by the instruction cache.

I did not know this specification. I need to put a fence.i instruction in self-modifying code. I could actually verify that the two behaviors match by putting a fence.i instruction.

Dolu1990, 2022-09-26:
Should be somewere lost in the :
The RISC-V Instruction Set Manual Volume I: Unprivileged ISA

The spec is kind of so large, hard to find stuff we don't know about in it :D
Even more chaotic with the privileged ISA spec XD

mmxsrup, 2022-09-26:
Yes, I agree.
RISC-V is simpler than other ISAs, but I still feel complex...