SpinalHDL/NaxRiscv #14: Cooperate with rust code

ducdracaena, 2022-10-15:
I wrote the firmware in rust code with read function, I tried to run this program in Naxriscv but stuck in read function. 

Another side, I tried that function in vexriscv ,this worked fine! 

This is the firmware:
unsafe fn main() -> ! {
    // led debug
    let LED_ADDR: u32 = 0x10000000;//0x10000000 with vexrisc and 0xFF000000 with naxrisc
    let mut led: u32 = 0x00000000;
    //let mut tcp_status;

    // TCP interface
    //let TCP_STATUS_ADDR: u32 = 0xFF000000;
    //let TCP_RX_ADDR: u32 = 0xff000020;
    //let TCP_TX_ADDR: u32 = 0xff000030;

    // init led
    (LED_ADDR as *mut u32).write_volatile(led);
    loop {
       led = (LED_ADDR as *mut u32).read();
       
       led = led ^ 0xFFFF000;
        (LED_ADDR as *mut u32).write_volatile(led);
        
        delay(100);
    }
}

Can you please check that?

Thank you,
Duc

Dolu1990, 2022-10-17:
Hi,

So far, what i tested with rust was compiling and running it in NaxRiscv debian.
Seems it was going well.

Could you provide a way to reproduce your issue ?
binaries i could run using that simulation environnement : 
https://github.com/SpinalHDL/NaxRiscv/tree/main/src/test/cpp/naxriscv

or a wave of your simulation.

ducdracaena, 2022-10-17:
Hello,

Can you please try this 2 files: 
https://file.io/tK8EdXJSbwoD

Thank you.

ducdracaena, 2022-10-20:
Hi, 

Do you need anymore information or elf or vcd file?

Thank you,
Duc

Dolu1990, 2022-10-20:
Hi ^^

So, the https://file.io/tK8EdXJSbwoD files are deleted.
I need the wave and the elf :)

Thanks
Charles

ducdracaena, 2022-10-21:
Hi,

Please check this :) 
https://file.io/e7vow2RhHvI5

Thank you,
Duc

Dolu1990, 2022-10-21:
Hi,
Seems like your issue is comming from the peripheral / interconnect side of things.
See : 
[image]

Getting an AXIlite r response with zero latency isn't something AXIlite compatible. (and it confuse NaxRiscv).

Seems like the issue is between : 
the outputs of cpu_LsuPeripheralAxiLite4_logic_axi_readOnlyDecoder
-> 
peripheral_axil

closed 2022-10-27 by Dolu1990