SpinalHDL/NaxRiscv #16: Find the plugin for PeripheralAXI4

ducdracaena, 2022-11-06:
Hi,

I the the plugin for PeripheralAXILite4

plugins += new LsuPeripheralAxiLite4(ioDataWidth = xlen)

Is there any plugins for AXI4? Also, where can I find a list of plugins for naxrisc? 

Thank you so much,
Duc

Dolu1990, 2022-11-06:
Hi,

There is currently no maintained list of available plugin. If you do "grep -r 'extends Plugin' ."  you should get them all.

For the Axi4 stuff, here is an example to have full AXI interface :

https://github.com/SpinalHDL/NaxRiscv/blob/0fd620074dd486a7bd9accbf783edc96c9ad65fd/src/main/scala/naxriscv/platform/Litex.scala#L27

So you need those 3 plugins and that should be all good.

Regards
Charles

ducdracaena, 2022-11-07:
A question, with those AXI4,

  plugins += new FetchAxi4(
    ramDataWidth = ramDataWidth,
    ioDataWidth  = ioDataWidth,
    toPeripheral = cmd => toPeripheral(cmd.address)
  )
  plugins += new DataCacheAxi4(
    dataWidth = ramDataWidth
  )

 I can access the external RAM through AXI4? Do you have any example?

Right now, I can access the external ram through LsuPeripheralAxiLite4 but that is AXIlite, so it will be slow.

Thank you so much,
Duc

Dolu1990, 2022-11-07:
> Right now, I can access the external ram through LsuPeripheralAxiLite4 but that is AXIlite, so it will be slow.

This is specified by the MMU ioRange parameter 
The io region wil gon on the AxiLite, while the non io region will go on the AXI4.

Regards
Charles

Dolu1990, 2022-11-07:
Sure, for instance : 
https://github.com/SpinalHDL/NaxRiscv/blob/0fd620074dd486a7bd9accbf783edc96c9ad65fd/src/main/scala/naxriscv/Gen.scala#L407

Could be ajusted, currently it mean that all the access in [0x10000000-0x1FFFFFFF] are io region.

ducdracaena, 2022-11-07:
Cool, 

I will check it. thank you so much. 

Duc

Dolu1990, 2022-11-10:
hmm that seems ok to me. Can you send me a wave file ?

Dolu1990, 2022-11-10:
(and the .asm too)

ducdracaena, 2022-11-10:
Hi,

I would like to upload the code 

https://www.file.io/uZXH/download/cxLkN0NVE83g

Thank you so much,
Duc

ducdracaena, 2022-11-10:
Hi,

I forget the vcd file. 
https://file.io/IZX4wujdsPkc

Thanks,
Duc

ducdracaena, 2022-11-11:
Hi,

I am still confused. So, how can I make it write AXI to the address 0x80000000

Thank you,
Duc

Dolu1990, 2022-11-11:
Hi,

You need to explicitly flush the data cache
there is a custom instruction for this, encoded as 0x0000500F

ducdracaena, 2022-11-11:
Hi, 

The problem is I dont know which command in rust to complie to make flush the data cache.

Dolu1990, 2022-11-11:
hmm, can you put some inline assembly code in rust ?
I don't know rust much.

ducdracaena, 2022-11-11:
Hi, 

This one is example of inline assembly

ducdracaena, 2022-11-11:
Hi,

This one is example of inline assembly and can complie good. 

fn delay(cycles: u32) {
    for _ in 0..cycles {
        unsafe {
            asm!("nop");
        }
    }
}

This is nop code. Do you know the opcode name of 0x0000500F

Thank you so much,
Duc

cklarhorst, 2022-11-11:
I have no rust compiler but [here](https://docs.rs/xuantie/latest/src/xuantie/asm.rs.html#29-31) is rust code for another riscv cpu.
From that I think you need something like:
`asm!(".insn i 0x500F");`

ducdracaena, 2022-11-12:
Hello, 

This worked! Thank you so much. 

Duc

Dolu1990, 2022-11-13:
@cklarhorst Thanks :D

closed 2022-11-14 by Dolu1990