SpinalHDL/NaxRiscv #18: Trap executing CSRW

kleinai, 2022-12-18:
I am having some trouble brining up a Nax core and the Rust riscv-rt library.

My code builds and loads into the softcore fine but traps quickly when accessing CSRs.

Source assembly is here: https://github.com/rust-embedded/riscv-rt/blob/master/asm.S
Here's the first few lines of my binary disassembly:
```
minisoc:        file format elf32-littleriscv
                                             
Disassembly of section .text:                
                                             
80000000 <_start>:                           
80000000: b7 00 00 80   lui     ra, 524288   
                                             
80000004 <.L0 >:                             
80000004: 67 80 80 00   jr      8(ra)        
                                             
80000008 <_abs_start>:                       
80000008: 73 50 40 30   csrwi   mie, 0       
                                             
8000000c <.L0 >:                             
8000000c: 73 50 40 34   csrwi   mip, 0       
                                             
80000010 <.L0 >:                             
80000010: 81 40         li      ra, 0        
                                             
80000012 <.L0 >:                             
80000012: 01 41         li      sp, 0        
                                             
80000014 <.L0 >:                             
80000014: 81 41         li      gp, 0
```

The core traps after `0x8000000C` which jumps it back to `0x00000000`.

Configuration is `Config.plugins(withDebug = true, withEmbeddedJtagTap = true)` with some peripherals.

I did try to hack in pieces of the interrupt controllers so maybe it's grabbing an erroneous external interrupt?

Any suggestions are appreciated!

Dolu1990, 2022-12-18:
Hi,

Yes that's "normal", as mip is a read only register in NaxRiscv : 
https://github.com/SpinalHDL/NaxRiscv/blob/4a6c4f67e9e1d0f2ca4bd36e87bc42730b89977f/src/main/scala/naxriscv/misc/PrivilegedPlugin.scala#L428

We could easily turn it into allowing write.

csr.allowCsr(CSR.MIP)

kleinai, 2022-12-18:
Is this something that should be "fixed"? That rust library seems to think the machine and supervisor CSRs should be writable, or at the least not an error.

Dolu1990, 2022-12-18:
> Is this something that should be "fixed"? 

Yes ^^

Dolu1990, 2022-12-18:
hardware fix required

kleinai, 2022-12-19:
I patched in `csr.allowCsr(CSR.MIP)` but that hasn't fixed the bug :/
I also tried adding a single bit to MIP as read/write but that didn't fix the trap either.

Dolu1990, 2022-12-19:
I just tested in sim, and the core i have do not trap on mip writes (even without allowCsr)

Do you have a simulation wave ? Could i look at it ?

Dolu1990, 2022-12-19:
AHHHHHH
You have compressed instruction set enabled in your assembly ( 81 40         li      ra, 0       )

While the core do not implement it by default. you have to turn on withRvc = true ^^

kleinai, 2022-12-19:
That was it! Thanks so much!

closed 2022-12-19 by kleinai