SpinalHDL/NaxRiscv #19: Cannot synthesize in Vivado

ducdracaena, 2023-01-13:
Hi, 

I used the generated code to synthesize in Vivado, but it doesnt work. This is the error. 

Unexpected Assertion error in File r:/wall1/workspaces/wall724/sub/REL/2022.1/src/shared/synth/rtx/syn/gen/HARTGRamGenInfo.cxx Line 4583 
An unrecoverable error has occurred, synthesis cancelled.
TclStackFree: incorrect freePtr. Call out of sequence?

Could you please let me know how to solve it?
Thank you.
Duc

Dolu1990, 2023-01-13:
Hi, never seen such error, i guess that's a  pure vivado issue ? maybe trying to change vivado version would fix it ?

Regards
Charles

ducdracaena, 2023-01-13:
Hmm I tried some Vivado version but doesnt work.

Dolu1990, 2023-01-16:
Some propose to start vivado with more stack : 
```shell
vivado -stack 2000 
```

wyp7788, 2024-12-04:
> Some propose to start vivado with more stack :
> 
> ```shell
> vivado -stack 2000 
> ```

wrong command

Dolu1990, 2024-12-05:
No idea then.