SpinalHDL/NaxRiscv #20: Linux dts/dtb instructions

AEW2015, 2023-02-02:
There are hints for getting the  linux-on-litex image working by modifying the dts/dtb. 
I am unsure what needs to be done.
Would it be possible to get some more instructions for this?

Dolu1990, 2023-02-03:
So the  thing which need to be adjusted are potencialy the address of peripherals in the DTS and potentialy the opensbi image address need adjustement too.
The best may just be to force / ensure that peripheral address are mapped at the same place (uart / plic / clint)

roryt12, 2023-02-03:
Maybe my steps here : https://github.com/roryt12/qmtech_wukong_debian_on_litex_naxriscv can assist you.

Dolu1990, 2023-02-03:
Hooo nice @roryt12 Thanks :)

Dolu1990, 2023-02-03:
once SMP / multicore is there (working on it), i will have to go through the whole again, i will try my best to document every steps.
First time i did it, it was a bit too much of a war field between bugs and my lack of knowledge XD

roryt12, 2023-02-03:
> once SMP / multicore is there (working on it)

Yes! :)