SpinalHDL/NaxRiscv #23: 64-bit version NaxRiscv linux support

YiminGao0113, 2023-03-09:
Hi,

I wanted to ask if the 64-bit NaxRiscv can run Linux. I followed the steps to run Linux with naxriscv rv32ima and it works well, but ran into issues (failures) when I switched to a 64-bit core. 
Could you please give me some pointers to run Linux using 64-bit core? Thanks!

[image]

Dolu1990, 2023-03-09:
Hi,

Should have worked. Can you check : 

```shell
NaxRiscv $ ls ext/NaxSoftware/buildroot/images/rv64ima
fw_dynamic.bin  fw_dynamic.elf  fw_jump.bin  fw_jump.elf  Image  linux.dtb  rootfs.cpio  vmlinux
```

?

Thanks ^^

YiminGao0113, 2023-03-09:
Hi,
It seems the same on my end: 
[image]
Thanks.

Dolu1990, 2023-03-09:
seems like you had the $NAXRISCV path not set.
Should be set to the absolute path to the NaxRiscv  :)

YiminGao0113, 2023-03-09:
Thanks for your reply! 
But I had set up the $NAXRISCV in terminal when I installed and set up everything, this is a new tab in the terminal
I just added it to my .bashrc file, but still facing the same issue. :0
[image]

Dolu1990, 2023-03-09:
Ahh , you need to regenerate the CPU in 64 bits mode, and regenerate the simulator

YiminGao0113, 2023-03-09:
Got it! 
However, (this could be a silly question) but I haven't found any instructions for configuring the core in 64bits. Which file do I need to change to generate 64-bit? 
Thanks!

Dolu1990, 2023-03-09:
sbt “runMain naxriscv.Gen64” :)

YiminGao0113, 2023-03-09:
Got it! It is working now, thank you very much for your help!

closed 2023-03-09 by YiminGao0113