SpinalHDL/NaxRiscv #24: How to change to write through DCache and invalidate from NOC

franktaTian, 2023-03-12:
Hi,
 Im trying to build a smp soc using NaxRiscv. I can find similar work as Litex done with VexRiscv. But I noticed that NaxRiscv's DCache is the one of write-back . 
How to get the DCache read/write info from NaxRiscv ? and  How to Invalidate /Modify DCache line externally(From SoC). It's too complex.
I think the simple start point is to change the write-back DCache with write-through one, and find a way so that invalidate the "dirty" line.
Thanks.

Dolu1990, 2023-03-12:
Hi,

I'm currently working on adding memory coherency to NaxRiscv, WIP :) 

If i remember well, i think you can flush + invalidate the whole data cache using the custom instruction  0x0000500F (a bit like vexriscv)

Issue of write through  is to kinda to ensure that inflight write ordering against yunger read, kinda have to keep track of them all, also write-through of VexRiscv was "often" creating issues for memory system not designed to handle such many little transaction.

Currently, the plan is to have memory coherency via write-back + tilelink

franktaTian, 2023-03-12:
Cool