SpinalHDL/NaxRiscv #26: How to start GUI

franktaTian, 2023-03-16:
Hi,
   I just  bought a digilent nexsys video FPGA card and run step -by step to boot Linux according which NaxRiscv document desicribed.
I can boot !!! and login from uart terminal.
But for HDMI output , It just shows a login prompt:

"Debian GNU/Linux bookworm/sid sid-rv64 tty1
sid-rv64 login: root"

Interesting! 
Since the card does not have keyboard , I cannot type on the 'HDMI screen'. 

Is there any more introduction to login from 'HDMI screen' and start the GUI ?
Thanks

Dolu1990, 2023-03-21:
Hi ^^

> I can boot !!! and login from uart terminal.

Nice !!

> Since the card does not have keyboard , I cannot type on the 'HDMI screen'.

So, until the memory coherency is implemented, we can't use the OHCI USB hardware support. That's WIP.
What i'm doing to try out GUI stuff, is to use a few x11 terminal commands. If i got the right copy past : 

```sh
cp /var/run/lightdm/root/:0 /tmp/lightdmauth
chmod a+r /tmp/lightdmauth
export XAUTHORITY=/tmp/lightdmauth
export DISPLAY=unix:0

xdotool type root
xdotool key Tab
xdotool type root
xdotool key Return
```

So wait login screen before doing it, but don't wait too much, else it will go into eternal sleep mode XD

Then to move the mouse, same, xdotool XD
not great, but enough to test ^^

Dolu1990, 2023-03-21:
> But for HDMI output , It just shows a login prompt:

Ahhh i'm sorry, i may have missread it.

So currently on HDMI, you are in terminal mode right ? i were thinking you had some desktop environnement, but you have none installed yet ?

franktaTian, 2023-03-22:
> > But for HDMI output , It just shows a login prompt:
> 
> Ahhh i'm sorry, i may have missread it.
> 
> So currently on HDMI, you are in terminal mode right ? i were thinking you had some desktop environnement, but you have none installed yet ?

I just installed software from NaxRiscv Document step-by-step. Does it include desktop environment yet? Or where can I download /install desktop environment  directly without having to rebuild using RV comipler? What I want is just to have some kind of GUI /Game to display through HDMI.

Dolu1990, 2023-03-23:
Ahh, have you done  : 

"Adding packages, like xfce-desktop, chocolate-doom, openttd, visualboyadvance you can get things as following :" ?

> without having to rebuild using RV compiler

Just need some apt-install, main thing is it take quite long XD
Else, there is a few app out there which may work using /dev/fb0, but there isn't many.

franktaTian, 2023-03-24:
> 
No, I have not start the rebuild work yet. I think it is hard for an IC engineer. So, Where can download "something" directly?

Dolu1990, 2023-03-24:
> I think it is hard for an IC engineer

Yes i have the same issue XD 
debian stuff start to go behond my confort zone by a lot.

I'm not  aware of a better process than running apt-get XXX on the running debian

roryt12, 2023-03-24:
May I assist on this @franktaTian  ? 

As far as the initial question, litex produces a terminal output (which is exactly the same as with the serial port output). If you really need to have keyboard (and not use the serial port), then a new device is needed, ie a USB host where you will plug a USB keyboard, or this https://github.com/Dolu1990/pmod_usb_host_x4 from Charles or maybe https://github.com/rdolbeau/PS2forLitex from Romain with a PS/2 keyboard . None of these come with the standard litex, you have to figure out how to add them manually (hardware, litex and linux kernel). Otherwise stick with the serial login, it does what it is needed.

As for the software (even though this has nothing to do with NaxRiscv, this is a Debian usage question), you have two options: either you have a working ethernet on the board and you are connected to Internet and you download what you need, or you have a RISCV chrooted filesystem on another PC, with binfmt support and qemu (see https://wiki.debian.org/RISC-V#Creating_a_riscv64_chroot) where you download what you need with apt from Debian's RISCV repository and then you copy all the filesystem to your sdcard. 

But do not expect too much from an FPGA that runs with 50 or 100MHz, limited RAM and one core. I have tested with a simple icewm environment, which I triggered by running vncserver on the FPGA, which then I accessed via network with vncviewer  (in the past that I could use liteeth, not it is not working for me).

Check also this : https://github.com/roryt12/qmtech_wukong_debian_on_litex_naxriscv , it needs some refresh after the last commits, but maybe will give you some hints.

franktaTian, 2023-03-27:
> May I assist on this @franktaTian ?
> 
> As far as the initial question, litex produces a terminal output (which is exactly the same as with the serial port output). If you really need to have keyboard (and not use the serial port), then a new device is needed, ie a USB host where you will plug a USB keyboard, or this https://github.com/Dolu1990/pmod_usb_host_x4 from Charles or maybe https://github.com/rdolbeau/PS2forLitex from Romain with a PS/2 keyboard . None of these come with the standard litex, you have to figure out how to add them manually (hardware, litex and linux kernel). Otherwise stick with the serial login, it does what it is needed.
> 
> As for the software (even though this has nothing to do with NaxRiscv, this is a Debian usage question), you have two options: either you have a working ethernet on the board and you are connected to Internet and you download what you need, or you have a RISCV chrooted filesystem on another PC, with binfmt support and qemu (see https://wiki.debian.org/RISC-V#Creating_a_riscv64_chroot) where you download what you need with apt from Debian's RISCV repository and then you copy all the filesystem to your sdcard.
> 
> But do not expect too much from an FPGA that runs with 50 or 100MHz, limited RAM and one core. I have tested with a simple icewm environment, which I triggered by running vncserver on the FPGA, which then I accessed via network with vncviewer (in the past that I could use liteeth, it is not working for me now).
> 
> Check also this : https://github.com/roryt12/qmtech_wukong_debian_on_litex_naxriscv , it needs some refresh after the last commits, but maybe will give you some hints.

OK, I will try it. Thanks for your help.

Dolu1990, 2023-03-27:
Note OHCI USB host isn't supported yet, as it require memory coherency.