SpinalHDL/NaxRiscv #34: Issue with Make Compile in setup

Pavitra07, 2023-06-22:
On running make compile, i get the following errors. Kindly help.
[image]

Dolu1990, 2023-06-22:
Hi,

which version of GCC do you have ?

(me gcc 10.1.0)

Pavitra07, 2023-06-22:
Hello,

I have the following version:

gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1)

Dolu1990, 2023-06-22:
Ahh it may bit it, string_view seems to be from C++17, gcc 9.4 may be too old ?

Pavitra07, 2023-06-22:
I have upgraded it to the following:
gcc version 10.3.0 (Ubuntu 10.3.0-1ubuntu1~20.04) 
Still showing the same error for make compile.
Should I restart the process?

Dolu1990, 2023-06-22:
hmm pretty sure it is related to  different version of GCC / C++ library.
Issue is that string_view seems c++ 17, but verilator is only c++ 14, so kind of a missmatch XD

Looking at https://github.com/verilator/verilator/issues/3081, seems like it may not be compatible with C++20
May worth a try ?

Getting a Very recent verilator version, and updating https://github.com/SpinalHDL/NaxRiscv/blob/6d1d48379609c5a390f8f4f02527282b21b311b6/src/test/cpp/naxriscv/makefile#L81   with -std=c++20 instead of -std=c++14 ?

Pavitra07, 2023-06-23:
I did the update in the make file regarding c++20, but still facing the same errors. Is there any way I could get through this?

Dolu1990, 2023-06-23:
Found it :)

So, there was a elfio update which broke things.

You just need to reinstall elfio by : 

```
sudo rm -rf /usr/include/elfio
git clone https://github.com/serge1/ELFIO.git
cd ELFIO
git checkout d251da09a07dff40af0b63b8f6c8ae71d2d1938d # Avoid C++17
sudo cp -R elfio /usr/include
```

closed 2023-06-23 by Dolu1990

Pavitra07, 2023-06-23:
Thanks a lot for the help! This absolutely worked :)

commit d7213a3f (2023-06-23): Fix #34 elfio version