SpinalHDL/NaxRiscv #39: Some Riscv-Tests Can't Pass

xie-1399, 2023-07-10:
Hi !
When I excute riscv-tests with the config 
```
    val l = Config.plugins(
      sideChannels = false, //WARNING, FOR TEST PURPOSES ONLY, turn to false for real stuff <3
      xlen = 64,
      withRdTime = false,
      aluCount = 2,
      decodeCount = 2,
      withRvc = true,
      withDebug = false,
      withEmbeddedJtagTap = false,
      debugTriggers = 0,
      withFloat = false,
      withDouble = false,
      lqSize = 16,
      sqSize = 16,
      resetVector = 0x80000000l
    )
```
I find the rv64si-p-wfi rv64mi-p-csr rv64mi-p-scall can't pass the test，can you tell me how to solve the problem? Thank You!

Dolu1990, 2023-07-11:
Hi,

I'm currently in a trip on a dying notebook ^^
I will test it when i'm back in the office (home <3)

Does it tell you where the test are failing ?
It can give a clue ^^

xie-1399, 2023-07-11:
they all fails on the ecall trap，I see the wave about the commit PC.They can commit until pass symbol
[image]
[image]

[image]
[image]

Dolu1990, 2023-07-11:
Ahhhhhhhhh
I were thinking you were using the python script to generate the test list : 
https://github.com/SpinalHDL/NaxRiscv/blob/main/src/test/cpp/naxriscv/testsGen.py

But you aren't right ?
You compiled the test and run them by yourself ?

So one thing about riscv-test, is that some test do assumption about having a core which fully implement the privileged spec, which isn't necessarly the case of NaxRiscv (ex PMP)

How did you ran the tests ?