SpinalHDL/NaxRiscv #40: Performance counters and custom opensbi

denishoornaert, 2023-07-25:
Hey!

I am using Linux on NaxRiscv, and I would like to monitor CPU performance using the `hpmcountern` registers. A quick test led me to believe that the associated bits of `mcountinhibit` are not cleared as reading `hpmcounter3`, `hpmcounter4`, `hpmcounter5`, and `hpmcounter6` return 0.

I understand that I must modify the opensbi in order to change the content of `mcountinhibit`. Even after following the steps listed in [linux-on-vexriscv](https://github.com/litex-hub/linux-on-litex-vexriscv#-generating-the-opensbi-binary-optional), I do not manage to compile a opensbi firmware that successfully executes (i.e., it freezes at "liftoff!"). My guess is that the compiler I use is not the correct one. Could you indicate to me what compiler you are referring to with `riscv-none-embed-` or if I am missing something?

Also, do you think that `mcountinhibit` is the reason why the `hpmcountern` registers are always read as 0?

Thank you in advance for the help!

denishoornaert, 2023-07-27:
I figured out what was wrong with the openSBI. Turns out I must precise `rv32ima_zicsr_zifence` as ISA instead of `rv32imac_zicsr_zifence`.
The compiler I used (compiled from [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain)) was perfectly fine.
For completion, once on the [0.8-linux-on-litex-vexriscv branch](https://github.com/litex-hub/opensbi/tree/0.8-linux-on-litex-vexriscv), I successfully compiled openSBI with:
```bash
make PLATFORM=litex/vexriscv CROSS_COMPILE=/opt/riscv32/bin/riscv32-unknown-elf- PLATFORM_RISCV_XLEN=32 PLATFORM_RISCV_ISA=rv32ima_zicsr_zifencei -j$(nproc)
```

For the performance counters and the `mcountinhibit` register, I will keep on investigating.

Dolu1990, 2023-07-27:
Hi ^^

About the mcountinhibit, i think that feature isn't implemented.
Here is the current implementation : 
https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/misc/PerformanceCounterPlugin.scala

It is quite a bit a "complicated" implementation, reasons are : 
- the 64 bits value of each counter has its 55 MSB bits store into a globaly shared memory in NaxRiscv
- Each counter physicaly has only a physical 7 bits counter dedicated
- For each counter, once that 7 bits counter MSB is set, it will update the 55 MSB bits in the globaly shared memory
  That is done by a shared state machine
  
All of this to save FPGA space XD
- only 7 bits physical counter
- CSR read has 55 bits wide mux done for "free" by the shared memory

denishoornaert, 2023-07-27:
Thanks for the answer!

> About the `mcountinhibit`, i think that feature isn't implemented.
 
That explains why I could not find the register in the sources :sweat_smile:

> It is quite a bit a "complicated" implementation, reasons are :
> - the 64 bits value of each counter has its 55 MSB bits store into a globaly shared memory in NaxRiscv
> - Each counter physicaly has [...]
    That is done by a shared state machine

Thanks for the clarification! That was the understanding I got after reading the sources but I could not pin-point the exact reason of the split.

This is probably for the same reason that the set of currently implemented performance counters is limited, right?

Dolu1990, 2023-07-27:
> This is probably for the same reason that the set of currently implemented performance counters is limited, right?

Hmm so far, i think that if something isn't implemented, that's mostly because it wasn't needed so far.

Using : 
https://github.com/SpinalHDL/NaxRiscv/blob/d7213a3f90ecaf5fb0d1caaa4abfe5af9c06936c/src/main/scala/naxriscv/misc/PerformanceCounterPlugin.scala#L21

You can create new events comming from another plugin

denishoornaert, 2023-08-07:
For the record, it works perfectly on Linux (as expected).

I just had to modify the opensbi to have access to more `mhpmeventX` registers at the same time. If you use the opensbi for VexRiscv, only `mhpmevent3` and `mhpmevent4` can be accessed.

I will play around in the next days and add other performance counters :)

Dolu1990, 2023-08-07:
Ahhh nice ^^
One question is, how do you use the performance counters in linux ?
I never tried ^^

denishoornaert, 2023-08-09:
So far, I am not doing it the smartest way. I just read the counter of interest before and after the execution of my benchmarks :sweat_smile: To ease/automate benchmarking, I use [RT-Bench](https://gitlab.com/rt-bench/rt-bench); a project I work on :smile: 

The obvious limitation is that the performance counter also include the OS activity or other processes.

Using `perf` would be the ideal but I am not up to date with its current state for risc-v.