SpinalHDL/NaxRiscv #41: error for sbt "runMain naxriscv.Gen"

yanjingpo, 2023-08-13:
Log
`[info] welcome to sbt 1.6.0 (Ubuntu Java 11.0.20)
[info] loading settings for project naxriscv-build from plugins.sbt ...
[info] loading project definition from /home/jingpo/Downloads/NaxRiscv/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project spinalhdl-build from plugin.sbt ...
[info] loading project definition from /home/jingpo/Downloads/SpinalHDL/project
[info] loading settings for project all from build.sbt ...
[info] set current project to NaxRiscv (in build file:/home/jingpo/Downloads/NaxRiscv/)
[info] compiling 108 Scala sources to /home/jingpo/Downloads/NaxRiscv/target/scala-2.11/classes ...
[error] error while loading <root>, Error accessing /home/jingpo/.cache/coursier/v1/https/repo1.maven.org/maven2/org/scalactic/scalactic_2.11/3.2.5/scalactic_2.11-3.2.5.jar
[error] ## Exception when compiling 108 sources to /home/jingpo/Downloads/NaxRiscv/target/scala-2.11/classes
[error] scala.reflect.internal.MissingRequirementError: object java.lang.Object in compiler mirror not found.
[error] scala.reflect.internal.MissingRequirementError$.signal(MissingRequirementError.scala:17)
[error] scala.reflect.internal.MissingRequirementError$.notFound(MissingRequirementError.scala:18)
[error] scala.reflect.internal.Mirrors$RootsBase.getModuleOrClass(Mirrors.scala:53)
[error] scala.reflect.internal.Mirrors$RootsBase.getModuleOrClass(Mirrors.scala:45)
[error] scala.reflect.internal.Mirrors$RootsBase.getModuleOrClass(Mirrors.scala:45)
[error] scala.reflect.internal.Mirrors$RootsBase.getModuleOrClass(Mirrors.scala:66)
[error] scala.reflect.internal.Mirrors$RootsBase.getClassByName(Mirrors.scala:102)
[error] scala.reflect.internal.Mirrors$RootsBase.getRequiredClass(Mirrors.scala:105)
[error] scala.reflect.internal.Definitions$DefinitionsClass.ObjectClass$lzycompute(Definitions.scala:257)
[error] scala.reflect.internal.Definitions$DefinitionsClass.ObjectClass(Definitions.scala:257)
[error] scala.reflect.internal.Definitions$DefinitionsClass.init(Definitions.scala:1390)
[error] scala.tools.nsc.Global$Run.<init>(Global.scala:1242)
[error] xsbt.ZincCompiler$ZincRun.<init>(CallbackGlobal.scala:80)
[error] xsbt.CachedCompiler0.run(CompilerBridge.scala:161)
[error] xsbt.CachedCompiler0.run(CompilerBridge.scala:134)
[error] xsbt.CompilerBridge.run(CompilerBridge.scala:39)
[error] sbt.internal.inc.AnalyzingCompiler.compile(AnalyzingCompiler.scala:91)
[error] sbt.internal.inc.MixedAnalyzingCompiler.$anonfun$compile$7(MixedAnalyzingCompiler.scala:192)
[error] scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23)
[error] sbt.internal.inc.MixedAnalyzingCompiler.timed(MixedAnalyzingCompiler.scala:247)
[error] sbt.internal.inc.MixedAnalyzingCompiler.$anonfun$compile$4(MixedAnalyzingCompiler.scala:182)
[error] sbt.internal.inc.MixedAnalyzingCompiler.$anonfun$compile$4$adapted(MixedAnalyzingCompiler.scala:163)
[error] sbt.internal.inc.JarUtils$.withPreviousJar(JarUtils.scala:239)
[error] sbt.internal.inc.MixedAnalyzingCompiler.compileScala$1(MixedAnalyzingCompiler.scala:163)
[error] sbt.internal.inc.MixedAnalyzingCompiler.compile(MixedAnalyzingCompiler.scala:210)
[error] sbt.internal.inc.IncrementalCompilerImpl.$anonfun$compileInternal$1(IncrementalCompilerImpl.scala:528)
[error] sbt.internal.inc.IncrementalCompilerImpl.$anonfun$compileInternal$1$adapted(IncrementalCompilerImpl.scala:528)
[error] sbt.internal.inc.Incremental$.$anonfun$apply$5(Incremental.scala:177)
[error] sbt.internal.inc.Incremental$.$anonfun$apply$5$adapted(Incremental.scala:175)
[error] sbt.internal.inc.Incremental$$anon$2.run(Incremental.scala:461)
[error] sbt.internal.inc.IncrementalCommon$CycleState.next(IncrementalCommon.scala:116)
[error] sbt.internal.inc.IncrementalCommon$$anon$1.next(IncrementalCommon.scala:56)
[error] sbt.internal.inc.IncrementalCommon$$anon$1.next(IncrementalCommon.scala:52)
[error] sbt.internal.inc.IncrementalCommon.cycle(IncrementalCommon.scala:263)
[error] sbt.internal.inc.Incremental$.$anonfun$incrementalCompile$8(Incremental.scala:416)
[error] sbt.internal.inc.Incremental$.withClassfileManager(Incremental.scala:503)
[error] sbt.internal.inc.Incremental$.incrementalCompile(Incremental.scala:403)
[error] sbt.internal.inc.Incremental$.apply(Incremental.scala:169)
[error] sbt.internal.inc.IncrementalCompilerImpl.compileInternal(IncrementalCompilerImpl.scala:528)
[error] sbt.internal.inc.IncrementalCompilerImpl.$anonfun$compileIncrementally$1(IncrementalCompilerImpl.scala:482)
[error] sbt.internal.inc.IncrementalCompilerImpl.handleCompilationError(IncrementalCompilerImpl.scala:332)
[error] sbt.internal.inc.IncrementalCompilerImpl.compileIncrementally(IncrementalCompilerImpl.scala:420)
[error] sbt.internal.inc.IncrementalCompilerImpl.compile(IncrementalCompilerImpl.scala:137)
[error] sbt.Defaults$.compileIncrementalTaskImpl(Defaults.scala:2366)
[error] sbt.Defaults$.$anonfun$compileIncrementalTask$2(Defaults.scala:2316)
[error] sbt.internal.server.BspCompileTask$.$anonfun$compute$1(BspCompileTask.scala:30)
[error] sbt.internal.io.Retry$.apply(Retry.scala:46)
[error] sbt.internal.io.Retry$.apply(Retry.scala:28)
[error] sbt.internal.io.Retry$.apply(Retry.scala:23)
[error] sbt.internal.server.BspCompileTask$.compute(BspCompileTask.scala:30)
[error] sbt.Defaults$.$anonfun$compileIncrementalTask$1(Defaults.scala:2314)
[error] scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error] sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error] sbt.std.Transform$$anon$4.work(Transform.scala:68)
[error] sbt.Execute.$anonfun$submit$2(Execute.scala:282)
[error] sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
[error] sbt.Execute.work(Execute.scala:291)
[... 54 lines cut ...]
[error] 	at sbt.Defaults$.compileIncrementalTaskImpl(Defaults.scala:2366)
[error] 	at sbt.Defaults$.$anonfun$compileIncrementalTask$2(Defaults.scala:2316)
[error] 	at sbt.internal.server.BspCompileTask$.$anonfun$compute$1(BspCompileTask.scala:30)
[error] 	at sbt.internal.io.Retry$.apply(Retry.scala:46)
[error] 	at sbt.internal.io.Retry$.apply(Retry.scala:28)
[error] 	at sbt.internal.io.Retry$.apply(Retry.scala:23)
[error] 	at sbt.internal.server.BspCompileTask$.compute(BspCompileTask.scala:30)
[error] 	at sbt.Defaults$.$anonfun$compileIncrementalTask$1(Defaults.scala:2314)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error] 	at sbt.std.Transform$$anon$4.work(Transform.scala:68)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:282)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
[error] 	at sbt.Execute.work(Execute.scala:291)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:282)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:265)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:64)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
[error] 	at java.base/java.lang.Thread.run(Thread.java:829)
[error] (Compile / compileIncremental) scala.reflect.internal.MissingRequirementError: object java.lang.Object in compiler mirror not found.
[error] Total time: 4 s, completed Aug 13, 2023, 2:00:20 PM`

Dolu1990, 2023-08-13:
Hi,
Damn flow which break itself in the web XD

Could you try by updating https://github.com/SpinalHDL/NaxRiscv/blob/main/project/build.properties to 1.9.3 ?

sssl-yjp, 2023-08-15:
> Hi, Damn flow which break itself in the web XD
> 
> Could you try by updating https://github.com/SpinalHDL/NaxRiscv/blob/main/project/build.properties to 1.9.3 ?

Thanks. 
My solution:
1-update https://github.com/SpinalHDL/NaxRiscv/blob/main/project/build.properties to 1.9.3
2-install jdk-8 `sudo apt install openjdk-8-jdk`
3-change the java version to jdk-8  `sudo update-alternatives --config java` ` sudo update-alternatives --config javac`
4-increase the memory for sbt `export SBT_OPTS="-Xmx4G"`

Dolu1990, 2023-08-15:
Thanks :D

So, step 2,3,4 were necessary ? step 1 wasn't enough :( ?

sssl-yjp, 2023-08-15:
> Thanks :D
> 
> So, step 2,3,4 were necessary ? step 1 wasn't enough :( ?

yes, the problem is the java version,but it also will report a errror about gc overhead limit exceeded when i run the command by using java8.
so step4 increase the memory to solve gc overhead limit exceeded。

my english is poor :)

Dolu1990, 2023-08-15:
Pushed a fix : 
https://github.com/SpinalHDL/SpinalHDL/commit/c8844920e8a0d5e10952d6ef834cc24ede61ab28

This should make it happy even without step 1,2,3,4

closed 2023-08-16 by sssl-yjp