SpinalHDL/NaxRiscv #45: in Verilator terminal for debugging purpose for NAX CPU

HimalSubedi787, 2023-09-21:
If i have to run in Verilator terminal as of II. b) Debug Briey SoC with Verilator +OpenOCD + GDB
https://thuchoang90.github.io/project/2020/07/23/VexRiscv.html 

like src/openocd -f tcl/interface/jtag_tcp.cfg -c "set NAX_CPU0_YAML /home/hsubedi/RISC_V_PROJECT/NaxRiscv/cpu0.yaml" -f tcl/target/Nax.cfg  

Why there is no Nax.cfg in https://github.com/SpinalHDL/openocd_riscv/tree/riscv_spinal/tcl/target

Dolu1990, 2023-09-22:
Mostly, NaxRiscv used the regular RISC-V debug spec instead of a custom one, so doesn't realy need the openocd fork.

tcl script used are in https://github.com/SpinalHDL/NaxRiscv/tree/main/src/main/tcl/openocd

HimalSubedi787, 2023-09-22:
Thank you
 
https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/debug/EmbeddedJtagPlugin.scala#L85

in this github link in from line 85 there is few useful tool to debug but i am not sure how to use it

Dolu1990, 2023-10-31:
EMBEDDED_JTAG is automaticaly set by scala code via getService[DocPlugin].property("EMBEDDED_JTAG","1") which endup into nax.h (generated by DocPlugin) and then i used by the testbench cpp

> who is instantiating EmbeddedJtagPlugin?

It is a optional part of the Config.plugins function : 
https://github.com/SpinalHDL/NaxRiscv/blob/3727ce674d26c98df68e257778aa93da93837c40/src/main/scala/naxriscv/Gen.scala#L91

> Does changing withDebug and withEmbeddedJtagTap to true would help in debugging ?

Yes, you need them both

Dolu1990, 2023-10-31:
Then, in a compiled openocd, i use src/openocd -f ../NaxRiscv/src/main/tcl/openocd/naxriscv_sim.tcl to connect to a running simulation

Dolu1990, 2023-10-31:
But you have to run the simulation with  ./obj_dir/VNaxRiscv --timeout-disable --spike-disable