SpinalHDL/NaxRiscv #51: Whole Soc Component like the Vexriscv

xie-1399, 2023-10-17:
Hi !
Do you have any plans about implement the the whole soc (with spi，uart) , which can boot the linux simulation and show results from the uart?

Dolu1990, 2023-10-17:
Hi,

Are you looking for a soc which can run in simulation, or a soc which can run on hardware ? 

Currently there is :
- toplevel : https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/platform/tilelinkdemo/SocDemo.scala
- testbench : https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/platform/tilelinkdemo/SocSim.scala

which can run multi core linux simulation, but that soc is only meant to be used in simulation via a simulated peripheral/memory

franktaTian, 2023-10-22:
> Hi,
>
> Are you looking for a soc which can run in simulation, or a soc which can run on hardware ?
>
> Currently there is :
>
> * toplevel : [...]

Can you update the document for how to config multicore and run SoCSim with Linux ?Thanks.

Dolu1990, 2023-10-23:
Added via https://github.com/SpinalHDL/NaxRiscv/blob/c8ee3a2ca5253f46f07241effe981aabed731801/src/main/scala/naxriscv/platform/tilelinkdemo/SocSim.scala#L20

Now you can use arguments to run the sim.
Also added some reference to it from the RTD

franktaTian, 2023-10-25:
> Added via
> 
> https://github.com/SpinalHDL/NaxRiscv/blob/c8ee3a2ca5253f46f07241effe981aabed731801/src/main/scala/naxriscv/platform/tilelinkdemo/SocSim.scala#L20
> 
> Now you can use arguments to run the sim. Also added some reference to it from the RTD

In SocSim.scala :
--load-bin 80F80000,ext/NaxSoftware/buildroot/images/rv32ima/linux_1c.dtb   does not  exit 
only linux.dtb, it can boot linux , but there is only one core

And ,since
val compiled = sc.compile(new SocDemoSim(cpuCount = 1))
naxcount indeed does not take effect.

Dolu1990, 2023-10-25:
Hoo right, i pushed fixes.
Thanks

Dolu1990, 2023-10-25:
Here are the other dtb 
[dtb_dts.zip](https://github.com/SpinalHDL/NaxRiscv/files/13162576/dtb_dts.zip)