SpinalHDL/NaxRiscv #57: Placing memory region at address 0 causes MatchError exception

pftbest, 2023-10-28:
Hello!
I'm trying to create a custom configuration where DDR memory starts at address 0 and other regions are at 0x40000000+.

But as soon as I remove "p" type region from address 0 I get an exception. Could you please help me figure out what is wrong?

```
$ sbt "runMain naxriscv.platform.litex.NaxGen \
--netlist-name=NaxRiscvLitex_5x --netlist-directory=. \
--reset-vector=0 --xlen=32 --cpu-count=1 --l2-bytes=131072 \
--l2-ways=8 --litedram-width=128 \
--memory-region=1073741824,3221225472,io,p \
--memory-region=0,1073741824,rwxc,m \
--memory-region=1073741824,1073741824,rw,p \
--memory-region=4026531840,65536,rw,p \
--scala-file=/Users/vadzim/Downloads/ecp5/litex/pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/configs/gen.scala"

[info] welcome to sbt 1.6.0 (Azul Systems, Inc. Java 17.0.7)
[info] loading settings for project naxriscv-build from plugins.sbt ...
[info] loading project definition from /Users/vadzim/Downloads/ecp5/litex/pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/ext/NaxRiscv/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project spinalhdl-build from plugin.sbt ...
[info] loading project definition from /Users/vadzim/Downloads/ecp5/litex/pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/ext/NaxRiscv/ext/SpinalHDL/project
[info] loading settings for project all from build.sbt ...
[info] set current project to NaxRiscv (in build file:/Users/vadzim/Downloads/ecp5/litex/pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/ext/NaxRiscv/)
[info] running (fork) naxriscv.platform.litex.NaxGen --netlist-name=NaxRiscvLitex_5x --netlist-directory=. --reset-vector=0 --xlen=32 --cpu-count=1 --l2-bytes=131072 --l2-ways=8 --litedram-width=128 --memory-region=1073741824,3221225472,io,p --memory-region=0,1073741824,rwxc,m --memory-region=1073741824,1073741824,rw,p --memory-region=4026531840,65536,rw,p --scala-file=/Users/vadzim/Downloads/ecp5/litex/pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/configs/gen.scala
[info] [Runtime] SpinalHDL dev    git head : 9c82ff197e1ca17e98af75a77b470a0fda61fcfd
[info] [Runtime] JVM max memory : 4096.0MiB
[info] [Runtime] Current date : 2023.10.28 22:22:56
[info] [Progress] at 0.000 : Elaborate components
[info] memoryRegions: Seq[naxriscv.platform.litex.LitexMemoryRegion] = ArrayBuffer(LitexMemoryRegion(SM(0x40000000, 0xc0000000),io,p), LitexMemoryRegion(SM(0x0, 0x40000000),rwxc,m), LitexMemoryRegion(SM(0x40000000, 0x40000000),rw,p), LitexMemoryRegion(SM(0xf0000000, 0x10000),rw,p))
[info] LitexMemoryRegion(SM(0x40000000, 0xc0000000),io,p)
[info] LitexMemoryRegion(SM(0x0, 0x40000000),rwxc,m)
[info] LitexMemoryRegion(SM(0x40000000, 0x40000000),rw,p)
[info] LitexMemoryRegion(SM(0xf0000000, 0x10000),rw,p)
[info] import scala.collection.mutable.ArrayBuffer
[info] import naxriscv.utilities.Plugin
[info] import naxriscv.platform.litex.LitexMemoryRegion
[info] import spinal.lib.bus.misc.SizeMapping
[info] plugins: scala.collection.mutable.ArrayBuffer[naxriscv.utilities.Plugin] = ArrayBuffer(DocPlugin, MmuPlugin, FetchPlugin, PcPlugin, FetchCachePlugin, AlignerPlugin, FrontendPlugin, DecompressorPlugin, DecoderPlugin, integer_RfTranslationPlugin, RfDependencyPlugin, integer_RfAllocationPlugin, DispatchPlugin, BranchContextPlugin, HistoryPlugin, DecoderPredictionPlugin, BtbPlugin, GSharePlugin, Lsu2Plugin, DataCachePlugin, RobPlugin, CommitPlugin, integer_RegFilePlugin, CommitDebugFilterPlugin, CsrRamPlugin, PrivilegedPlugin, PerformanceCounterPlugin, ALU0_ExecutionUnitBase, ALU0_IntFormatPlugin, ALU0_SrcPlugin, ALU0_IntAluPlug...
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 2.709 : Elaborate components
[info] memoryRegions: Seq[naxriscv.platform.litex.LitexMemoryRegion] = ArrayBuffer(LitexMemoryRegion(SM(0x40000000, 0xc0000000),io,p), LitexMemoryRegion(SM(0x0, 0x40000000),rwxc,m), LitexMemoryRegion(SM(0x40000000, 0x40000000),rw,p), LitexMemoryRegion(SM(0xf0000000, 0x10000),rw,p))
[info] LitexMemoryRegion(SM(0x40000000, 0xc0000000),io,p)
[info] LitexMemoryRegion(SM(0x0, 0x40000000),rwxc,m)
[info] LitexMemoryRegion(SM(0x40000000, 0x40000000),rw,p)
[info] LitexMemoryRegion(SM(0xf0000000, 0x10000),rw,p)
[info] import scala.collection.mutable.ArrayBuffer
[info] import naxriscv.utilities.Plugin
[info] import naxriscv.platform.litex.LitexMemoryRegion
[info] import spinal.lib.bus.misc.SizeMapping
[info] plugins: scala.collection.mutable.ArrayBuffer[naxriscv.utilities.Plugin] = ArrayBuffer(DocPlugin, MmuPlugin, FetchPlugin, PcPlugin, FetchCachePlugin, AlignerPlugin, FrontendPlugin, DecompressorPlugin, DecoderPlugin, integer_RfTranslationPlugin, RfDependencyPlugin, integer_RfAllocationPlugin, DispatchPlugin, BranchContextPlugin, HistoryPlugin, DecoderPredictionPlugin, BtbPlugin, GSharePlugin, Lsu2Plugin, DataCachePlugin, RobPlugin, CommitPlugin, integer_RegFilePlugin, CommitDebugFilterPlugin, CsrRamPlugin, PrivilegedPlugin, PerformanceCounterPlugin, ALU0_ExecutionUnitBase, ALU0_IntFormatPlugin, ALU0_SrcPlugin, ALU0_IntAluPlug...
[error] Exception in thread "main" scala.MatchError: spinal.lib.bus.misc.NeverMapping$@4340533c (of class spinal.lib.bus.misc.NeverMapping$)
[error] 	at spinal.lib.bus.misc.SizeMapping.spinal$lib$bus$misc$SizeMapping$$rec$1(Misc.scala:300)
[error] 	at spinal.lib.bus.misc.SizeMapping.intersectImpl(Misc.scala:315)
[error] 	at spinal.lib.bus.misc.AddressMapping$class.intersect(Misc.scala:111)
[error] 	at spinal.lib.bus.misc.SizeMapping.intersect(Misc.scala:236)
[error] 	at spinal.lib.system.tag.MemoryConnection$$anonfun$getMemoryTransfers$2$$anonfun$2.apply(Bus.scala:105)
[error] 	at spinal.lib.system.tag.MemoryConnection$$anonfun$getMemoryTransfers$2$$anonfun$2.apply(Bus.scala:102)
[error] 	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:234)
[error] 	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:234)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at scala.collection.TraversableLike$class.map(TraversableLike.scala:234)
[error] 	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
[error] 	at spinal.lib.system.tag.MemoryConnection$$anonfun$getMemoryTransfers$2.apply(Bus.scala:102)
[error] 	at spinal.lib.system.tag.MemoryConnection$$anonfun$getMemoryTransfers$2.apply(Bus.scala:100)
[error] 	at spinal.lib.system.tag.MappedNode$$anonfun$foreachSlave$1.apply(Bus.scala:69)
[error] 	at spinal.lib.system.tag.MappedNode$$anonfun$foreachSlave$1.apply(Bus.scala:66)
[error] 	at scala.collection.mutable.LinkedHashSet.foreach(LinkedHashSet.scala:93)
[error] 	at spinal.core.SpinalTagReady$class.foreachTag(Trait.scala:696)
[error] 	at spinal.lib.bus.fabric.Node.foreachTag(Node.scala:6)
[error] 	at spinal.lib.system.tag.MappedNode.foreachSlave(Bus.scala:66)
[error] 	at spinal.lib.system.tag.MemoryConnection$.getMemoryTransfers(Bus.scala:100)
[error] 	at spinal.lib.system.tag.MemoryConnection$.getMemoryTransfers(Bus.scala:84)
[error] 	at spinal.lib.bus.tilelink.fabric.TransferFilter$$anonfun$1$$anon$1.<init>(TransferFilter.scala:64)
[error] 	at spinal.lib.bus.tilelink.fabric.TransferFilter$$anonfun$1.apply(TransferFilter.scala:42)
[error] 	at spinal.lib.bus.tilelink.fabric.TransferFilter$$anonfun$1.apply(TransferFilter.scala:42)
[error] 	at spinal.core.fiber.Fiber$$anonfun$1.apply(Fiber.scala:32)
[error] 	at spinal.core.fiber.package$$anonfun$1.apply$mcV$sp(package.scala:16)
[error] 	at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
[error] 	at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
[error] 	at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 7 s, completed Oct 28, 2023, 10:22:59 PM
```

Dolu1990, 2023-10-30:
Hi,

I'm looking into it, seems something is wrong on the interconnect elaboration side of things, not your fault ^^

I'm investigating

Dolu1990, 2023-10-30:
Hi,
I pushed the fix into NaxRiscv main branch
Should be good now, let's me know if you have any other issue ^^

closed 2023-10-30 by Dolu1990

pftbest, 2023-10-30:
Thank you very much! It works fine now.

commit 00c8c7f3 (2023-10-30): Fix #57 litex memory mapping at offset 0