SpinalHDL/NaxRiscv #59: Slow SRAM

zyn810039594, 2023-11-01:
I just want to use Nax as a embedded core of MCU, but the SRAM the fab provides can't run at the frequency that CPU can do. My solution is to integrate a half-speed SRAM. So, is there anyway to increase the number of cycles for cache accessing the SRAM?  Or, is there a better idea?

Dolu1990, 2023-11-01:
the issue is in the data cache / instruction cache ? or you mean the main SRAM of the chip, (the peripheral)

zyn810039594, 2023-11-01:
Yes, the sram in the data cache and instruction cache. The sram that fab gives can only run at 200MHz, but the logic of the nax core can run much higher in this process.

Dolu1990, 2023-11-02:
natively, there is no option to have a multi cycle ram for the caches.
Technicaly speaking, i think it is easily faisable for the I$, at the cost of performance.

It may also be faisable for the D$, but that a bit more tricky, as the LSU is interacting with it.

Hmm, so, even with a fully relax data path on the SRAM read data, the fmax is 200 Mhz ?
or 200 Mhz is including the datapath of sram + nax ?

Dolu1990, 2023-11-02:
May i ask which fab it is ?

zyn810039594, 2023-11-02:
> natively, there is no option to have a multi cycle ram for the caches. Technicaly speaking, i think it is easily faisable for the I$, at the cost of performance.
>
> It may also be faisable for the D$, [...]

The max speed of SRAM block itself is about 280MHz, but I need to support ECC... (The fmax of nax without SRAM is up to 350MHz in this process)

> May i ask which fab it is ?

Ahhhh, UMC, with eflash process.

zyn810039594, 2023-11-08:
And the large rams in Nax (BTB,GShare,Cache) are dual-port rams. In simulation, these pieces of RAM all have the situation of simultaneously reading and writing data to the same address. So are they read-first? Or write-first?

Dolu1990, 2023-11-10:
> And the large rams in Nax (BTB,GShare,Cache) are dual-port rams. In simulation, these pieces of RAM all have the situation of simultaneously reading and writing data to the same address. So are they read-first? Or write-first?

The CPU is designed to not care about the readed value when there is conflicts. The only thing which matther to it is that it doesn't generate metastable value i would say.

closed 2024-01-08 by zyn810039594