SpinalHDL/NaxRiscv #64: naxriscv boot on tang_primer 20k

54o8, 2023-11-09:
When I tried to build the soc for naxriscv on tang_primer_20k, there was an error. Do you have any suggestions?
[image]
I used the following command
[image]

Dolu1990, 2023-11-09:
Hi,

NaxRiscv is likely too big for it. Mainly because it assumes that the FPGA support distributed ram (lut ram, meaning 1w port, 1r asyncronous port)
It extensively use them to infer the various data structure required for out of order stuff.

54o8, 2023-11-09:
Thank you for your answer
I  then used another 138k board of Gao Yun, but after the bit stream download, I got stuck in litex-term, and nothing happened when I pressed the enter key

 
[image]

enjoy-digital, 2023-11-09:
@54o8: Can you share your build command? We could do some test here with @trabucayre that also has the board. Is it working with other CPU (ex with --cpu-type=vexriscv?) Since it seems there is an issue on the UART TX/RX pins.

enjoy-digital, 2023-11-09:
@54o8: It could be worth testing with https://github.com/litex-hub/litex-boards/commit/17a0152ef9d8297585ce390e08afe1ebc84f9e56 and upstream LiteX/LiteDRAM.

54o8, 2023-11-09:
@enjoy-digital 
hi,I will try my best to explain the problems I encountered in the 138k board, it may be a little long, hope you can be patient to read it

The command I use is as follows
[image]
When I use cpu-type=vexriscv_smp
[image]
At least when litex-term is used(cpu-type=vexriscv_smp),   LITEX >   can appear;(Output to another screen via hdmi),when cpu-type=vexriscv_smp ,The screen has no output, and litex-term is unresponsive
[image]

by the way，How do you solve the DDR problem of the gowin138k board?

As far as I know, gowin doesn't give a DDRPHY for 138k, similar to GW2DDRPHY for 20k boards, so does that mean I have to write my own DDRPHY when I want to use DDR on 138k board?
[image]

or as in target.sipeed_tang_mega_138k.py using external sdram to solve this problem
[image]
But when I tried this method, I also made a mistake. When I did the sdram_test in litex
[image]

[image]

enjoy-digital, 2023-11-09:
So, there are different issues.

For NaxRiscv, from what I undersand, it has never been tested on a Gowin FPGA, so it would be interesting to carefully check the synthesis logs and also to make sure it passes timings. You could eventually try to reduce `sys_clk_freq`.

For the DDR3, it's not yet supported with LiteX/LiteDRAM, but we are working on it (ex: https://github.com/enjoy-digital/litedram/pull/351).

For the SDRAM, with upstream code, it should be working with `--with-sdram`.

Dolu1990, 2023-11-09:
> For NaxRiscv, from what I undersand, it has never been tested on a Gowin FPGA, so it would be interesting to carefully check the synthesis logs and also to make sure it passes timings

Yes right.

Also on thing which will help a lot with the size of the design is : 
--scala-args='rvc=false,rvf=false,rvd=false,mmu=false,alu-count=1,decode-count=1' 

(not i do not often use mmu=false)

54o8, 2023-11-09:
@enjoy-digital  
hi , 
i set with_ddr3 = turn ,
[image]
but it doesn't seem to work (no ddr3 connected to the cpu)
[image]
[image]
Do you have any solutions?

trabucayre, 2023-11-09:
please use `--with-ddr3`: default value for constructor is overriden by args

54o8, 2023-11-09:
HI,can you give me some advice?
bad thing happened again, litex sdram-init failed[image]

54o8, 2023-11-09:
> 嗨，你能给我一些建议吗？坏事又发生了，litex sdram-init 失败了[image]

Do I need to change the timing or anything? Is there any way to solve this problem

trabucayre, 2023-11-09:
DDR3 for tang Mega it's currently not working. I try to find why but gowin doesn't provides any real/effective documentations so it's not really easy.

54o8, 2023-11-09:
Thanks for contributing to this, bset wish!

54o8, 2023-11-09:
by the way ,i also tried another method, which was to use peripheral sdram, but there were still data errors
[image]

 I think it may be the dm problem, because the dm signal line of the sdram peripheral I used is multiplexed with A11 and A12 address lines (the schematic diagram is as follows).
[image]
So I re-modified the signal line, used the dm as a separate pin, and made a new pcb
[image]
 
[image]
I wonder if this modification can have a positive effect?
Do you have any suggestions for that？

54o8, 2023-11-21:
hi，
I used the peripheral sdram and passed sdram_init,
[image]
 but linux still wouldn't boot properly
[image]
Is the problem caused by the dtb file?

Dolu1990, 2023-11-25:
Hi,

It seems more like some memory issue. 0x5c2ff isn't a valid instruction. Can you check the linux binary,? what is at 0xC0053200 ?