SpinalHDL/NaxRiscv #67: [Performance bug] Incorrect register dependency tracking

ronan-lashermes, 2023-11-13:
Got a slight issue with the following Spectre-BTB gadget. It appears that there is a dependency where it shouldn’t.
```
.global gadget
gadget:
  # Windowing gadget
  addi sp, sp,-8
  li t0, 1
  # Slow down the commits using dependencies, also slow down the branch execution to avoid pipeline flush too early
  div t0, jump_dst, t0
  # Triggering speculation
  jalr x0, t0

.global gadget_middle
gadget_middle:
  # Access secret
  lb t0, 0(secret_ptr)
  # Select one bit of secret
  srl t0, t0, bit_select # shift bit
  andi t0, t0, 1 # select bit
  slli t0, t0, 10 # 6 = LOG2_BYTES_PER_LINE # set to size of cache line or 0 
  or t0, t0, shared_mem_ptr # compute address in shared mem depending on bit
  # address is therefore either shared_mem_ptr or shared_mem_ptr + 64
  # Access shared memory
  lb t0, 0(t0)

.global gadget_end
gadget_end:
  addi sp, sp, 8
  ret
```
gives with konata.
[image]

The prediction is correct but the `lb` after the jump does not execute (because of `t0` dependency ?).
The correct speculative execution is flushed and reexecuted.

If I replace `t0` with `t1` in the windowing gadget
```
li t1, 1
div t1, jump_dst, t1
# Triggering speculation
jalr x0, t1
```
everything is working correctly.

[image]

I am not familiar enough with the core for a proper debug. But if you have some ideas where to look for, I can give it a try.

Dolu1990, 2023-11-14:
Hi,

Yes, that's curious.
Can your software code + binaries + assembly code so i can reproduce ? Woud need to check what is reaching the LSU / dispatch

thanks ^^

ronan-lashermes, 2023-11-14:
You can find all that here : [L4-SpectreBTB.zip](https://github.com/SpinalHDL/NaxRiscv/files/13347942/L4-SpectreBTB.zip)
This is derived from your SideChannels folder so you may copy/paste only parts of the code and it should work.
(I am preparing a practical on microarchitecture security with these)

Dolu1990, 2023-11-14:
Thanks for the zip ^^
i can reproduce.

I checked and all of this seems normal, for non-obvious reasons XD
So, but your sit belt : 
What is happening on your first screenshot is the following : 
1) PC=0x80000578 is interpreted by the post decode predictor as a JALR which should use the RAS to pop where to go.
https://github.com/SpinalHDL/NaxRiscv/blob/ec3ee4dc0d7e6351f2df8c2ca7ad136c224d6321/src/main/scala/naxriscv/prediction/DecoderPredictionPlugin.scala#L162
From the RISC-V spec : 
-"A JAL instruction should push the return address onto a return-address stack (RAS) only when rd=x1/x5"
- x5 => t0
This will override the prediction made by the BTB, by the one from the RAS, throwing the 0x8000057C aways and "wrongly" correcting the PC to 0x80000294
So, avoid using t0 for jumps ^^

closed 2023-11-14 by Dolu1990

ronan-lashermes, 2023-11-15:
Wow, nice catch indeed. I forgot that t0 was allowed to impact the RAS. Thanks !

Dolu1990, 2023-11-15:
> I forgot that t0 was allowed to impact the RAS

Same XD