SpinalHDL/NaxRiscv #69: Failing to reproduce the NaxRiscv/Debian setup

newinnovations, 2023-11-15:
I am trying to reproduce the NaxRiscv/Debian setup as described [here](https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/hardware/index.html) and am running into some issues.

Steps I have untertaken:

* Downloaded the files from the google drive and put them on an sd-card according to the instructions.
* Created the gateware:

```text
python3 -m litex_boards.targets.digilent_nexys_video --build --no-compile-gateware \
    --cpu-type=naxriscv \
    --bus-standard axi-lite \
    --with-video-framebuffer \
    --with-spi-sdcard \
    --with-ethernet \
    --xlen=64 --scala-args='rvc=true,rvf=true,rvd=true' \
    --csr-csv $HOME/build/csr.csv \
    --csr-json $HOME/build/csr.json
```

* Created the bitstream using Vivado 2023.2
* Loaded the bitstream to my Digilent Nexys Video board

```text
openFPGALoader -b nexysVideo --freq 15M digilent_nexys_video.bin
```

## First issue

The fist issue I ran into is a "hang" after printing the memory write speed. Pressing reset the first time leads to a "hang" at the same spot. After pressing reset a second time the boot continues and I am running into the second issue.

```text
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Nov 15 2023 07:56:26
 BIOS CRC passed (70194846)

 LiteX git sha1: 6d9cacd4

--=============== SoC ==================--
CPU:            NaxRiscv @ 100MHz
BUS:            AXI-LITE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
L2:             8.0KiB
SDRAM:          512.0MiB 16-bit @ 800MT/s (CL-7 CWL-5)
MAIN-RAM:       512.0MiB

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |01111111111111100000000000000000| delays: 07+-06
  m0, b03: |00000000000000000011111111111110| delays: 24+-06
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 07+-06
  m1, b00: |00000000000000000000000000000000| delays: -
[... 292 lines cut ...]
[   48.254092] [<ffffffff80397966>] mmc_start_host+0x80/0x8c
[   48.259560] [<ffffffff80398cf0>] mmc_add_host+0x28/0x3c
[   48.264856] [<ffffffff803a5cc6>] mmc_spi_probe+0x2e8/0x558
[   48.270412] [<ffffffff8060007e>] rdinit_setup+0x28/0x2c
[   48.275711] [<ffffffff80348e06>] really_probe.part.0+0x54/0x22c
[   48.281700] [<ffffffff803490b2>] driver_probe_device+0x26/0xf0
[   48.287605] [<ffffffff80349386>] __driver_attach+0x86/0x150
[   48.293245] [<ffffffff803492fc>] __device_attach_driver+0xfc/0x100
[   48.299498] [<ffffffff803476d2>] bus_for_each_dev+0x3c/0x64
[   48.305143] [<ffffffff80347d4c>] bus_add_driver+0xd2/0x180
[   48.310698] [<ffffffff80349d38>] driver_register+0x46/0xd8
[   48.316254] [<ffffffff80615da2>] mmc_blk_init+0x10c/0x110
[   48.321723] [<ffffffff80600c9e>] do_one_initcall+0x5e/0x110
[   48.327367] [<ffffffff8060007e>] rdinit_setup+0x28/0x2c
[   48.332666] [<ffffffff80600f5c>] kernel_init_freeable+0x1a8/0x226
[   48.338831] [<ffffffff804cacb4>] rest_init+0xc8/0xcc
[   48.343861] [<ffffffff804cacce>] kernel_init+0x16/0xfe
[   48.349070] [<ffffffff804cacb4>] rest_init+0xc8/0xcc
[   48.354105] [<ffffffff80001c94>] ret_from_syscall_rejected+0x8/0xc
```

This is bitstream I am using: 
[digilent_nexys_video.bin.gz](https://github.com/SpinalHDL/NaxRiscv/files/13364307/digilent_nexys_video.bin.gz)

Hope you can help me, I really would like to get this working.

Dolu1990, 2023-11-15:
Hi,

1) Yes i noticed that sometime, there is some weird hang in the litex bios, so far after doing reset it goes away most of the time.
I didn't tracked the exact source of the issue so far.

2) Ahh i recently tested using https://github.com/SpinalHDL/NaxSoftware/blob/main/debian_litex/README.md which use the native MMC litex interface (not the SPI one)

What kernel were you using ? all the ones from the readthedoc ? you didn't recompiled anything right ?
Maybe the address mapping of the SPI peripheral moved. What dts are you using / does the spi controller address of the generated soc match the dts/dtb you are using ?

Note, i'm mostly based on https://github.com/SpinalHDL/NaxSoftware/blob/main/debian_litex/README.md

newinnovations, 2023-11-15:
Thanks for your quick reaction. 

Reassuring to hear that you experienced the same bios/boot hang. I was wondering whether it may be a problem with my board. Not a major issue, but it would be nice if we could resolve it. 

I was indeed using all the files from the readthedoc, including the dtb. And your guess was correct, the address mapping of the SPI has moved from spi@f0004800 to spi@f0005800.

After updating the dts and putting the corresponding dtb on the card, I was able to boot debian. Nice! Thanks!

I will now try to create a system based on your newer instructions.

Dolu1990, 2023-11-15:
Nice :D

Thanks for the update.
Let's me know if you have any issue with the new instructions / there is missing information, especialy on the litex side of things.

newinnovations, 2023-11-16:
I started with upgrading my config to the new instructions on step at a time:

1. opensbi
2. kernel
3. naxriscv bitstream with dma/non-spi sd/multi cpu
4. rootfs

Upgrading the opensbi was not successful. This is what I did:

(I changed the CROSS_COMPILE to riscv64-unknown-elf-)

```
git clone https://github.com/litex-hub/opensbi --branch 1.3.1-linux-on-litex-vexriscv
cd opensbi/
make CROSS_COMPILE=riscv64-unknown-elf- PLATFORM=litex/vexriscv
cp build/platform/litex/vexriscv/firmware/fw_jump.bin ../part1/opensbi.bin
```
This leads to a hang at liftoff:

```
...
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 163.6MiB/s
   Read speed: 215.4MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
Copying Image to 0x41000000 (15451440 bytes)...
[########################################]
Copying linux.dtb to 0x46000000 (2630 bytes)...
[########################################]
Copying opensbi.bin to 0x40f00000 (132040 bytes)...
[########################################]
Executing booted program at 0x40f00000

--============= Liftoff! ===============--
<<<HANGS>>>
```

What I find strange is that you create the opensbi with the config of VexRiscv, which leads to a 32-bit executable.

Then I created the new linux 6.1-rc2 linux kernel per your instructions, which went fine and booted correctly with the old opensbi:

```
...
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 164.2MiB/s
   Read speed: 243.5MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Booting from SDCard in SPI-Mode...
Booting from boot.json...
Copying Image to 0x41000000 (17603072 bytes)...
[########################################]
Copying linux.dtb to 0x46000000 (2630 bytes)...
[########################################]
Copying opensbi.bin to 0x40f00000 (49992 bytes)...
[########################################]
Executing booted program at 0x40f00000

--============= Liftoff! ===============--

OpenSBI v0.9
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : LiteX / VexRiscv-SMP
Platform Features         : timer,mfdeleg
Platform HART Count       : 8
Firmware Base             : 0x40f00000
Firmware Size             : 124 KB
Runtime SBI Version       : 0.2

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*,2*,3*,4*,5*,6*,7*
Domain0 Region00          : 0x0000000040f00000-0x0000000040f1ffff ()
Domain0 Region01          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000041000000
Domain0 Next Arg1         : 0x0000000046000000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART ISA             : rv64imafdcsu
Boot HART Features        : scounteren,mcounteren,time
Boot HART PMP Count       : 0
Boot HART PMP Granularity : 0
Boot HART PMP Address Bits: 0
Boot HART MHPM Count      : 4
Boot HART MHPM Count      : 4
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109
[    0.000000] Linux version 6.1.0-rc2+ (embryonic@litex) (riscv64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP Thu Nov 16 12:27:06 UTC 2023
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000041000000-0x000000005fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000041000000-0x000000005fffffff]
[    0.000000] SBI specification v0.2 detected
[    0.000000] SBI implementation ID=0x1 Version=0x9
[    0.000000] SBI TIME extension detected
[    0.000000] SBI IPI extension detected
[    0.000000] SBI RFENCE extension detected
[    0.000000] SBI HSM extension detected
[    0.000000] riscv: base ISA extensions acdfim
[    0.000000] riscv: ELF capabilities acdfim
....
``` 

I will now start working on updating the bitstreams, but wanted to let you know about my progress and hear your ideas on the opensbi issue.

Dolu1990, 2023-11-16:
Ahhhh right, i had some local changes about opensbi.
I created a fork, and changed the readme instructions.
with that updated https://github.com/SpinalHDL/NaxSoftware/blob/main/debian_litex/README.md#compile-opensbi it should be good.
Let's me know ^^

newinnovations, 2023-11-16:
That solved it:

```
--============= Liftoff! ===============--

OpenSBI v0.9-614-g034d47a
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : LiteX / NaxRiscv-SMP
Platform Features         : medeleg
Platform HART Count       : 8
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 100000000Hz
Platform Console Device   : litex_uart
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Platform Suspend Device   : ---
Platform CPPC Device      : ---
Firmware Base             : 0x40f00000
Firmware Size             : 248 KB
Firmware RW Offset        : 0x20000
Firmware RW Size          : 120 KB
Firmware Heap Offset      : 0x32000
Firmware Heap Size        : 48 KB (total), 3 KB (reserved), 8 KB (used), 36 KB (free)
Firmware Scratch Size     : 4096 B (total), 736 B (used), 3360 B (free)
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*,2*,3*,4*,5*,6*,7*
Domain0 Region00          : 0x00000000f0018000-0x00000000f001bfff M: (I,R,W) S/U: ()
Domain0 Region01          : 0x00000000f0010000-0x00000000f0017fff M: (I,R,W) S/U: ()
Domain0 Region02          : 0x0000000040f00000-0x0000000040f1ffff M: (R,X) S/U: ()
Domain0 Region03          : 0x0000000040f20000-0x0000000040f3ffff M: (R,W) S/U: ()
Domain0 Region04          : 0x0000000000000000-0xffffffffffffffff M: (R,W,X) S/U: (R,W,X)
Domain0 Next Address      : 0x0000000041000000
Domain0 Next Arg1         : 0x0000000046000000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.10
Boot HART Base ISA        : rv64imafdc
Boot HART ISA Extensions  : zicntr,zihpm
Boot HART PMP Count       : 0
Boot HART PMP Granularity : 0
Boot HART PMP Address Bits: 0
Boot HART MHPM Count      : 4
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109
[    0.000000] Linux version 6.1.0-rc2+ (embryonic@litex) (riscv64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP Thu Nov 16 12:27:06 UTC 2023
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Zone ranges:
```

Dolu1990, 2023-11-16:
great :D

newinnovations, 2023-11-20:
Now I am running into problems with the network. DHCP is not working and when statically configured I get:

```
root@sid-rv64:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 7e:4a:a0:19:59:68 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.50/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::7c4a:a0ff:fe19:5968/64 scope link
       valid_lft forever preferred_lft forever
root@sid-rv64:~# ping 192.168.1.100
PING 192.168.1.100 (192.168.1.100) 56(84) bytes of data.
From 192.168.1.50 icmp_seq=1 Destination Host Unreachable
From 192.168.1.50 icmp_seq=2 Destination Host Unreachable
From 192.168.1.50 icmp_seq=3 Destination Host Unreachable
From 192.168.1.50 icmp_seq=4 Destination Host Unreachable
From 192.168.1.50 icmp_seq=5 Destination Host Unreachable
From 192.168.1.50 icmp_seq=6 Destination Host Unreachable
^C
--- 192.168.1.100 ping statistics ---
7 packets transmitted, 0 received, +6 errors, 100% packet loss, time 6074ms
pipe 3
root@sid-rv64:~# ifdown eth0
root@sid-rv64:~# ifup eth0
[  158.218664] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
root@sid-rv64:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 7e:4a:a0:19:59:68 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.50/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::7c4a:a0ff:fe19:5968/64 scope link
       valid_lft forever preferred_lft forever
root@sid-rv64:~# ping 192.168.1.100
PING 192.168.1.100 (192.168.1.100) 56(84) bytes of data.
From 192.168.1.50 icmp_seq=1 Destination Host Unreachable
From 192.168.1.50 icmp_seq=2 Destination Host Unreachable
From 192.168.1.50 icmp_seq=3 Destination Host Unreachable
From 192.168.1.50 icmp_seq=4 Destination Host Unreachable
^C
--- 192.168.1.100 ping statistics ---
7 packets transmitted, 0 received, +4 errors, 100% packet loss, time 6073ms
pipe 3
root@sid-rv64:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 7e:4a:a0:19:59:68 brd ff:ff:ff:ff:ff:ff
```

Dolu1990, 2023-11-23:
Hi,

I didn't had particular with ethernet in debian, could it be aswell that the DTS memory address / interrupt id for the ethernet peripheral changed ?
Can you check the eth0 stats ? (ifconfig eth0)
        RX packets 55249  bytes 65328784 (65.3 MB)
        RX errors 0  dropped 300  overruns 0  frame 0
        TX packets 12857  bytes 2470859 (2.4 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

to see if the trafic goes both way

newinnovations, 2023-11-24:
For your information my setup is now as follows. I created the bitmap using the following command:

```
python3 -m litex_boards.targets.digilent_nexys_video --build --no-compile-gateware \
    --cpu-type=naxriscv \
    --bus-standard axi-lite \
    --with-video-framebuffer \
    --with-coherent-dma \
    --with-sdcard \
    --with-ethernet \
    --xlen=64 --scala-args='rvc=true,rvf=true,rvd=true,alu-count=2,decode-count=2' \
    --uart-baudrate 3000000 \
    --with-jtag-tap \
    --sys-clk-freq 75000000 \
    --cpu-count 2 \
    --csr-csv $HOME/build/csr.csv \
    --csr-json $HOME/build/csr.json
```

I reduced the speed to 75MHz as I noticed that Vivado signals that timing cannot be met at 100MHz. At 75MHz the booting hangs are completely resolved and I can use the SD card in non-SPI mode. At 100MHz the core was unable to read the files reliable from SD to bootstrap everything. And I increased the serial speed to 3Mbps (for serial boots).

The dts is generated from csr.json, with the following modifications:

* CPU ISA changed to `rv64imafdc`
* Memory starts at `0x41000000` length reduced to `0x1f000000`
* Reserved memory areas removed
* LEDs removed
* Chosen initrd fields removed
* Bootargs set to `console=hvc0 earlycon=sbi root=/dev/mmcblk0p2 rootfstype=ext4 rw swiotlb=noforce rootwait rcupdate.rcu_cpu_stall_timeout=2`

These are all the files I use, if you want to reproduce:

[naxriscv.tar.gz](https://github.com/SpinalHDL/NaxRiscv/files/13456611/naxriscv.tar.gz)

Below are the requested ifconfig logs:

```
sid-rv64 login: root
Password:
Linux sid-rv64 6.1.0-rc2+ #1 SMP Thu Nov 16 12:27:06 UTC 2023 riscv64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Aug 27 21:39:55 UTC 2022 on hvc0
root@sid-rv64:~# ifconfig eth0
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.50  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::c086:f0ff:fe24:4714  prefixlen 64  scopeid 0x20<link>
        ether c2:86:f0:24:47:14  txqueuelen 1000  (Ethernet)
        RX packets 4  bytes 366 (366.0 B)
        RX errors 0  dropped 4  overruns 0  frame 0
        TX packets 79  bytes 3694 (3.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 1

root@sid-rv64:~# ping 192.168.1.100
PING 192.168.1.100 (192.168.1.100) 56(84) bytes of data.
From 192.168.1.50 icmp_seq=1 Destination Host Unreachable
From 192.168.1.50 icmp_seq=2 Destination Host Unreachable
From 192.168.1.50 icmp_seq=3 Destination Host Unreachable
From 192.168.1.50 icmp_seq=6 Destination Host Unreachable
From 192.168.1.50 icmp_seq=7 Destination Host Unreachable
From 192.168.1.50 icmp_seq=8 Destination Host Unreachable
From 192.168.1.50 icmp_seq=9 Destination Host Unreachable
From 192.168.1.50 icmp_seq=10 Destination Host Unreachable
From 192.168.1.50 icmp_seq=11 Destination Host Unreachable
^C
--- 192.168.1.100 ping statistics ---
11 packets transmitted, 0 received, +9 errors, 100% packet loss, time 10139ms
pipe 4
root@sid-rv64:~# ifconfig eth0
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.50  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::c086:f0ff:fe24:4714  prefixlen 64  scopeid 0x20<link>
        ether c2:86:f0:24:47:14  txqueuelen 1000  (Ethernet)
        RX packets 4  bytes 366 (366.0 B)
        RX errors 0  dropped 4  overruns 0  frame 0
        TX packets 147  bytes 6578 (6.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 1

root@sid-rv64:~# echo Just pinged to the naxriscv
Just pinged to the naxriscv
root@sid-rv64:~# ifconfig eth0
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.50  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::c086:f0ff:fe24:4714  prefixlen 64  scopeid 0x20<link>
        ether c2:86:f0:24:47:14  txqueuelen 1000  (Ethernet)
        RX packets 13  bytes 906 (906.0 B)
        RX errors 0  dropped 13  overruns 0  frame 0
        TX packets 207  bytes 9126 (8.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 1
```

Dolu1990, 2023-11-24:
> 75MHz 

Ahh maybe on my devkit i more lucky, (vivado give me slack violation of around -0.6 ns)

One thing you can do to increase fmax is to set alu-count=1,decode-count=1 instead

The thing which is botthering with vivado, is that -0.6 ns violation mean it could run at  94 Mhz. But when i change setting to 90 Mhz, then vivado relax too much and still get timing violation XD
maybe the best would be to fake a higher clock frequancy to vivado  than what it realy is to stress him well

Ahhh one thing to diagnostic is : 
cat /proc/interrupts

To see if interrupts happen or not.

newinnovations, 2023-11-24:
Interrupts info:

```
root@sid-rv64:~# cat /proc/interrupts
           CPU0       CPU1
  1:        154          0  SiFive PLIC   3 Edge      eth0
  2:      10813          0  SiFive PLIC   4 Edge      litex-mmc
  5:      51092      51065  RISC-V INTC   5 Edge      riscv-timer
IPI0:        29         56  Rescheduling interrupts
IPI1:      2238       6428  Function call interrupts
IPI2:         0          0  CPU stop interrupts
IPI3:         0          0  IRQ work interrupts
IPI4:         0          0  Timer broadcast interrupts
root@sid-rv64:~# cat /proc/interrupts
           CPU0       CPU1
  1:        277          0  SiFive PLIC   3 Edge      eth0
  2:      10815          0  SiFive PLIC   4 Edge      litex-mmc
  5:      71969      71942  RISC-V INTC   5 Edge      riscv-timer
IPI0:        31         56  Rescheduling interrupts
IPI1:      2239       6505  Function call interrupts
IPI2:         0          0  CPU stop interrupts
IPI3:         0          0  IRQ work interrupts
IPI4:         0          0  Timer broadcast interrupts
```

I was wondering if you can reproduce with the files I provided.

Dolu1990, 2023-11-25:
hmm doesn't look bad.
I will give a try.

Dolu1990, 2023-11-27:
Note if you try to revert litex to https://github.com/Dolu1990/litex.git 5e482d64
i had to change litex-boards/litex_boards/targets/digilent_nexys_video.py
pll.create_clkout(self.cd_sys,       sys_clk_freq, reset_buf="bufg")
into 
pll.create_clkout(self.cd_sys,       sys_clk_freq)

Dolu1990, 2023-12-11:
Should have been fixed with https://github.com/enjoy-digital/litex/issues/1831#issuecomment-1847283321 ?

newinnovations, 2023-12-11:
Last week I was able to get networking running with reverting to the commit you mentioned and modifying the board file. Will try the "official fix" this week.

Dolu1990, 2023-12-11:
ok, thanks :D

Dolu1990, 2023-12-15:
Hi,

I had time to test. With litex upstream all seems good now.
Also, it seems it may have fixed the random freeze i had in litex bios / early linux boot. (@100 Mhz)

let's me know how things goes on your side :D

newinnovations, 2023-12-15:
Working for me as well. But only tested at 75MHz.

Will test for 100Mhz later and report back. Thanks again.

newinnovations, 2023-12-15:
With 100Mhz, indeed the memory hangs seem to be resolved. However, the lite-x bios fails to recognize my SD-card in non-SPI mode or has read failures, so not usable for me at 100MHz.

newinnovations, 2023-12-15:
Some are. I have several paths with negative slack, all related to `crg_s7mmcm0_clkout0`

```
Clock                          WNS(ns)      TNS(ns)  TNS Failing Endpoints  TNS Total Endpoints      WHS(ns)      THS(ns)  THS Failing Endpoints  THS Total Endpoints     WPWS(ns)     TPWS(ns)  TPWS Failing Endpoints  TPWS Total Endpoints
-----                          -------      -------  ---------------------  -------------------      -------      -------  ---------------------  -------------------     --------     --------  ----------------------  --------------------
clk100                           8.006        0.000                      0                    7        0.154        0.000                      0                    7        3.000        0.000                       0                    10
  basesoc_s7mmcm0_mmcm_fb                                                                                                                                                    8.751        0.000                       0                     2
  basesoc_s7mmcm1_mmcm_fb                                                                                                                                                    8.751        0.000                       0                     2
  crg_s7mmcm0_clkout0           -0.808     -751.404                   3788               198206        0.015        0.000                      0               198206        3.750        0.000                       0                 57657
  crg_s7mmcm0_clkout1                                                                                                                                                        0.345        0.000                       0                    77
  crg_s7mmcm0_clkout2                                                                                                                                                        0.345        0.000                       0                     4
  crg_s7mmcm0_clkout3            1.215        0.000                      0                   14        0.160        0.000                      0                   14        0.264        0.000                       0                    11
  crg_s7mmcm1_clkout0            1.200        0.000                      0                  489        0.061        0.000                      0                  489       12.000        0.000                       0                   444
  crg_s7mmcm1_clkout1                                                                                                                                                        2.845        0.000                       0                     8
eth_clocks_rx                                                                                                                                                                5.845        0.000                       0                     1
eth_rx_clk                       1.241        0.000                      0                  445        0.090        0.000                      0                  445        2.000        0.000                       0                   158
  basesoc_pll_fb                                                                                                                                                             6.751        0.000                       0                     2
  ethphy_clkout0                                                                                                                                                             5.845        0.000                       0                     2
  ethphy_clkout1                                                                                                                                                             5.845        0.000                       0                     3
eth_tx_clk                       0.746        0.000                      0                  336        0.153        0.000                      0                  336        3.500        0.000                       0                   140
```

This one seems RAM related for example:

```
--------------------------------------------------------------------------------------
Slack (VIOLATED) :        -0.808ns  (required time - arrival time)
  Source:                 NaxRiscvLitex_a79cc6017676fe6fb88c46e096a4295b/naxes_0_thread_core/DataCachePlugin_logic_cache/load_pipeline_stages_2_WAYS_HITS_reg[3]/C
                            (rising edge-triggered cell FDRE clocked by crg_s7mmcm0_clkout0  {rise@0.000ns fall@5.000ns period=10.000ns})
  Destination:            NaxRiscvLitex_a79cc6017676fe6fb88c46e096a4295b/naxes_0_thread_core/ALU0_ExecutionUnitBase_pipeline_fetch_1_SrcStageables_SRC1_reg[38]/D
                            (rising edge-triggered cell FDRE clocked by crg_s7mmcm0_clkout0  {rise@0.000ns fall@5.000ns period=10.000ns})
  Path Group:             crg_s7mmcm0_clkout0
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            10.000ns  (crg_s7mmcm0_clkout0 rise@10.000ns - crg_s7mmcm0_clkout0 rise@0.000ns)
  Data Path Delay:        10.512ns  (logic 1.784ns (16.972%)  route 8.728ns (83.028%))
  Logic Levels:           10  (LUT3=1 LUT4=3 LUT5=5 LUT6=1)
  Clock Path Skew:        -0.304ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    5.994ns = ( 15.994 - 10.000 )
    Source Clock Delay      (SCD):    6.543ns
[... cut ...]
    SLICE_X65Y127        FDRE (Setup_fdre_C_D)        0.075    16.247    NaxRiscvLitex_a79cc6017676fe6fb88c46e096a4295b/naxes_0_thread_core/ALU0_ExecutionUnitBase_pipeline_fetch_1_SrcStageables_SRC1_reg[38]
  -------------------------------------------------------------------
                         required time                         16.247
                         arrival time                         -17.055
  -------------------------------------------------------------------
                         slack                                 -0.808
```