SpinalHDL/NaxRiscv #70: RVLS and testsGen.py

Bill94l, 2023-11-30:
Hello, 

1) What is the difference between checking with these two methods which uses spike as a reference model?
- testsGen.py: which generates a pass or fail for each test (as described in src/test/cpp/naxriscv/README.md)
- SocSim.scala: which uses RVLS to check CPU execution traces

2) How can we make all the existing tests to run through SocSim.scala, as is done in the following step explained in the README?
```shell
cd $NAXRISCV/src/test/cpp/naxriscv
./testsGen.py
make clean compile
make test-clean && make test-all -j$(nproc); make test-report
```

3) Is the verilator & spike check done simultaneously at each execution step when compiling ext/rvls? or it's equivalent. If ext/rvls is not compiled and we use --no-rvls argument, we will have an output trace in (simWorkspace/SoCDemo/test/tracer.log) We then use this trace as input to RVLS for verification of the execution steps. 
Are these steps equivalent to lock-step simulation?

Dolu1990, 2023-11-30:
Hi,

> What is the difference between checking with these two methods which uses spike as a reference model?

testsGen.py is based on a  C++ testbench which directly use verilator to run the sim. That testbench also manualy use spike as a golden model. It run all the regression from the ext/NaxriscvSoftware repo.
So testGen.py is used to run automated regressions.

SocSim.scala is a very different testbench. It is based on Scala / SpinalSim, but still use verilator indirectly to run the sim (but in a scala runtime, not c++). That testbench has the capability to run multicore / memory coherent NaxRiscv while using RVLS as golden model to check that the cores are doing things right. It can even boot linux in multi core while keeping RVLS in sync.
So SocSim.scala was used to verify memory coherency / multi-core / SoC level stuff, mostly by booting linux on it, which take maybe around 30 mn. 
RVLS is using spike under the hood

> How can we make all the existing tests to run through SocSim.scala, as is done in the following step explained in the README?

Currently, there is nothing which automate it. You would have to run each test manualy. So far it wasn't realy the purpose of that testbench.

> Is the verilator & spike check done simultaneously at each execution step

Yes

> compiling ext/rvls

Is only used by SocSim.scala as golden model to check the CPU

> If ext/rvls is not compiled and we use --no-rvls argument, we will have an output trace in

If RVLS isn't complied, the sim will run, but no check will be done, so the CPU can go crazy without any error report.

> simWorkspace/SoCDemo/test/tracer.log

So, there is mostly 2 way to use the SocSim.scala checking.
Either you let it use rvls by itself, and SocSim.scala will by itself do some calls to RVLS to check on each commit / trap that things are in sync
EIther you instead let SocSim.scala generate simWorkspace/SoCDemo/test/tracer.log, and manualy in a terminal ask rvls to check the tracer.log.

The good thing about tracer.log, is that it allow to do things offline and to inspect what happened manualy. But the issue is that the check isn't done during sim, and that the file can grow quite big quite fast XD

Bill94l, 2023-12-01:
Dear Charles, 

Thank you for your quick reply.

My ultimate goal is to build a lockstep verification environment based on NaxRiscv & RVLS into which I can integrate the RISCV-DV framework (). I'd like to know what you think is the best way to do this.

1) How can I automate the NaxRiscv kernel check to run all the regression tests used in testsGen.py (basic tests + bootLinux) with RVLS (via SocSim.scala)? 

2) Could you tell me more about the functions (doTest & doCmd) and the command line that allows running src/test/scala/naxriscv/NaxRiscvRegression.scala to be launched?

3)  I see that (NaxRiscvRegression.scala ) automates the running of all regression tests. I would like to confirm that the following lines in NaxRiscvRegression.scala (L95 - L99) are being used for the checking of the output test results?
```scala
      val passed = doCmd(s"find output -name PASS", env :_*).lines.count()
      val failed = doCmd(s"find output -name FAIL", env :_*).lines.count()
      println(s"PASS = $passed")
      println(s"FAIL = $failed")
      if(failed != 0 || passed == 0) throw new Exception("Failure")
```

4) To automate regression testing with RVLS using SocSim.scala, would you advise me to take NaxRiscvRegression.scala and modify it?

5) Then I would like to analyse the RVLS output file (spike.log) to check for execution errors in the execution, but I don't know if this is provided for in RVLS (src/ascii_frontend.cpp/checkFile)? Secondly, I still don't know where to find the definition of various error or success codes that appear in the spike.log (like number 3 in the log below)?
```log
core   0: 0x80000000 (0x00001fb7) lui     t6, 0x1
core   0: 3 0x80000000 (0x00001fb7) x31 0x00001000

core   0: 0x80000004 (0x388f8f93) addi    t6, t6, 904
core   0: 3 0x80000004 (0x388f8f93) x31 0x00001388

core   0: 0x80000008 (0x00001f17) auipc   t5, 0x1
core   0: 3 0x80000008 (0x00001f17) x30 0x80001008

core   0: 0x8000000c (0xff8f0f13) addi    t5, t5, -8
core   0: 3 0x8000000c (0xff8f0f13) x30 0x80001000

core   0: 0x80000010 (0x00012eb7) lui     t4, 0x12
core   0: 3 0x80000010 (0x00012eb7) x29 0x00012000
```

Dolu1990, 2023-12-04:
> RISCV-DV framework

Ahhh i don't realy know what is the testing interface of RISC-V DV.

> NaxRiscv kernel check

What do you mean by it ? (i'm not sure ^^)

> How can I automate the NaxRiscv kernel check to run all the regression tests used in testsGen.py (basic tests + bootLinux) with RVLS (via SocSim.scala)?

in a first time, you can do some : 
sbt "runMain naxriscv.platform.tilelinkdemo.SocSim --load-elf ext/NaxSoftware/baremetal/dhrystone/build/rv32ima/dhrystone.elf"
sbt "runMain naxriscv.platform.tilelinkdemo.SocSim --load-elf ..."

for each tests you want to run.
That will be slow, because it will recompile the NaxRiscv each time between each sim XD

In VexiiRiscv (very fresh stuff i'm working on), i did something a bit different by added a simulation server : 
https://github.com/SpinalHDL/VexiiRiscv/blob/623c6fdb76c83a4e1c8980501a2ead157277c5c1/src/test/scala/vexiiriscv/tester/TestBench.scala#L259

Basicaly, you run it, then it stay "warm" and listen to a TCP socket for simulation requests. for instance you can run a simulation using : 
echo '--load-elf ext/NaxSoftware/baremetal/dhrystone/build/rv32ima/dhrystone.elf   --with-all' | nc localhost  8189

(I used it to integrate with embench test suite)

> Could you tell me more about the functions (doTest & doCmd) and the command line that allows running src/test/scala/naxriscv/NaxRiscvRegression.scala to be launched?

This stuffed is all based on the verilator testbench, not the SocSim.
- doCmd run a shell command from scala while redirecting stdout
- doTest run the generation / compilation / verilator testbench on a naxriscv config

"sbt test" i guess should run all the scala regressions, including that one.
but to specifcaly run it : 
sbt "testOnly *.NaxRiscvRegression"

> lines in NaxRiscvRegression.scala (L95 - L99) are being used for the checking of the output test results?

Yes

> To automate regression testing with RVLS using SocSim.scala, would you advise me to take NaxRiscvRegression.scala and modify it?

Hmm naaa, not realy, those are completly different things.

> Then I would like to analyse the RVLS output file (spike.log) to check for execution errors in the execution, but I don't know if this is provided for in RVLS 

Yes it is, you have to enable the debug mode, see       if(withRvls) rvls.debug() in the SocSim

> src/ascii_frontend.cpp/checkFile

That's something else, that's when you want to check a trace file which is in the rvls format (not spke)

> Secondly, I still don't know where to find the definition of various error or success codes that appear in the spike.log (like number 3 in the log below)?

That's spike stuff, i think the number 3 there is the privilege mode in which the instruction was executed
3 => machne
1 => supervisor
0 => user

So overall, i have a few questions : 
- in which context are doing this work ? (study / work / pleasure)
- you try overall to get riscv-dv to run SocSim with rvls, so, how riscv-dv expect people to connect things ? Is it like "Run this binary on your CPU, collect the trace, check those against my reference trace" ?

Bill94l, 2024-01-18:
> in which context are doing this work ? (study / work / pleasure).

I'm working with Olivier on integrating side-channel countermeasures into NaxRiscv64core.

> Hmm naaa, not realy, those are completly different things.

I am currently working to launch the execution of all the tests used in NaxRiscvRegression with RVLS in order to have a lock-step verification. I made modifications in the following files :
- SocSim.scala
- SocDemo.scala
- DualSimTracer.scala
- TilelinkNaxRiscvFiber.scala
- Makefile
in order to generate a NaxRiscv core once for each configuration and to use it to launch a list of ELF executables that are adapted to the core configuration.

Now I just have to write a Scala file "lockstep_regression" equivalent to "NaxRiscvRegression" and a Python file to generate a Makefile according to the desired configuration (equivalent to testGen.py).

> you try overall to get riscv-dv to run SocSim with rvls, so, how riscv-dv expect people to connect things ? Is it like "Run this binary on your CPU, collect the trace, check those against my reference trace" ?

I will start working on the integration of RISCV-DV as soon as I have finished launching the tests used by NaxRiscvRegression in lockstep with RVLS.

Dolu1990, 2024-01-18:
Cool thanks :D

Bill94l, 2024-01-25:
Hello Charles,

I am trying to run the riscv test programs with the RVLS through SocSim. The basic tests pass well (riscv-tests, riscvTestMemory, riscvTestMul, riscvTestDiv, riscvTestAmo). Even the "riscv32TestRvc" tests pass.

However, the tests fail when I run the riscv32TestFloat and riscv64TestDouble tests. In the <reset_vector>  PMP ( (csrw pmpaddr0,t0)) throws an exception in spike.log (the output of RVLS).  Knowing that I have added the options "F" / "D" / "C" in src/main/scala/naxriscv/platform/NaxriscvProbe.scala 
> **line 154:** tracer.newCpu(hartId, s"RV${xlen } IMAFDC", "MSU", 32, HartId)

**For example :**
> sbt "runMain naxriscv.platform.tilelinkdemo.SocSim --xlen 32 --nax-count 1 --withFloat --dual-sim --start-symbol _start --pass-symbol pass --start-add 0 --load- elf NaxRiscv/ext/NaxSoftware/riscv-tests/rv32uf-p-fmadd"

- Terminal output:
[info] *** CSR WRITE ADDRESS DUT=3 REF=2 ***
[info] commit error
[info] - std::exception
[info] *** CSR WRITE ADDRESS DUT=3 REF=2 ***
[info] commit error
[info] - std::exception

- Spike.log:
core   0: 3 0x800000f0 (0xfff28293) x 5 0x7fffffff
core   0: 0x800000f4 (0x3b029073) csrw    pmpaddr0, t0
core   0: exception trap_illegal_instruction, epc 0x800000f4
core   0:           tval 0x3b029073
core   0: 0x80000100 (0x30405073) csrwi   mie, 0
core   0: 3 0x80000100 (0x30405073) c772_mie 0x00000000

Do you have any suggestions for a solution to the problem?

Thank you

closed 2024-06-22 by Bill94l