SpinalHDL/NaxRiscv #71: Request for Guidance: Starting with a Minimal Framework

skylayer, 2023-12-27:
Hi,

I've been working with SpinalHDL for a few months and am inspired by the design of NaxRiscv. I aim to deepen my understanding by building a toy CPU from scratch. While I appreciate the complexity and sophistication of NaxRiscv, I'm seeking an empty or minimal framework as a starting point for my project. Could you provide or direct me to such a simplified version of your framework? This would greatly aid in learning and applying your design methodology.

Dolu1990, 2023-12-29:
Hi,

So i kinda had to recently make such framework because i wanted to start working on VexiiRiscv, which kinda adopt / improve many aspect of NaxRiscv : 
https://github.com/SpinalHDL/VexiiRiscv

What aspect of the framework are you insterrested in ?
- Plugin
- Pipelining API
- ... 

Like something with the plugin + one main pipeline ?

Dolu1990, 2024-01-11:
Hi,

Yes, the things merged into SpinalHDL 1.10.0 should be enough.
I would say, just copy past the https://github.com/SpinalHDL/VexiiRiscv/blob/dev/src/main/scala/spinal/lib/misc/database/DataBase.scala into your project if you want to use the database stuff ^^

Also, here are the things i did (in order) for VexiiRiscv : 
https://github.com/SpinalHDL/VexiiRiscv/issues/1

In general, for cpu design, you need to go little step by little step, it realy help.

MahirAbbas, 2024-04-19:
@skylayer here are some textbooks to help (in order of complexity, from beginner to advanced)
Digital Design and Computer Architecture RISC-V Edition, Sarah Harris, David Harris
Computer Organization and Design RISC-V Edition The Hardware Software Interface (David A. Patterson, John L. Hennessy)
Next-Gen Computer Architecture: Till the End of Silicon, Smruti R. Sarangi
Computer Architecture: A Quantitative Approach, Sixth Edition, John L. Hennessy & David A. Patterson

hasaranga, 2024-12-30:
@Dolu1990 Can you provide a simple example of CPU using the current SpinalHDL features?

Dolu1990, 2025-01-02:
@hasaranga what do you mean by simple example of CPU ?
There is VexiiRiscv : https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/Framework/index.html
But it is a big design

hasaranga, 2025-01-02:
> @hasaranga what do you mean by simple example of CPU ?
> 
> There is VexiiRiscv : https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/Framework/index.html
> 
> But it is a big design

Just a few instructions. I wanted to see how to begin with new APIs.

Dolu1990, 2025-01-02:
Ahhh then https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/Framework/index.html should help.
There is also the pipeline API : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Libraries/Pipeline/introduction.html

hasaranga, 2025-01-02:
> Ahhh then https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/Framework/index.html should help.
> 
> There is also the pipeline API : 
> 
> https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Libraries/Pipeline/introduction.html

I just wanted to see how those new  APIs fit together! Just a skeleton of cpu!

Dolu1990, 2025-01-02:
Ahhh there is nothing like this in the doc.
mostly a few decoupled samples.