SpinalHDL/NaxRiscv #72: NaxRiscv running custom test cases prompts FAILURE ???

LMiaoH, 2024-01-03:
Hi，
When running my custom-designed test cases with NaxRiscv, I encountered a failure and received the following information:
```
DUT is blocked in a endless trap cycle of death
TIME=456
LAST PC COMMIT=80000440
INCOMING SPIKE PC=0
ROB_ID=x0
FAILURE ???
STATS :
  IPC               0.0044843
  cycles            223
  commits           1
  reschedules       0
  trap              0
  branch miss       0
  jump miss         0
  storeToLoadHazard 0
  loadHitMiss       0

```
But I am unsure about the significance of this information or how to modify my input. Can you provide guidance on this issue?

Other pieces of information：
1. run test case：
```
SEED=input_nax.elf
STRAT_SYM=_inst_main
PASS_SYM=write_tohost

TIMEOUT=10000

./obj_dir/VNaxRiscv --load-elf $SEED --start-symbol $STRAT_SYM --pass-symbol $PASS_SYM  --timeout $TIMEOUT --stats-print 
```
2. Compilation parameters for the test case file：
```
riscv64-unknown-elf-gcc -march=rv64gc -mabi=lp64d -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I include/ -T include/link.ld -I include/p input_nax.S -o input_nax.elf
```

Dolu1990, 2024-01-03:
Hi,

The error message seems to indicate that Spike (golden model) want to throw an RISC-V exception.

What software are you running (exactly / .c / .asm) ?

LMiaoH, 2024-01-03:
I want to execute the .asm file.

Additionally, I attempted to use `--spike-disabled` to disable the lockstep check performed on the testbench, but I still encountered some issues:
```
NO PROGRESS the cpu hasn't commited anything since too long
TIME=20394
LAST PC COMMIT=80000080
INCOMING SPIKE PC=80000000
ROB_ID=x22
FAILURE ???
```
Even though I followed the instructions provided in [https://github.com/SpinalHDL/NaxRiscv/blob/main/src/test/cpp/naxriscv/README.md](url).
Due to my limited experience with NaxRiscv, I would appreciate some assistance

Dolu1990, 2024-01-03:
Did you used the Gen64  to generate naxriscv ?
Also note that RVC is disabled by default : 
https://github.com/SpinalHDL/NaxRiscv/blob/6af1a20158f6402196c1f17f47680b96e0a88b86/src/main/scala/naxriscv/Gen.scala#L532
RVC is one of the issue you likely have.

LMiaoH, 2024-01-03:
Thank you, I think I might not have done that.
But I am not sure how to use Gen64 to generate naxriscv. May it be as expressed in `src/test/cpp/naxriscv/README.md`?

Dolu1990, 2024-01-03:
just remplace Gen by Gen64 in the command line ^^

LMiaoH, 2024-01-03:
Thank you, your approach is correct,

but there is still an issue: encountering a floating-point register or fcsr register triggers a 'trap_illegal_instruction' message."

```
core   0: 0x00000000800003d2 (0x00305073) csrwi   fcsr, 0
core   0: exception trap_illegal_instruction, epc 0x00000000800003d2
core   0:           tval 0x0000000000305073
```
or 
```
core   0: 0x000000008000040a (0x204b06d3) fsgnj.s fa3, fs6, ft4
core   0: exception trap_illegal_instruction, epc 0x000000008000040a
core   0:           tval 0x00000000204b06d3
```
Can you offer me some advice?

Dolu1990, 2024-01-03:
Yes, in the mstatus csr, there is the FS field, it should be set to somthing else than 0b00

LMiaoH, 2024-01-08:
Unfortunately, the FS field in the mstatus CSR is not 0b00, and it seems that the issue still exists. Additionally, similar problems may occur with registers related to pmpaddr. 
```
core   0: 0x00000000800003b8 (0x3b029073) csrw    pmpaddr0, t0
core   0: exception trap_illegal_instruction, epc 0x00000000800003b8
core   0:           tval 0x000000003b029073
```
However, it's noteworthy that the pmpaddr register doesn't interrupt the execution of my test cases, whereas issues with floating-point registers or the fcsr register directly lead to the termination of my program.

LMiaoH, 2024-01-17:
Hello, I'm trying to inspect the values of certain memory addresses in `soc->memory` within the main.cpp file. However, the values obtained through the `get` function always return as 0. Could you help me take a look at what might be causing this issue?
```
u64 socTextReadAddress = elf->getSymbolAddress("test_address");
u8* socMemoryData = soc->memory.get(socTextReadAddress);
printf("socTextReadAddress: %x, socMemoryData:%02x\n",socTextReadAddress, *socMemoryData);
```
socTextReadAddress: 80002038, socMemoryData:00
Memory value at the same address in Spike: 38

Dolu1990, 2024-01-17:
Hi,

That's curious, i would say it should have worked.
At that stage, maybe adding some printf in the memory.h file to figure where / how accesses are done could help.