SpinalHDL/NaxRiscv #73: Loop when change GShare RAM

zyn810039594, 2024-01-08:
Hello!

By default, gshare have a 4096×8 ram for gshare counter. However, the shape of the 4096×8 ram is too long to run at a high period. So I want to change the 4096×8 ram to a 512×64 ram. My solution is to write data by 8-bits mask, and read data by judging the last 3-bits addresses. But spinal calls loop...
I've only changed the `GSharePlugin.scala`, and here's the code I changed:

```scala
class GSharePlugin(var historyWidth : Int,
                   var entries : Int = 0,
                   var memBytes : BigInt = null,
                   var readAt : Int = 0,
                   var counterWidth : Int = 2,
                   var readAsync : Boolean = false) extends Plugin with FetchConditionalPrediction with HistoryUser{

  override def useHistoryAt = readAt
  override def historyWidthUsed = historyWidth
  override def getPredictionAt(stageId: Int) = getService[FetchPlugin].getStage(stageId)(setup.keys.GSHARE_COUNTER).map(_.msb)

  val setup = create early new Area{
    val fetch = getService[FetchPlugin]
    val frontend = getService[FrontendPlugin]
    val branchContext = getService[BranchContextPlugin]
    val aligner = getService[AlignerPlugin]

    fetch.retain()
    frontend.retain()
    branchContext.retain()

    val keys = new AreaRoot {
      val GSHARE_COUNTER = Stageable(Vec.fill(SLICE_COUNT)(UInt(counterWidth bits)))
    }

    aligner.addLastWordContext(
      keys.GSHARE_COUNTER
    )
    branchContext.dispatchWrite(
      keys.GSHARE_COUNTER
    )
  }

  val logic = create late new Area{
    val fetch = getService[FetchPlugin]
    val branchContext = getService[BranchContextPlugin]
    val predictor = getService[DecoderPredictionPlugin]
    val frontend = getService[FrontendPlugin]
    val BRANCH_HISTORY = getService[HistoryPlugin].keys.BRANCH_HISTORY

    var words = entries/SLICE_COUNT
    if(memBytes != null) words = (1 <<(log2Up(memBytes.toInt*8/counterWidth+1)-1)) / SLICE_COUNT
    assert(words != 0)
    assert(isPow2(words))

    val keys = setup.keys

    def hashWidth = log2Up(words)
    def gshareHash(address : UInt, history : Bits) = address(SLICE_RANGE.high + 1, hashWidth bits).reversed ^ U(history).resized

    val mem = new Area{ //TODO bypass read durring write ?
      // ----------------------------------------------------------------
      // Where I changed
      // ----------------------------------------------------------------
      val counter = Mem.fill(words/8)(Vec(keys.GSHARE_COUNTER,8))
      val write = counter.writePortWithMask(8)
      if(GenerationFlags.simulation){
        counter.initBigInt(List.fill(counter.wordCount)(BigInt(0)))
      }
    }

    val BYPASS = Stageable(cloneOf(mem.write))
    val HASH = Stageable(UInt(hashWidth bits))

[... 156 lines cut ...]
    >>> (toplevel/FetchPlugin_stages_1_GSHARE_COUNTER_0 :  UInt[2 bits]) at naxriscv.prediction.GSharePlugin$$anonfun$3$$anon$2$$anon$1$$anonfun$4$$anonfun$apply$1.apply(GSharePlugin.scala:42) >>>

  Full chain :
    (toplevel/BtbPlugin_logic_applyIt_prediction :  Bool)
    (toplevel/BtbPlugin_setup_historyPush_taken :  Bits[1 bits])
    Bits(Int)
    Bits -> Bits
    Bits ## Bits
    resize(Bits,24 bits)
    (toplevel/HistoryPlugin_logic_update_pushes_0_stateNext_1 :  Bits[24 bits])
    (toplevel/HistoryPlugin_logic_onFetch_valueNext :  Bits[24 bits])
    (toplevel/FetchPlugin_stages_0_BRANCH_HISTORY :  Bits[24 bits])
    (Bits -> UInt of 24 bits)
    resize(UInt,5 bits)
    (UInt ^ UInt)[5 bits]
    (toplevel/FetchPlugin_stages_0_GSharePlugin_logic_HASH :  UInt[5 bits])
    UInt(Int downto Int)
    (toplevel/??? :  UInt[3 bits])
    mux of UInt[2 bits]
    (toplevel/FetchPlugin_stages_1_GSHARE_COUNTER_0 :  UInt[2 bits])
    UInt(Int)
    mux of Bool
```

That's all. If any other details are needed, I can provide them. Thanks for your help!

Dolu1990, 2024-01-11:
Hi,

Here is the fix : 

```scala
      def readMem[T <: Data](mem : Mem[Vec[T]]) = {
        assert(!readAsync)
        val addrReal = readCmd.stage(HASH)(widthOf(HASH) - 1 downto 3)
        val addrChoose = this.stage(HASH)(2 downto 0)
        readAsync match {
          case false => mem.readSync(addrReal, readCmd.stage.isReady)(addrChoose)
          case true  => mem.readAsync(addrReal)(addrChoose)
        }
      }
```
The issue was that the code was using the address of stage 0 to mux the output of the readSync, so jumping stage -> creating combinatorial loop with history/pc updates

zyn810039594, 2024-01-14:
Thanks for help! Problem resolved.

closed 2024-01-14 by zyn810039594