SpinalHDL/NaxRiscv #83: riscv tests with virtual memory enabled fail

Bill94l, 2024-02-23:
Hi Charles,

I generated tests with virtual memory enabled from the riscv-tests.

When I start the tests from the "test_3" symbol, the execution finishes successfully, otherwise the execution fails when I start the tests from the "_start" symbol. I would like to know the reason for the failure of the test ?
#### start-symbol : test_3
 ```Bash
./obj_dir/VNaxRiscv  --load-elf /tests/riscv-tests/isa/rv64ui-v-addi --start-symbol test_3  --pass-symbol pass --fail-symbol fail --trace-ref  --output-dir output
SUCCESS ???
```
#### start-symbol : _start
 ```Bash
./obj_dir/VNaxRiscv --load-elf  /tests/riscv-tests/isa/rv64ui-v-addi --start-symbol test_3  --pass-symbol pass --fail-symbol fail --trace-ref  --output-dir output --spike-debug

*** MISSMATCH PC DUT=80000008 REF=2bc0 ***

TIME=3032
LAST PC COMMIT=80000008
INCOMING SPIKE PC=ffffffffffe00154
ROB_ID=x12
FAILURE ???
```
Please find attached the "spike.log" of the previous command with the start symbol = _start
[spike.log](https://github.com/SpinalHDL/NaxRiscv/files/14387127/spike.log)
[rv64ui-v-addi.dump](https://github.com/SpinalHDL/NaxRiscv/files/14387163/rv64ui-v-addi.dump.log)

--------------------------------------------------------------------------

#### SocSim without RVLS & start-symbol : _start 
When launching the code with SocSim the simulation remains blocked at the beginning
 ```Bash
sbt "runMain naxriscv.platform.tilelinkdemo.SocSim --xlen 64 --nax-count 1 --trace --no-rvls --load-elf /tests/riscv-tests/isa/rv64ui-v-addi"

[info] [Progress] Start SocDemo rv64ui-v-addi simulation with seed 2
[info] Sim starting <3
```

Please find attached the "tracer.log" from SocSim
[tracer.log](https://github.com/SpinalHDL/NaxRiscv/files/14387243/tracer.log)

Thank you !

Dolu1990, 2024-02-26:
Hi ^^

> I generated tests with virtual memory enabled from the riscv-tests.

Ahhh i never tried those. I didn't knew it even existed. How does this works ? it is all the regular test but running in virtual memory ?

> *** MISSMATCH PC DUT=80000008 REF=2bc0 ***

So looking at spike.log, i can see 80000008 refer to the mtvec, so NaxRiscv did trap to machine mode, which seems to be correct, as spike log say : 
core   0: exception trap_instruction_page_fault

The thing i don't get is why  REF=2bc0, so why the spike comparison say spike commited  PC 2bc0

Would probably need to go step by step with gdb / eclipse in the testbench/spike code to figure it out

Bill94l, 2024-02-26:
Hi,

>  How does this works ? it is all the regular test but running in virtual memory ?

This is a regular test but executed with virtual memory enabled and program execution failed in virtual memory initialization

Frome the [README of riscv-tests](https://github.com/riscv-software-src/riscv-tests/blob/master/README.md) 
<p dir="auto">A test program for RISC-V is written within a single assembly language file,
which is passed through the C preprocessor, and all regular assembly
directives can be used. An example test program is shown below. Each test
program should first include the <code>riscv_test.h</code> header file, which defines the
macros used by the TVM. The header file will have different contents depending
on the target environment for which the test will be built.  One of the goals
of the various TVMs is to allow the same test program to be compiled and run
on very different target environments yet still produce the same results. The
following table shows the target environment currently defined.</p>

Target Environment Name | Description
-- | --
p | virtual memory is disabled, only core 0 boots up
pm | virtual memory is disabled, all cores boot up
pt | virtual memory is disabled, timer interrupt fires every 100 cycles
v | virtual memory is enabled

Here is the execution result when launching the code with SocSim and RVLS : 
```Shell
sbt "runMain naxriscv.platform.tilelinkdemo.SocSim --xlen 64 --nax-count 1 --dual-sim --start-symbol _start --pass-symbol pass --start-add 0 --load-elf  /tests/riscv-tests/isa/rv64ui-v-addi"

[info] [Progress] Start SocDemo rv64ui-v-addi_explorer simulation with seed 2
[info] PeripheralEmulator IS created
[info] Sim starting <3
[info] PC COMMIT dut=80000000 ref=80000000
[info] PC COMMIT dut=8000000c ref=8000000c
[info] PC COMMIT dut=80000010 ref=80000010
[info] PC COMMIT dut=80000014 ref=80000014
...
...
...
[info] PC COMMIT dut=800000e8 ref=800000e8
[info] PC COMMIT dut=800000ec ref=800000ec
[info] PC COMMIT dut=800000f0 ref=800000f0
[info] [Error] Simulation failed at time=22080
[info] null
[info] naxriscv.platform.RvlsBackend.trap(Tracer.scala:197)
[info] naxriscv.platform.NaxriscvProbe$$anonfun$checkTrap$1.apply(NaxriscvProbe.scala:210)
[info] naxriscv.platform.NaxriscvProbe$$anonfun$checkTrap$1.apply(NaxriscvProbe.scala:210)
[info] scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[info] scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[info] naxriscv.platform.NaxriscvProbe.checkTrap(NaxriscvProbe.scala:210)
[info] naxriscv.platform.NaxriscvProbe$$anonfun$1.apply$mcV$sp(NaxriscvProbe.scala:342)
[info] spinal.core.sim.package$SimClockDomainPimper$$anonfun$onSamplings$1$$anonfun$apply$mcV$sp$1.apply(package.scala:969)
[info] spinal.core.sim.package$SimClockDomainPimper$$anonfun$onSamplings$1$$anonfun$apply$mcV$sp$1.apply(package.scala:969)
[info] scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[info] scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[info] spinal.core.sim.package$SimClockDomainPimper$$anonfun$onSamplings$1.apply$mcV$sp(package.scala:969)
[info] spinal.core.sim.package$$anon$1.update(package.scala:194)
[info] spinal.sim.SimManager.runWhile(SimManager.scala:324)
[info] spinal.sim.SimManager.runAll(SimManager.scala:246)
[info] spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:606)
[info] spinal.core.sim.SimCompiled.doSimUntilVoid(SimBootstraps.scala:568)
[info] spinal.lib.misc.test.DualSimTracer$$anonfun$1.apply$mcV$sp(DualSimTracer.scala:24)
[info] spinal.lib.misc.test.AsyncJob$$anonfun$1$$anonfun$apply$mcV$sp$1$$anonfun$apply$mcV$sp$2.apply$mcV$sp(MultithreadedTester.scala:24)
[info] spinal.lib.misc.test.AsyncJob$$anonfun$1$$anonfun$apply$mcV$sp$1$$anonfun$apply$mcV$sp$2.apply(MultithreadedTester.scala:23)
[info] spinal.lib.misc.test.AsyncJob$$anonfun$1$$anonfun$apply$mcV$sp$1$$anonfun$apply$mcV$sp$2.apply(MultithreadedTester.scala:23)
[info] scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[info] scala.Console$.withErr(Console.scala:92)
[info] spinal.lib.misc.test.AsyncJob$$anonfun$1$$anonfun$apply$mcV$sp$1.apply$mcV$sp(MultithreadedTester.scala:22)
[info] spinal.lib.misc.test.AsyncJob$$anonfun$1$$anonfun$apply$mcV$sp$1.apply(MultithreadedTester.scala:22)
[info] spinal.lib.misc.test.AsyncJob$$anonfun$1$$anonfun$apply$mcV$sp$1.apply(MultithreadedTester.scala:22)
[info] scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[... 23 lines cut ...]
[info] PC COMMIT dut=80000120 ref=80000120
[info] PC COMMIT dut=80000124 ref=80000124
[info] PC COMMIT dut=80000128 ref=80000128
[info] PC COMMIT dut=8000012c ref=8000012c
[info] PC COMMIT dut=80000130 ref=80000130
[info] PC COMMIT dut=80000134 ref=80000134
[info] PC COMMIT dut=80000138 ref=80000138
[info] PC COMMIT dut=8000013c ref=8000013c
[info] PC COMMIT dut=80000140 ref=80000140
[info] PC COMMIT dut=80000144 ref=80000144
[info] PC COMMIT dut=80000148 ref=80000148
[info] PC COMMIT dut=8000014c ref=8000014c
[info] DUT did trap on 80000150
[info] Code 2
[info] commit error
[info] - std::exception

[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
``` 
Please find attached the "spike.log" & "tracer.log" of the previous command rv64ui-v-addi with socSim+RVLS
[spike.log](https://github.com/SpinalHDL/NaxRiscv/files/14405389/spike.log)
[tracer.log](https://github.com/SpinalHDL/NaxRiscv/files/14405393/tracer.log)

Thank you !

Dolu1990, 2024-03-04:
At this point would need /tests/riscv-tests/isa/rv64ui-v-addi to recreate the issue to understand what is happening