SpinalHDL/NaxRiscv #88: The mmu_sv39.elf fail with SocSim

Bill94l, 2024-03-22:
Hi Charles,

The mmu_sv39.elf fail with SocSim, here is the command I used 
```Bash
sbt "runMain naxriscv.platform.tilelinkdemo.SocSim --xlen 64 --nax-count 1 --load-elf  ext/NaxSoftware/baremetal/mmu_sv39/build/rv64ima/mmu_sv39.elf --start-symbol _start --pass-symbol pass --trace"
```

According to the execution result, there is a missmatch between the trap code returned by the DUT and REF, here is the execution result which shows that the DUT returns code 13 which is equivalent to LOAD_PAGE_FAULT and the REF returns code 5 which is equivalent to LOAD_ACCESS_FAULT. Do you have any idea about this missmatch?

```Bash
[info] [Error] Simulation failed at time=47650
[error] Exception in thread "main" java.lang.Exception
[error] 	at naxriscv.platform.RvlsBackend.trap(Tracer.scala:197)
[error] 	at naxriscv.platform.NaxriscvProbe$$anonfun$checkTrap$1.apply(NaxriscvProbe.scala:210)
[error] 	at naxriscv.platform.NaxriscvProbe$$anonfun$checkTrap$1.apply(NaxriscvProbe.scala:210)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at naxriscv.platform.NaxriscvProbe.checkTrap(NaxriscvProbe.scala:210)
[error] 	at naxriscv.platform.NaxriscvProbe$$anonfun$1.apply$mcV$sp(NaxriscvProbe.scala:342)
[error] 	at spinal.core.sim.package$SimClockDomainPimper$$anonfun$onSamplings$1$$anonfun$apply$mcV$sp$1.apply(package.scala:970)
[error] 	at spinal.core.sim.package$SimClockDomainPimper$$anonfun$onSamplings$1$$anonfun$apply$mcV$sp$1.apply(package.scala:970)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at spinal.core.sim.package$SimClockDomainPimper$$anonfun$onSamplings$1.apply$mcV$sp(package.scala:970)
[error] 	at spinal.core.sim.package$$anon$1.update(package.scala:195)
[error] 	at spinal.sim.SimManager.runWhile(SimManager.scala:324)
[error] 	at spinal.sim.SimManager.runAll(SimManager.scala:246)
[error] 	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:606)
[error] 	at spinal.core.sim.SimCompiled.doSimUntilVoid(SimBootstraps.scala:568)
[error] 	at naxriscv.platform.tilelinkdemo.SocSim$$anonfun$34.apply(SocSim.scala:198)
[error] 	at naxriscv.platform.tilelinkdemo.SocSim$$anonfun$34.apply(SocSim.scala:193)
[error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
[error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error] 	at naxriscv.platform.tilelinkdemo.SocSim$.delayedEndpoint$naxriscv$platform$tilelinkdemo$SocSim$1(SocSim.scala:193)
[error] 	at naxriscv.platform.tilelinkdemo.SocSim$delayedInit$body.apply(SocSim.scala:55)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:392)
[error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] 	at scala.App$class.main(App.scala:76)
[error] 	at naxriscv.platform.tilelinkdemo.SocSim$.main(SocSim.scala:55)
[error] 	at naxriscv.platform.tilelinkdemo.SocSim.main(SocSim.scala)
[info] PC: 80000588
[info] Code: 15
[info] Interrupt: 0
[info] PC: 800005b8
[info] Code: 15
[info] Interrupt: 0
[info] PC: 800005e8
[info] Code: 15
[info] Interrupt: 0
[info] PC: 1000000000
[info] Code: 12
[info] Interrupt: 0
[info] PC: 2000000000
[info] Code: 12
[info] Interrupt: 0
[info] PC: 4000000000
[info] Code: 12
[info] Interrupt: 0
[info] PC: ffffffe000000000
[info] Code: 12
[info] Interrupt: 0
[info] PC: ffffffc000000000
[info] Code: 12
[info] Interrupt: 0
[info] PC: ffffff8000000000
[info] Code: 12
[info] Interrupt: 0
[info] PC: 80000740
[info] Code: 13
[info] Interrupt: 0
[info] *** DUT code missmatch DUT=d REF=5 ***
[info] commit error
[info] - std::exception
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 16 s, completed 22 mars 2024 11:25:03
```
Thank you !

Dolu1990, 2024-03-22:
Hi, 

I think this should be expected the the SoCSim can't pass mmu_sv39.elf, as the memory layout isn't the same than the pure verilator regression testbench.

Bill94l, 2024-03-22:
Pourquoi ça marche bien sans rvls?
#### command 
```Bash
sbt "runMain naxriscv.platform.tilelinkdemo.SocSim --xlen 64 --nax-count 1  --load-elf  ext/NaxSoftware/baremetal/mmu_sv39/build/rv64ima/mmu_sv39.elf --start-symbol _start --pass-symbol pass --trace --no-rvls"
```
#### execution result 
```Bash
[info] Sim starting <3
[info] Hart 0
[info] - i$ refill = 60
[info] - d$ refill = 15
[info] TEST PASS
[info] PASS FILE ./simWorkspace/SocDemo/logs/nax/mmu_sv39/mmu_sv39/PASS
[info] [Done] Simulation done in 582,732 ms
[info] 1 TESTS PASSED & 0 TESTS FAILED
[info] LIST OF TESTS FAILED
[success] Total time: 17 s, completed 22 mars 2024 15:15:47
```

> As the memory layout isn't the same than the pure verilator regression testbench.

Could you tell me more to know what I need to modify for the test to succeed?

Dolu1990, 2024-03-22:
Many of the hand written assembly test aren't self tested, but instead rely on the lock-step checker to trigger error on missmatch between the model and the hardware.
SocSim still trap on this => OK. The mcause code aren't checked by the software itself.

Bill94l, 2024-04-04:
Hi Charles,

There is a mismatch between the trap code returned by the DUT and REF according to the execution result of mmu_sv39.elf. The execution result shows that the DUT returns code 13 which is equivalent to LOAD_PAGE_FAULT and the REF returns code 5 which is equivalent to LOAD_ACCESS_FAULT.

```Shell
*Spike.log*
core   0: 0x0000000080000740 (0x0000a083) lw      ra, 0(ra)
core   0: exception trap_load_access_fault, epc 0x0000000080000740
core   0:           tval 0x0000001000000000
```
Here are the available storage regions:
```Shell
[info] Type: 0 Base 80000000 Size 80000000
[info] Type: 1 Base 10010000 Size 10000
[info] Type: 1 Base 10c00000 Size 400000
[info] Type: 1 Base 10000000 Size 10000
[info] Type: 1 Base 10020000 Size be0000
[info] Type: 1 Base 11000000 Size 3000000
[info] Type: 1 Base 14001000 Size bfff000
[info] Type: 1 Base 14000000 Size 1000
```
So the address 0x0000001000000000 does not exist in the memory regions, so spike returns an access fault, the DUT should also return an access fault, but currently returns a page fault.

At what level does the DUT's MMU check for existing memory regions?

Dolu1990, 2024-04-09:
Hi,

In which testbench this happened ? (the regression C++ one, or the SoCSim one) ?

> At what level does the DUT's MMU check for existing memory regions?

The MMU doesn't check for existing memory regions in NaxRiscv. instead,  the L1 cache will know at run time via memory bus read response if the given asked region is OK or not. And that information will propagate to the LSU during load/store execute.

What mmu_sv39.elf are you using ? i don't know what 0x0000000080000740 is about

Bill94l, 2024-04-11:
Hi, 
> In which testbench this happened ? (the regression C++ one, or the SoCSim one) ?

In SocSim testbench

> What mmu_sv39.elf are you using ? i don't know what 0x0000000080000740 is about

To be able to launch Linux and pass the mmu_sv39.elf test and avoid the missmatch, I made modifications that you will find in the attached file.

[rvls_mmu_modif.txt](https://github.com/SpinalHDL/NaxRiscv/files/14949790/rvls_mmu_modif.txt)

With these modifications 
- The mmu tests passed correctly 
- The Linux boot passes correctly for the different 32-bit ISA configurations

But the Linux boot fails for the different 64-bit ISA configurations and boot fails at this level
```Shell
info] Firmware Size       : 72 KB
[info] Runtime SBI Version : 0.2
[info] MIDELEG : 0x0000000000000222
[info] MEDELEG : 0x000000000000b109
[info] [    0.000000] Linux version 5.10.1 (rawrr@rawrr) (riscv64-buildroot-linux-uclibc-gcc.br_real (Buildroot 2020.11-rc3-8-g9ef54b7d0b) 10.2.0, GNU ld (GNU Binutils) 2.34) #8 SMP Mon Mar 14 10:26:33 CET 2022
[info] [    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[info] [    0.000000] printk: bootconsole [sbi0] enabled
[info] [    0.000000] Initial ramdisk at: 0x(____ptrval____) (8388608 bytes)
[info] [    0.000000] Zone ranges:
[info] [    0.000000]   DMA32    [mem 0x0000000080400000-0x000000008fffffff]
[info] [    0.000000]   Normal   empty
[info] [    0.000000] Movable zone start for each node
[info] [    0.000000] Early memory node ranges
[info] [    0.000000]   node   0: [mem 0x0000000080400000-0x000000008fffffff]
[info] [    0.000000] Initmem setup node 0 [mem 0x0000000080400000-0x000000008fffffff]
[info] [    0.000000] software IO TLB: mapped [mem 0x000000008bc8c000-0x000000008fc8c000] (64MB)
[info] [    0.000000] SBI specification v0.2 detected
[info] [    0.000000] SBI implementation ID=0x1 Version=0x8
[info] [    0.000000] SBI v0.2 TIME extension detected
[info] [    0.000000] SBI v0.2 IPI extension detected
[info] [    0.000000] SBI v0.2 RFENCE extension detected
[info] [    0.000000] SBI v0.2 HSM extension detected
[info] [    0.000000] riscv: ISA extensions aim
[info] [    0.000000] riscv: ELF capabilities aim
[info] [    0.000000] percpu: Embedded 14 pages/cpu s25560 r0 d31784 u57344
[info] [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 63630
[info] [    0.000000] Kernel command line: rootwait console=hvc0 earlycon=sbi root=/dev/ram0 init=/sbin/init
[info] [    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[info] [    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[info] [    0.000000] Sorting __ex_table...
[info] [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[info] [    0.000000] Memory: 173364K/258048K available (4703K kernel code, 679K rwdata, 718K rodata, 176K init, 262K bss, 84684K reserved, 0K cma-reserved)
[info] [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[info] [    0.000000] rcu: Hierarchical RCU implementation.
[info] [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=1.
[info] [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[info] [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[info] [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[info] [    0.000000] riscv-intc: 64 local interrupts mapped
[info] [    0.000000] random: get_random_bytes called from start_kernel+0x364/0x4f8 with crng_init=0
[info] [    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[info] [    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x171024e7e0, max_idle_ns: 440795205315 ns
[info] [    0.000060] sched_clock: 64 bits at 100MHz, resolution 10ns, wraps every 4398046511100ns
[info] [    0.001386] Console: colour dummy device 80x25
[info] [    0.001855] printk: console [hvc0] enabled
[info] [    0.001855] printk: console [hvc0] enabled
[info] [    0.002537] printk: bootconsole [sbi0] disabled
[info] [    0.002537] printk: bootconsole [sbi0] disabled
[info] [    0.003408] Calibrating delay loop (skipped), value calculated using timer frequency.. 200.00 BogoMIPS (lpj=400000)
[info] [    0.004326] pid_max: default: 32768 minimum: 301
[info] [    0.005908] Mount-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[info] [    0.006596] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[info] [    0.017190] rcu: Hierarchical SRCU implementation.
[info] mmio_load 1001bff8 8

```

Dolu1990, 2024-04-12:
> [info] [    0.003408] Calibrating delay loop (skipped), value calculated using timer frequency.. 200.00 BogoMIPS (lpj=400000)
[info] [    0.004326] pid_max: default: 32768 minimum: 301
[info] [    0.005908] Mount-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[info] [    0.006596] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[info] [    0.017190] rcu: Hierarchical SRCU implementation.
[info] mmio_load 1001bff8 8

SoCSim was kinda only used for 32 bits RISC-V so far.
The reason why you get that mmio_load fault, is because : 
```scala
class SocDemo ....
    val emulated = new tilelink.fabric.SlaveBus(
      M2sSupport(
        addressWidth = 28,
        dataWidth = 32,
        transfers = M2sTransfers(
          get = SizeRange(4),   //<- 32 bits only
          putFull = SizeRange(4) //<- 32 bits only
        )
      )
    )
```

> To be able to launch Linux and pass the mmu_sv39.elf test and avoid the missmatch, I made modifications that you will find in the attached file.

Nice thanks ^^
How do i apply your [rvls_mmu_modif.txt](https://github.com/SpinalHDL/NaxRiscv/files/14949790/rvls_mmu_modif.txt) to a local tree ?

git apply rvls_mmu_modif.txt
error: patch with only garbage at line 42

Saaaad XD

Dolu1990, 2024-04-15:
(SizeRange(4)) mean only 4 byte access can go to it,  can write SizeRange(4, 8) instead i think.

Bill94l, 2024-04-15:
> How do i apply your [rvls_mmu_modif.txt](https://github.com/SpinalHDL/NaxRiscv/files/14949790/rvls_mmu_modif.txt) to a local tree ?

Because I had already made other changes to be able to run all the regression tests, I didn't make the changes on a clean version. I will generate a clean patch when finished.

> SoCSim was kinda only used for 32 bits RISC-V so far.

With my current modifications, the 64-bit Linux boot works correctly with RVLS. However, it takes almost 3 hours.

```Scala
class SocDemo ....
    val emulated = new tilelink.fabric.SlaveBus(
      M2sSupport(
        addressWidth = 28,
        dataWidth = 32,
        transfers = M2sTransfers(
          get = SizeRange(4),   //<- 32 bits only
          putFull = SizeRange(4) //<- 32 bits only
        )
      )
    )
```
- Maybe if I apply the modification you suggested SizeRange(4, 8) it could speed up the boot time? (Perhaps this will increase the data access rate)
- if [SizeRange(4)] means only 4 bytes, [SizeRange(4, 8)] what does it mean?
- For the 64-bit version, the dataWidth will still be 32 bits?

Dolu1990, 2024-04-16:
> SizeRange(4, 8) it could speed up the boot time? 

no

> SizeRange(4, 8)] what does it mean?

Both 32 bits and 64 bits burst supported

> For the 64-bit version, the dataWidth will still be 32 bits?

clint access will stay with 32 bits data bus

>  9 hours. Is this normal?

Not it isn't normal at all. Maybe double check the size of the simWorkspace, to see if it has captured some waveform.

Else, mabe your verilator version is cursed ?

Uncomment     cd.forkSimSpeedPrinter(1.0)  to get an idea of the simulation speed.

How many naxriscv cpu are you simulating ?

Bill94l, 2024-04-16:
> Else, mabe your verilator version is cursed ?

I use Verilator v4.216

> How many naxriscv cpu are you simulating ?

I simulate just one naxriscv cpu

> Not it isn't normal at all. Maybe double check the size of the simWorkspace, to see if it has captured some waveform.

La taille des fichier générer par la simulation
 - nax_buildroot_run1.fst 147.82GB (waveform)
 - spike.log 17.95GB
 - tracer.log 11.03GB

Compared to the size of the output files, maybe 9h is normal ?

> Uncomment cd.forkSimSpeedPrinter(1.0) to get an idea of the simulation speed.

- Without  --trace  parameter : Speedup = 69 kcycles/s
-  With  --trace  parameter : Speedup = 10 kcycles/s

I thought it would take longer to generate the waveforms. However, if I disable the --trace option, the waveforms are still generated, donc juste la génération de tracer.log fait baissé la vitesse de simulation d'un facteur de 7. il y'a un moyen de desactiver la generation des waveform avec SocSim to see the simulation speed without waveform generation ?

Dolu1990, 2024-04-22:
> La taille des fichier générer par la simulation

Ahhhhhh damned,
Yes so, if you have those being generated, you probably run 20 time slower than without.
That is the issue. So 7h to boot link with those option is normal. It should take around 20 mn without those for a single core nax sim.

> 69 kcycles/s 

How many nax CPU are you running ?

Bill94l, 2024-04-22:
> How many nax CPU are you running ?

I'm running just one nax CPU

Dolu1990, 2024-04-22:
On what computer config ?

Dolu1990, 2024-04-22:
hmm that is a bit slow for such CPU i would say.
Are you running in a VM ?

Bill94l, 2024-04-22:
> Are you running in a VM ?

I'm not running in a VM

Dolu1990, 2024-04-22:
Ahhh my bad', that test bench is slower than what i remember, i just tried on Ryzen 5800X3D, got : 
[Info] Simulation speed :   82 kcycles/s

Enabling allOptimization help a bit: 
  val sc = SimConfig
  sc.allOptimisation
  
[Info] Simulation speed :   92 kcycles/s

closed 2024-06-22 by Bill94l