SpinalHDL/NaxRiscv #96: about litex and dhrystone

duanjiulon, 2024-04-30:
Hi,dear dulo,
I am currently learning how to use the kernel generated by Litex to run the dhrystone algorithm on SOC. The specific operation is: 1. Generate a SOC system with a Nax core, and then based on the dhrystone source code and simulation steps provided by you in your Naxriscv, set the starting address of the linker script to 0x0000000. This address is the address of the ROM, which can run the program. However, once the drystone program is loaded, it freezes and cannot obtain the corresponding results in Verilator, May I ask why this is? Is it because I lack any environmental dependencies?

Dolu1990, 2024-04-30:
Hi,

Which dhrystone software are you using, i don't remember providing any dhrystone for litex.
Did you ported the IO ?

duanjiulon, 2024-04-30:
No,No,No,It's not about the dhrystone program for litex, but the dhrystone program and makefile in the ext/Maxsofware directory of your Naxriscv directory，

发自我的iPhone

------------------ Original ------------------
From: Dolu1990 ***@***.***&gt;
Date: Tue,Apr 30,2024 6:22 PM
To: SpinalHDL/NaxRiscv ***@***.***&gt;
Cc: duanjiulon ***@***.***&gt;, Author ***@***.***&gt;
Subject: Re: [SpinalHDL/NaxRiscv] about litex and dhrystone (Issue #96)

Dolu1990, 2024-04-30:
>  but the dhrystone program and makefile in the ext/Maxsofware

No software from ext/NaxSoftware is designed to run on litex (excepted the instruction for debian)

duanjiulon, 2024-05-19:
> No software from ext/NaxSoftware is designed to run on litex (excepted the instruction for debian)

OK,fine,dear dolu,I have done this by litex intergration, 
But his coremark performance is around 3.14. I would like to inquire about which key parameters need to be configured when configuring SOC parameters to achieve the best performance of nax_core, which is a coremark score of 5.02. Looking forward to your reply.！

Dolu1990, 2024-05-21:
Hi,

On litex, i generaly use : 
python3 -m litex_boards.targets.digilent_nexys_video --cpu-type=naxriscv  --bus-standard axi-lite --with-video-framebuffer --with-coherent-dma --with-sdcard --with-ethernet --xlen=64 --scala-args='rvc=true,rvf=true,rvd=true,alu-count=2,decode-count=2'  --with-jtag-tap --sys-clk-freq 100000000 --cpu-count 2 --soc-json build/digilent_nexys_video/csr.json --update-repo no --load

to get a debian ready core, so you can change it with  --xlen=32  and rvc=false,rvf=false,rvd=false

Then about coremark, it realy depend how you compiled it.
to get 5+ coremark, i used : 
https://github.com/SpinalHDL/NaxSoftware/blob/main/baremetal/coremark/naxriscv_sim/core_portme.mak#L33

Also, with binary in the main memory, not in io mapped one.

I didn't tested on litex specificaly, that coremark was for simulation only.

duanjiulon, 2024-05-22:
> I didn't tested on litex specificaly, that coremark was for simulation only.

Hi，dear Dolu，
1. I would like to ask for some ideas about FetchCachePluginAXI4. scala here. Why is the Fetch interface directly divided into two types of axial interfaces: axial and axial, and what is its purpose
```
class FetchAxi4(ramDataWidth : Int,
                ioDataWidth : Int,
                toPeripheral : FetchL1Cmd => Bool
                ) extends Plugin{
  val logic = create late new Area{
    val cache = getService[FetchCachePlugin]
    val native = cache.mem.setAsDirectionLess

    val (ram, peripheral) = native.split(toPeripheral)
    val axiRam = master(ram.resizer(ramDataWidth).toAxi4())
    val axiPeripheral  = master(peripheral.resizer(ioDataWidth).toAxiLite4())

    Axi4SpecRenamer(axiRam)
    AxiLite4SpecRenamer(axiPeripheral)
  }
}
```
2.There is a setting in Briey.scala regarding ibus and dbus:
```
Val core=new Area{
Val config=VexRiscvConfig（
Plugins=cpuPlugins+=new DebugPlugin (debugClockDomain)
）

Val CPU=new VexRiscv (config)
Var iBus: Axi4ReadOnly=null
Var dBus: Axi4Shared=null
For (plugin<- config. plugins) plugin match{
Case plugin: IBusSimplePlugin=>iBus=plugin. iBus. toAxi4ReadOnly()
Case plugin: IBusCachedPlugin=>iBus=plugin. iBus. toAxi4ReadOnly()
Case plugin: DBusSimplePlugin=>dBus=plugin. dBus. toAxi4Shared()
Case plugin: DBusCachedPlugin=>dBus=plugin. dBus. toAxi4Shared (true)
Case plugin: CsrPlugin=>{
Plugin. externalInterrupt:=BufferCC (io. coreInterrupt)
Plugin. timerInterrupt:=timerCtrl. io. interrupt
}
Case plugin: DebugPlugin=>debugClockDomain{
ResetCtrl. axiReset setWhen (RegNext (plugin. io. resetOut))
Io. jtag<>plugin. io. bus. frmJtag()
}
Case_=>
}
}
```
What are the differences between the settings of ibus and dbus in the two types of cores
in the scala file：
```
 val core = new NaxRiscv(l)
    core.plugins.foreach{
      case p : FetchCachePlugin => iBus.bus << p.mem.toTilelink()
      case p : DataCachePlugin =>  dBus.bus << p.mem.toTilelink()
      case p : Lsu2Plugin => pBus.bus << p.peripheralBus.toTilelink()
      case p : PrivilegedPlugin => {
        p.io.int.machine.timer := mti.flag
        p.io.int.machine.software := msi.flag
        p.io.int.machine.external := mei.node.flag
        if(p.p.withSupervisor) p.io.int.supervisor.external := sei.node.flag
        if(p.p.withRdTime) p.io.rdtime := clint.thread.core.io.time
      }
      case _ =>
    }
  }
```
finally，I want to develop an Ibus and dbus interface similar to vex_core, but I don't know how to handle pbus

Dolu1990, 2024-05-22:
Hi,

> I would like to ask for some ideas about FetchCachePluginAXI4. scala here. Why is the Fetch interface directly divided into two types of axial interfaces: axial and axial, and what is its purpose

I think the reason was to be compatible with litex, but overall, a single bus is fine.

> What are the differences between the settings of ibus and dbus in the two types of cores
in the scala file：

I'm not sure to understand, one is AXI one is tilelink, but fondamentaly the kind of access going through is the same, just that nax is writeback => less write trafic

> but I don't know how to handle pbus

Would need a bridge.

But overall why not using the FetchAxi4 in the 
val core = new NaxRiscv(l)
    core.plugins.foreach{
    
    ?
 I'm confused by the code above

duanjiulon, 2024-05-22:
> Would need a bridge.

There is currently no lib library available for converting pbus to AXI, and as far as I know, the LSUPlgins. scala file only has the axilite conversion interface.

>I'm not sure to understand, one is AXI one is tilelink, but fondamentaly the kind of access going through is the same, just that nax is writeback => less write trafic

What I mean is the difference in functionality between FetchPlgins with AXI4 interface and ibus-axi4readonly in vex, and datacache-axi4 interface and vex-axi4dbus in vex

>I'm confused by the code above

what you say is:
```
val core = new NaxRiscv(l)
    core.plugins.foreach{
      case p : FetchCachePlugin => iBus.bus << p.mem.toTilelink()
      case p : DataCachePlugin =>  dBus.bus << p.mem.toTilelink()
      case p : Lsu2Plugin => pBus.bus << p.peripheralBus.toTilelink()
      case p : PrivilegedPlugin => {
        p.io.int.machine.timer := mti.flag
        p.io.int.machine.software := msi.flag
        p.io.int.machine.external := mei.node.flag
        if(p.p.withSupervisor) p.io.int.supervisor.external := sei.node.flag
        if(p.p.withRdTime) p.io.rdtime := clint.thread.core.io.time
      }
      case _ =>
    }
  }
```
this one？

duanjiulon, 2024-05-22:
dear dulo， in the gen.scala, I add the code:
```
plugins += new DataCacheAxi4(
      dataWidth = 32
    )
    plugins += new FetchAxi4(
      dataWidth = 32
    )
    ////
    plugins += new LsuPeripheralAxiLite4(
      ioDataWidth = 32
    )
    ////

```
output wire          FetchAxi4_logic_axi_arvalid,
  input  wire          FetchAxi4_logic_axi_arready,
  output wire [31:0]   FetchAxi4_logic_axi_araddr,
  output wire [7:0]    FetchAxi4_logic_axi_arlen,
  output wire [2:0]    FetchAxi4_logic_axi_arsize,
  output wire [1:0]    FetchAxi4_logic_axi_arburst,
  output wire [2:0]    FetchAxi4_logic_axi_arprot,
  input  wire          FetchAxi4_logic_axi_rvalid,
  output wire          FetchAxi4_logic_axi_rready,
  input  wire [31:0]   FetchAxi4_logic_axi_rdata,
  input  wire [1:0]    FetchAxi4_logic_axi_rresp,
  input  wire          FetchAxi4_logic_axi_rlast,
  output wire          DataCacheAxi4_logic_axi_awvalid,
  input  wire          DataCacheAxi4_logic_axi_awready,
  output wire [31:0]   DataCacheAxi4_logic_axi_awaddr,
  output wire [0:0]    DataCacheAxi4_logic_axi_awid,
  output wire [7:0]    DataCacheAxi4_logic_axi_awlen,
  output wire [2:0]    DataCacheAxi4_logic_axi_awsize,
  output wire [1:0]    DataCacheAxi4_logic_axi_awburst,
  output wire [2:0]    DataCacheAxi4_logic_axi_awprot,
  output wire          DataCacheAxi4_logic_axi_wvalid,
  input  wire          DataCacheAxi4_logic_axi_wready,
  output wire [31:0]   DataCacheAxi4_logic_axi_wdata,
  output wire [3:0]    DataCacheAxi4_logic_axi_wstrb,
  output wire          DataCacheAxi4_logic_axi_wlast,
  input  wire          DataCacheAxi4_logic_axi_bvalid,
  output wire          DataCacheAxi4_logic_axi_bready,
  input  wire [0:0]    DataCacheAxi4_logic_axi_bid,
  input  wire [1:0]    DataCacheAxi4_logic_axi_bresp,
  output wire          DataCacheAxi4_logic_axi_arvalid,
  input  wire          DataCacheAxi4_logic_axi_arready,
  output wire [31:0]   DataCacheAxi4_logic_axi_araddr,
  output wire [0:0]    DataCacheAxi4_logic_axi_arid,
  output wire [7:0]    DataCacheAxi4_logic_axi_arlen,
  output wire [2:0]    DataCacheAxi4_logic_axi_arsize,
  output wire [1:0]    DataCacheAxi4_logic_axi_arburst,
  output wire [2:0]    DataCacheAxi4_logic_axi_arprot,
  input  wire          DataCacheAxi4_logic_axi_rvalid,
  output wire          DataCacheAxi4_logic_axi_rready,
  input  wire [31:0]   DataCacheAxi4_logic_axi_rdata,
  input  wire [0:0]    DataCacheAxi4_logic_axi_rid,
  input  wire [1:0]    DataCacheAxi4_logic_axi_rresp,
  input  wire          DataCacheAxi4_logic_axi_rlast,
  input  wire          reset,
  output wire          LsuPeripheralAxiLite4_logic_axi_awvalid,
  input  wire          LsuPeripheralAxiLite4_logic_axi_awready,
  output wire [31:0]   LsuPeripheralAxiLite4_logic_axi_awaddr,
  output wire [2:0]    LsuPeripheralAxiLite4_logic_axi_awprot,
  output wire          LsuPeripheralAxiLite4_logic_axi_wvalid,
  input  wire          LsuPeripheralAxiLite4_logic_axi_wready,
  output wire [31:0]   LsuPeripheralAxiLite4_logic_axi_wdata,
  output wire [3:0]    LsuPeripheralAxiLite4_logic_axi_wstrb,
  input  wire          LsuPeripheralAxiLite4_logic_axi_bvalid,
  output wire          LsuPeripheralAxiLite4_logic_axi_bready,
  input  wire [1:0]    LsuPeripheralAxiLite4_logic_axi_bresp,
  output wire          LsuPeripheralAxiLite4_logic_axi_arvalid,
  input  wire          LsuPeripheralAxiLite4_logic_axi_arready,
  output wire [31:0]   LsuPeripheralAxiLite4_logic_axi_araddr,
  output wire [2:0]    LsuPeripheralAxiLite4_logic_axi_arprot,
  input  wire          LsuPeripheralAxiLite4_logic_axi_rvalid,
  output wire          LsuPeripheralAxiLite4_logic_axi_rready,
  input  wire [31:0]   LsuPeripheralAxiLite4_logic_axi_rdata,
  input  wire [1:0]    LsuPeripheralAxiLite4_logic_axi_rresp,

```

I have modified fetchaxi4. scala to single port readonly axi4 and obtained the result. Can the three interfaces here be considered as ibus, dbus, and pbus。thank you so much

duanjiulon, 2024-06-05:
Hello, dear dulo,recently I tested multiple core configurations of naxriscv in your recommended emulation method (on git) emulation and litex separately, and found that for the first way: loading the elf file into a specific sim_mem for testing will have better results, 2-alu, 2-decode will have a dhrystone score of 2.93, but under the same configuration, I use litex_ SoC loading ROM for simulation or board testing, the result is only about 1 drystone score,please help me why is this,thanks

Dolu1990, 2024-06-05:
Could you send me the fst wave of the litex simulation ?

duanjiulon, 2024-06-06:
Hi，dear Dolu，Considering that I often use the suffix vcd file to analyze waveforms, and the fst wave file cannot be opened correctly on my computer, I have sent you the. vcd file and fst file together. The program flow of dhrystone is as follows: 1. Print version information to the serial port, you can check reslutREADME.txt, then read the start time, start running dhrystone, read the end time, and print the result to the serial port after running. The compressed file size may be around 2GB, please be patient and download it. Thank you for your suppor！

段玖龙
***@***.***

 
Hi,

 

从QQ邮箱发来的超大附件

fst wave.zip (2.23G, 2024年07月06日 11:08 到期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=25343463c6aaba420053485e1264061c1c400007515505554902015b064902515c0719525601511e01510400575705020151555634683455174014145512511d1e5d446309&t=exs_ftn_download&code=d44c4d43

duanjiulon, 2024-06-07:
dear dulo,I'm sorry to ask you another question. If I still want to get a lightweight core, which plugins do I need to close while ensuring that the core can run? For example, can branch prediction be closed? Can many random launches be achieved by turning off certain piungis and turning them into sequential launches, resulting in a core that occupies less LUT resources (similar to vex_core)，thank you so much！

Dolu1990, 2024-06-07:
Hi,

I checked the simulation wave, and i can see the the integer divider is never used, did you compiled with -march=rv32im ?

Dolu1990, 2024-06-07:
>  which plugins do I need to close while ensuring that the core can run?

You may remove the PerformanceCounterPlugin, reduce the LQ SQ size and swap the MMU for the StaticAddressTranslationPlugin, ensure the core if configure with single decoder, 1 alu, no RVC

Also, you may be interrested by VexiiRiscv is you look for small 64 bits support, but in order.

duanjiulon, 2024-06-07:
> Hi,
> 
> I checked the simulation wave, and i can see the the integer divider is never used, did you compiled with -march=rv32im ?

Besides that, I changed arch to im and my score increased to 1.09,Could you please take a look at the waveform again? Do you have any other questions

Dolu1990, 2024-06-07:
> Could you please take a look at the waveform again?

In the wave form you sent me, divider was never used => that wave form has no use anymore.

Can you send me your complied dhrystone ?

Dolu1990, 2024-06-07:
elf file

duanjiulon, 2024-06-07:
> elf file

I only modified the print function and time access function in the code, and did not modify anything else

Dolu1990, 2024-06-07:
The btb has only 32 entries ? BtbPlugin_logic_mem. If yes, that is way to few

Dolu1990, 2024-06-07:
Should be   (* ram_style = "block" *) reg [49:0] BtbPlugin_logic_mem [0:511];

duanjiulon, 2024-06-07:
I have not made any changes to the plugins related to BTB, and the entire SOC is generated through commands.

发自我的iPhone

Dolu1990, 2024-06-07:
Can you send pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/NaxRiscvLitex*.v ?

Dolu1990, 2024-06-07:
I still get   (* ram_style = "block" *) reg [50:0] BtbPlugin_logic_mem [0:511];

The point is to figure out what you are running exactly, so, realy need to check that on your generated verilog files.

duanjiulon, 2024-06-07:
> I still get (* ram_style = "block" *) reg [50:0] BtbPlugin_logic_mem [0:511];
> 
> The point is to figure out what you are running exactly, so, realy need to check that on your generated verilog files.

Sorry for keeping you waiting for so long
[NaxRiscvLitex_2b34dda3f33457e82fa832a726b15dfa.zip](https://github.com/user-attachments/files/15739099/NaxRiscvLitex_2b34dda3f33457e82fa832a726b15dfa.zip)

duanjiulon, 2024-06-07:
> I still get (* ram_style = "block" *) reg [50:0] BtbPlugin_logic_mem [0:511];
> 
> The point is to figure out what you are running exactly, so, realy need to check that on your generated verilog files.

dear dolu,I try to use the v file to test, this file BtbPlugin_logic_mem is [0:511],but the result dhrystone is 1.18
[NaxRiscvLitex_6b78f906f0b9e3bb22e097bcf4713986.zip](https://github.com/user-attachments/files/15739199/NaxRiscvLitex_6b78f906f0b9e3bb22e097bcf4713986.zip)

duanjiulon, 2024-06-07:
I will also send you the waveform of this v file. Could you please take a look at other issues

段玖龙
***@***.***

 

 
 

从QQ邮箱发来的超大附件

vcd.zip (1.42G, 2024年07月07日 21:22 到期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=096430395f806dc351034c04443502494d10510c5b5701561806530b0418075151001d0b0601074b040702005701515e5052075d6232301056001e430b45305b&t=exs_ftn_download&code=5d09b50f

Dolu1990, 2024-06-10:
> I will also send you the waveform of this v file

Thanks, waiting for it.

duanjiulon, 2024-06-10:
> > I will also send you the waveform of this v file
> 
> Thanks, waiting for it.

https://mail.qq.com/cgi-bin/ftnExs_download?k=096430395f806dc351034c04443502494d10510c5b5701561806530b0418075151001d0b0601074b040702005701515e5052075d6232301056001e430b45305b&t=exs_ftn_download&code=5d09b50f

Dolu1990, 2024-06-11:
Two things : 
- The CPU is getting a lot of store to load hazards, how did you compiled the dhrystone benchmark ? (exactly, also can you send the elf ?)
- The benchmark results given in the doc are without RVC, your software is running with RVC

duanjiulon, 2024-06-11:
1-

The program for dhrystone is based on the source files in your ext/Maxsofware directory, and then modifies the serial port printing function and timer function to obtain the elf file. The other source codes have not changed, and the makefile is also a makefile mainly composed of asm.mk in the common directory. I only used the rv32 compiler, and the rest have not changed
2-
Do I need to add a c to the arch parameter?

发自我的iPhone

duanjiulon, 2024-06-11:
The first v file sent to you has rvc, while the second one does not

发自我的iPhone

duanjiulon, 2024-06-12:
> Two things :
> 
>     * The CPU is getting a lot of store to load hazards, how did you compiled the dhrystone benchmark ? (exactly, also can you send the elf ?)

[dhrystone.zip](https://github.com/user-attachments/files/15796580/dhrystone.zip)

Dolu1990, 2024-06-13:
DEBUG?=yes

You need to turn that to no.

duanjiulon, 2024-06-13:
> DEBUG?=yes
> 
> You need to turn that to no.

```
/home/jlduan/rv-toolchain//bin/riscv32-anlogic-elf-gcc -fno-common -fno-inline -Wno-implicit-int -Wno-implicit-function-declaration -DTIME -DCORE_HZ=12000000ll -DUSE_GP -mabi=ilp32 -mcmodel=medany -O3  -I/home/jlduan/litex/NaxRiscv/ext/NaxSoftware/baremetal/include -I/home/jlduan/litex/NaxRiscv/ext/NaxSoftware/baremetal/driver -o build/test.elf build//home/jlduan/litex/NaxRiscv/ext/NaxSoftware/baremetal/dhrystone/src2/uart.o build//home/jlduan/litex/NaxRiscv/ext/NaxSoftware/baremetal/dhrystone/src2/main.o build//home/jlduan/litex/NaxRiscv/ext/NaxSoftware/baremetal/dhrystone/src2/dhrystone.o build//home/jlduan/litex/NaxRiscv/ext/NaxSoftware/baremetal/common/start.o -march=rv32ima_zicsr  -mabi=ilp32 -mcmodel=medany -lc -L/home/jlduan/litex/NaxRiscv/ext/NaxSoftware/baremetal/common -nostdlib -lgcc -nostartfiles -ffreestanding -Wl,-Bstatic,-T,/home/jlduan/litex/NaxRiscv/ext/NaxSoftware/baremetal/common/app.ld,-Map,build/test.map,--print-memory-usage
```
what you mean is argument -O3?

duanjiulon, 2024-09-12:
> > > I will also send you the waveform of this v file
> > 
> > 
> > Thanks, waiting for it.
> 
> https://mail.qq.com/cgi-bin/ftnExs_download?k=096430395f806dc351034c04443502494d10510c5b5701561806530b0418075151001d0b0601074b040702005701515e5052075d6232301056001e430b45305b&t=exs_ftn_download&code=5d09b50f The link is just three dots behind &nbsp, just click on it, and I'll send it to you again for your convenience Do you see anything wrong with the V-file?

Dear dulo，Do you still have the V-files of the waveforms I sent to you on your computer? I recently changed my computer and couldn't find those ones ，zhanks

duanjiulon, 2024-09-13:
> I'm in a travel, will be back the 18, will be able to check then, not sure.

thanks so much，I’ve found it。
https://github.com/SpinalHDL/NaxRiscv/blob/main/src/test/cpp/naxriscv/README.md
When I was running the simulation using the new Linux system according to the tutorial you provided, an error occurred in the last step. Can you please take a look at what happened
```
g++    main.o verilated.o verilated_dpi.o verilated_fst_c.o VNaxRiscv__ALL.a   -pthread -lSDL2 -lz /home/jlduan/work/NaxRiscv/ext/riscv-isa-sim/build/libspike_main.a /home/jlduan/work/NaxRiscv/ext/riscv-isa-sim/build/libriscv.a /home/jlduan/work/NaxRiscv/ext/riscv-isa-sim/build/libdisasm.a /home/jlduan/work/NaxRiscv/ext/riscv-isa-sim/build/libsoftfloat.a /home/jlduan/work/NaxRiscv/ext/riscv-isa-sim/build/libfesvr.a /home/jlduan/work/NaxRiscv/ext/riscv-isa-sim/build/libfdt.a  -lpthread -ldl -lboost_regex -lboost_system  -lpthread  -lboost_system -lboost_regex  -o VNaxRiscv
collect2: error: ld returned 1 exit status
make[1]: *** [VNaxRiscv.mk:85: VNaxRiscv] Error 1
rm VNaxRiscv__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/home/jlduan/work/NaxRiscv/src/test/cpp/naxriscv/obj_dir'
make: *** [makefile:94: compile] Error 2
```

Dolu1990, 2024-09-14:
I just followed the whole https://github.com/SpinalHDL/NaxRiscv/blob/main/src/test/cpp/naxriscv/README.md#how-to-setup-things with a fresh clone, all went well.
Was there any error message above the logs you showed ?
It is kinda weird that ld just return 1  and exits like that without any info XD