SpinalHDL/NaxRiscv #102: How to customize hardware architecture

phillippZZ, 2024-05-29:
Hi, Dolu and everyone.

I'm a first time user and want to learn how to configure NaxRiscv. Let's say, if I'm going to have a 4-core, 2 cache-level architecture, where and how should I set everything up? 
Plus, can I simply change the values of parameters defined in Gen.gala to change the performance of components? Is the upper bound of the parameters defined somewhere so that I can use them to make the architecture as fast as possbible without breaking anything?
I don't have related experience before. So please forgive me for asking these basic quesitons.

Thank you.

Dolu1990, 2024-05-29:
Hi,

> I'm a first time user and want to learn how to configure NaxRiscv. Let's say, if I'm going to have a 4-core, 2 cache-level architecture, where and how should I set everything up?

I would say it depend what you want to do, if you want to go on hardware easily, you should use the litex integration : 
https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/hardware/index.html#litex

There you have a set of parameters that you can tweek.

Else there is this kind of simulation SoC : 
https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/platform/tilelinkdemo/SocDemo.scala#L19

You would need to dig in the TilelinkNaxRiscvFiber.getCoherentConfig to modify parameters

> Plus, can I simply change the values of parameters defined in Gen.gala to change the performance of components? https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/Gen.scala

Yes

> Is the upper bound of the parameters defined somewhere so that I can use them to make the architecture as fast as possbible without breaking anything?

Not realy, i would say, the main upper bound is : 
              aluCount : Int = 2,
              decodeCount : Int = 2,
Otherwise it is mostly free range.

phillippZZ, 2024-06-01:
Hi, Dolu. 
I checked the following link: 
> https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/hardware/index.html#litex

Then I tried the command:

> sbt "runMain naxriscv.platform.tilelinkdemo.SocSim --load-elf ext/NaxSoftware/baremetal/dhrystone/build/rv32ima/dhrystone.elf --no-rvls --iverilog --asic"

 Somehow it raised an error:

>  [error] Exception in thread "main" java.io.IOException: Cannot run program "iverilog-vpi": error=2, No such file or directory
[error]         at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
[error]         at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
[error]         at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.scala:75)
[error]         at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang(ProcessBuilderImpl.scala:119)
[error]         at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.slurp(ProcessBuilderImpl.scala:135)
...

And it ran correctly as I removed "--iverilog".  Did I miss something during the installation stage?
> ...
[info] [Done] Simulation done in 14626.821 ms
[success] Total time: 19 s, completed May 31, 2024, 9:06:47 PM

Another question is that after I modify "SocDemo.scala" and use "sbt" command to generate naxriscv, should I use the following command to simulate the architecture and run linux on it for testing?
> cd $NAXRISCV/src/test/cpp/naxriscv
export LINUX_IMAGES=$NAXRISCV/ext/NaxSoftware/buildroot/images/rv32ima
./obj_dir/VNaxRiscv \
    --load-bin $LINUX_IMAGES/fw_jump.bin,0x80000000 \
    --load-bin $LINUX_IMAGES/linux.dtb,0x80F80000 \
    --load-bin $LINUX_IMAGES/Image,0x80400000 \
    --load-bin $LINUX_IMAGES/rootfs.cpio,0x81000000

Dolu1990, 2024-06-03:
Hi,

Did you installed icarus verilog ? 
sudo apt install -y iverilog

Note that the part of the readme you read is "exclusively" for ASIC, it isn't made to target FPGA.

> Another question is that after I modify "SocDemo.scala" and use "sbt" command to generate naxriscv, should I use the following command to simulate the architecture and run linux on it for testing?

The testbench you are using is only made to work with the Gen / Gen64 generators which generate raw naxriscv. SocDemo.scala is for a "full" soc.

If you want to simulate SocDemo, it is a different flow : 
- Run the simulation via naxriscv.platform.tilelinkdemo.SocSim
- With those arguments --load-bin 80000000,ext/NaxSoftware/buildroot/images/rv32ima/fw_jump.bin --load-bin 80F80000,ext/NaxSoftware/buildroot/images/rv32ima/linux.dtb --load-bin 80400000,ext/NaxSoftware/buildroot/images/rv32ima/Image --load-bin 81000000,ext/NaxSoftware/buildroot/images/rv32ima/rootfs.cpio --nax-count 1

phillippZZ, 2024-06-03:
Thank you, Dolu!

I tried your other project "linux-on-litex-vexriscv" yesterday and succeeded to simulate 4-core and 2-core architecture. I guess I'll stick with it for convenience.

[image]

But I haven't found the parameter to set 2-cache-level. May I ask is there a way to configure it, for example, using l2_cache_size = X?
> https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/sim.py
[image]

Dolu1990, 2024-06-03:
Don't use the litex l2 cache, because is is in the wishbone interconnect, while Vex / nax  directly go through high bandwidth interface to the sdram controller.

VexRiscv doesn't provide L2 cache support. NaxRiscv and VexiiRiscv have l2 option.
VexiiRiscv != VexRiscv :)

phillippZZ, 2024-06-03:
Could you please teach me where to add these options? Do I add them to sbt command like "sbt naxriscv.Gen --l2=true --cpu-count=4" or add them to the following command:
> ./obj_dir/VNaxRiscv
--load-bin $LINUX_IMAGES/fw_jump.bin,0x80000000
--load-bin $LINUX_IMAGES/linux.dtb,0x80F80000
--load-bin $LINUX_IMAGES/Image,0x80400000
--load-bin $LINUX_IMAGES/rootfs.cpio,0x81000000
--2=true --cpu-count=4

Sorry for keeping asking these simple questions.

phillippZZ, 2024-06-04:
As you mentioned before, SocSim isn't intended for FPGA. So I was wondering it there's a design intended for FPGA that I can simulate with 4 cores and 2 cache levels? I tried some objects defined in .scala files but couldn't run any of them successfully.
For example, "naxriscv.platform.litex.NaxGen" have options including "cpu-count" and "l2-bytes". But somehow only providing "cpu-count" in the "sbt" command caused an error since I didn't provide "memory-region" or some other parameters.

Dolu1990, 2024-06-04:
Currently the only SoC which is intended for FPGA is the one integrated into litex. Here is its SpinalHDL source code : 
https://github.com/SpinalHDL/NaxRiscv/tree/main/src/main/scala/naxriscv/platform/litex

So the best is probably to run linux : 
litex_sim  --cpu-type=naxriscv --help
litex_sim  --cpu-type=naxriscv --cpu-count=4

Thing is, the opensbi/linux image are different for that one XD
The ones specified by https://github.com/litex-hub/linux-on-litex-vexriscv should be ok for buildroot
If you want to run debian => https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/hardware/index.html#putting-debian-on-the-sdcard

phillippZZ, 2024-06-04:
Thank you Dolu. 
But I don't know how to load the images. 
> Thing is, the opensbi/linux image are different for that one

Should I modify some code or actually I can use an option to specify the image like "litex_sim --cpu-type=naxriscv --cpu-count=4 --load-elfxxx --load-binxxx"?

phillippZZ, 2024-06-04:
Thank you, Dolu.

I tried "litex/litex/tools/litex_sim.py --cpu-type=naxriscv --rom-init images/opensbi.bin  --ram-init images/boot.json" and got an error as:
>...
 [clocker] loaded
[xgmii_ethernet] loaded (0x55e4671a3190)
[gmii_ethernet] loaded (0x55e4671a3190)
[serial2tcp] loaded (0x55e4671a3190)
[spdeeprom] loaded (addr = 0x0)
[jtagremote] loaded (0x55e4671a3190)
[serial2console] loaded (0x55e4671a3190)
[ethernet] loaded (0x55e4671a3190)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0
FAILURE Tilelink decoder miss ???
\- /usr/src/pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/NaxRiscvLitex_76ae2c59eeb72af5d989a7c60ede0195.v:15362: Verilog $finish

Dolu1990, 2024-06-04:
With litex sim you need to provide a json file to specify what to load where. (--help)
You can find that json file in the linux-on-litex-vexriscv repo.

For instance : 
 --with-sdram --sdram-data-width=64 --sdram-init boot.json
 But be sure the dts memory config is smaller or equal to what that   --with-sdram --sdram-data-width=64 will provide you as emulated memory

Dolu1990, 2024-06-04:
>  --rom-init images/opensbi.bin 

Not good, should not have that argument, opensbi should be loaded via the boot.json

Dolu1990, 2024-06-05:
> The booting up of simulation and interaction with simulated linux console are quite slow. Would it help if I use "--thread=8" option?

Hmm, in general, multi-threaded simulation do not boost things much, as the overall design is too small for it to scale well.
Maybe 2-4 thread will help

> Is there'a way that I can store the simulation so that I don't have to repeat the long procedure of booting up every time?

Not that i'm aware of. But that would be usefull.

> How can I compile the softwares into the image file(The most important one).

Generaly, the idea with buildroot is to create a buildroot package and regenerate the whole thing. here is an example of package : 
https://github.com/SpinalHDL/buildroot-spinal-saxon/tree/main/package/mandelbrot

For how to generate the buildroot images, check the linux on litex vexriscv repo.

Good luck ^^

phillippZZ, 2024-06-05:
Thank you very much!!

closed 2024-06-10 by phillippZZ