SpinalHDL/NaxRiscv #109: Question about the sizes of cache L2 & L1

phillippZZ, 2024-06-18:
Hi, Dolu. 
I succeeded to simulate naxriscv with 4 cpu cores by using litex under your guidance. But later on I noticed that the size of cache L2 doesn't seem to be bigger than L1(I suppose "sram:8kb" represents the size of cache L1).
Could you tell me if I forgot to specify something? Just as a reminder, I used "litex/litex/tools/litex_sim.py --cpu-type=naxriscv --with-sdram --sdram-data-width=64 --sdram-init images/boot.json --cpu-count=4" to simulate.
[image]
[image]

Dolu1990, 2024-06-18:
Hi,

That's ok, that L2 cache is internal to litex and NaxRiscv will not read memory through it. Nax coherent infrastructure has its own l2 coherent cache which is 128 KB by default normaly.

Idealy, i would need to have it disabled automaticaly for Nax/Vexii cpus targets in litex.

phillippZZ, 2024-06-18:
Oh, I see. So the "l2" option in this case is actually related the wishbone stuff. I'll disable it. Thank you very much.

Dolu1990, 2024-08-02:
Hi ^^
Always 64 bytes

phillippZZ, 2024-08-02:
Got it! I'll remember it in my heart : )

phillippZZ, 2024-08-09:
Hi, @Dolu1990. Thank you for the help all the time. 
I've finished several tests on the naxriscv and everything goes well. 
But I do have one question and maybe few more in the future about the cache 😅
Given that l2 has total size as 128kb, cache line size as 64 kb and 8 ways by default, can I say that the number of sets are 256 by default? Do we decide the total size, line size and number of ways of l2 cache first, and then the number of sets can be calculated naturally?

Dolu1990, 2024-08-16:
> sets are 256 by default?

256 is correct

> Do we set the total size, line size and number of ways of l2 cache first,  and then the number of sets can be calculated automatically?

yes right

> L1 L2 

L1 are 4 way 64 sets
L2 is 8 way 256 sets

closed 2024-08-16 by phillippZZ