SpinalHDL/NaxRiscv #111: storeFresh unused : trap-store_access_fault

Bill94l, 2024-06-28:
Hi,

After running a program generated by riscv-dv for rv64imafdc, accessing address 0x7fffff20 generates an exception (trap_store_access_fault) in spike, but the DUT does the commit and RVLS detects a commit error. Here is the error message during execution
```Shell
[info] commit error
[info] - storeFresh unused ???
```
Dump 
```dump
0000000080004040 <mmode_exception_handler>:
    80004040:	1671                	addi	a2,a2,-4
    80004042:	e21c                	sd	a5,0(a2)
    80004044:	000607b3          	add	a5,a2,zero
    80004048:	f0078793          	addi	a5,a5,-256
    8000404c:	0017b423          	sd	ra,8(a5)
```
Spike log
```log
core   0: 0 0x00000000000009d6 (0x00c3543b) x 8 0x0000000008000000
core   0: 0x00000000000009da (0x0017b423) sd      ra, 8(a5)
core   0: exception trap_store_page_fault, epc 0x00000000000009da
core   0:           tval 0x000000000002cbd0
core   0: 0x0000000080004000 (0x0400006f) j       pc + 0x40
core   0: 3 0x0000000080004000 (0x0400006f)
core   0: 0x0000000080004040 (0x00001671) c.addi  a2, -4
core   0: 3 0x0000000080004040 (0x1671) x12 0x0000000080000020
core   0: 0x0000000080004042 (0x0000e21c) c.sd    a5, 0(a2)
core   0: 3 0x0000000080004042 (0xe21c) mem 0x0000000080000020 0x000000000002cbc8
core   0: 0x0000000080004044 (0x000607b3) add     a5, a2, zero
core   0: 3 0x0000000080004044 (0x000607b3) x15 0x0000000080000020
core   0: 0x0000000080004048 (0xf0078793) addi    a5, a5, -256
core   0: 3 0x0000000080004048 (0xf0078793) x15 0x000000007fffff20
core   0: 0x000000008000404c (0x0017b423) sd      ra, 8(a5)
core   0: exception trap_store_access_fault, epc 0x000000008000404c
core   0:           tval 0x000000007fffff28
```
Tracer log
```log
rv rf w 0 0 32 0000000008000000
rv commit 0 00000000000009d6
rv load flu 0
rv trap 0 0 15 183248
rv commit 0 0000000080004000
rv rf w 0 0 32 0000000080000020
rv commit 0 0000000080004040
rv store com 0 4 8 0000000080000020 000000000002cbc8
rv commit 0 0000000080004042
rv rf w 0 0 32 0000000080000020
rv commit 0 0000000080004044
rv rf w 0 0 32 000000007fffff20
rv commit 0 0000000080004048
rv store com 0 5 8 000000007fffff28 00000000000002f0
rv commit 0 000000008000404c
```
**Everything you need to debug or reproduce the execution is in the attached file**:
[debug_riscv-dv2.zip](https://github.com/user-attachments/files/16033797/debug_riscv-dv2.zip)

Dolu1990, 2024-07-01:
Hi,

i think this is the combination of a few things : 
- By design, NaxRiscv doesn't trap on cached memory region access fault (only on io access fault)
- NaxRiscv is by default only consider that as a io region : https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/Gen.scala#L480
which isn't in the 7fffff28 range
- rvls is only notified of those memory region : 
rv region add 0 0 0000000080000000 0000000080000000
rv region add 0 1 0000000010010000 0000000000010000
rv region add 0 1 0000000010c00000 0000000000400000
rv region add 0 1 0000000010000000 0000000000001000
rv region add 0 1 0000000014000000 0000000000001000

Maybe the best would be to provide NaxRiscv generation with instead of 
ioRange = a => a(31 downto 28) === 0x1
having 
ioRange = a => !a(31)

That way only rv region add 0 0 0000000080000000 0000000080000000 would be considered as cached by Nax
and everything else would be sensitive to access faults. And so should be in sync with rvls.

Bill94l, 2024-07-01:
> ioRange = a => !a(31)

Once this change has been made, the test becomes infinite: it loops between "trap_store_access_fault" and "trap_store_address_misaligned"

```Shel
[info] Sim starting <3
^C
[warn] Canceling execution...
```
Spike log
```log
core   0: 0x0000000080004000 (0x0400006f) j       pc + 0x40
core   0: 3 0x0000000080004000 (0x0400006f)
core   0: 0x0000000080004040 (0x00001671) c.addi  a2, -4
core   0: 3 0x0000000080004040 (0x1671) x12 0x0000000080000004
core   0: 0x0000000080004042 (0x0000e21c) c.sd    a5, 0(a2)
core   0: exception trap_store_address_misaligned, epc 0x0000000080004042
core   0:           tval 0x0000000080000004
core   0: 0x0000000080004000 (0x0400006f) j       pc + 0x40
core   0: 3 0x0000000080004000 (0x0400006f)
core   0: 0x0000000080004040 (0x00001671) c.addi  a2, -4
core   0: 3 0x0000000080004040 (0x1671) x12 0x0000000080000000
core   0: 0x0000000080004042 (0x0000e21c) c.sd    a5, 0(a2)
core   0: 3 0x0000000080004042 (0xe21c) mem 0x0000000080000000 0x000000007fffff08
core   0: 0x0000000080004044 (0x000607b3) add     a5, a2, zero
core   0: 3 0x0000000080004044 (0x000607b3) x15 0x0000000080000000
core   0: 0x0000000080004048 (0xf0078793) addi    a5, a5, -256
core   0: 3 0x0000000080004048 (0xf0078793) x15 0x000000007fffff00
core   0: 0x000000008000404c (0x0017b423) sd      ra, 8(a5)
core   0: exception trap_store_access_fault, epc 0x000000008000404c
core   0:           tval 0x000000007fffff08
core   0: 0x0000000080004000 (0x0400006f) j       pc + 0x40
core   0: 3 0x0000000080004000 (0x0400006f)
core   0: 0x0000000080004040 (0x00001671) c.addi  a2, -4
core   0: 3 0x0000000080004040 (0x1671) x12 0x000000007ffffffc
core   0: 0x0000000080004042 (0x0000e21c) c.sd    a5, 0(a2)
core   0: exception trap_store_address_misaligned, epc 0x0000000080004042
core   0:           tval 0x000000007ffffffc
core   0: 0x0000000080004000 (0x0400006f) j       pc + 0x40
core   0: 3 0x0000000080004000 (0x0400006f)
core   0: 0x0000000080004040 (0x00001671) c.addi  a2, -4
core   0: 3 0x0000000080004040 (0x1671) x12 0x000000007ffffff8
core   0: 0x0000000080004042 (0x0000e21c) c.sd    a5, 0(a2)
core   0: exception trap_store_access_fault, epc 0x0000000080004042
core   0:           tval 0x000000007ffffff8
core   0: 0x0000000080004000 (0x0400006f) j       pc + 0x40
core   0: 3 0x0000000080004000 (0x0400006f)
core   0: 0x0000000080004040 (0x00001671) c.addi  a2, -4
core   0: 3 0x0000000080004040 (0x1671) x12 0x000000007ffffff4
core   0: 0x0000000080004042 (0x0000e21c) c.sd    a5, 0(a2)
core   0: exception trap_store_address_misaligned, epc 0x0000000080004042
core   0:           tval 0x000000007ffffff4
core   0: 0x0000000080004000 (0x0400006f) j       pc + 0x40
core   0: 3 0x0000000080004000 (0x0400006f)
core   0: 0x0000000080004040 (0x00001671) c.addi  a2, -4
core   0: 3 0x0000000080004040 (0x1671) x12 0x000000007ffffff0
core   0: 0x0000000080004042 (0x0000e21c) c.sd    a5, 0(a2)
core   0: exception trap_store_access_fault, epc 0x0000000080004042
core   0:           tval 0x000000007ffffff0
core   0: 0x0000000080004000 (0x0400006f) j       pc + 0x40
core   0: 3 0x0000000080004000 (0x0400006f)
core   0: 0x0000000080004040 (0x00001671) c.addi  a2, -4
core   0: 3 0x0000000080004040 (0x1671) x12 0x000000007fffffec
core   0: 0x0000000080004042 (0x0000e21c) c.sd    a5, 0(a2)
core   0: exception trap_store_address_misaligned, epc 0x0000000080004042
core   0:           tval 0x000000007fffffec
```

Dolu1990, 2024-07-03:
> Once this change has been made, the test becomes infinite: it loops between "trap_store_access_fault" and "trap_store_address_misaligned"

As spike do agree with that behaviour, i would say that it is either OK, either that is the riscv-dv configuration/generation which need tweeks to avoid this.

> Separate issue: Does nax support unaligned instructions?

No it doesn't, when running linux, opensbi emulate them in machine mode.

Bill94l, 2024-07-03:
In this case :
you are going to make a commit to modify `ioRange = a => a(31 downto 28) === 0x1`  by  `ioRange = a => !a(31)`  in the following files or it is not necessary ?

https://github.com/SpinalHDL/NaxRiscv/blob/bc18f667c11ed05ddbc248dc90e255d4c16798ad/src/main/scala/naxriscv/Gen.scala#L43

https://github.com/SpinalHDL/NaxRiscv/blob/bc18f667c11ed05ddbc248dc90e255d4c16798ad/src/main/scala/naxriscv/Gen.scala#L480

https://github.com/SpinalHDL/NaxRiscv/blob/bc18f667c11ed05ddbc248dc90e255d4c16798ad/src/main/scala/naxriscv/platform/TilelinkNaxRiscvFiber.scala#L41

https://github.com/SpinalHDL/NaxRiscv/blob/bc18f667c11ed05ddbc248dc90e255d4c16798ad/src/main/scala/naxriscv/platform/asic/NaxAsicGen.scala#L61

closed 2024-10-15 by Bill94l