SpinalHDL/NaxRiscv #112: WFI instruction

Bill94l, 2024-07-01:
Hi,

The execution of the WFI instruction causes a "trap_illegal_instruction" on the spike, but not on the DUT ?

Here is the execution result

```shell
[info] *** DUT missed a trap ***
[info] commit error
[info] - std::exception
```

Dump
```dump
    80000340:	03141793          	slli	a5,s0,0x31
    80000344:	91a2                	add	gp,gp,s0
    80000346:	10500073          	wfi
```

Spike log
```log
core   0: 0x0000000000000340 (0x03141793) slli    a5, s0, 49
core   0: 0 0x0000000000000340 (0x03141793) x15 0x03a4000000000000
core   0: 0x0000000000000344 (0x000091a2) c.add   gp, s0
core   0: 0 0x0000000000000344 (0x91a2) x 3 0x00000000000001d2
core   0: 0x0000000000000346 (0x10500073) wfi
core   0: exception trap_illegal_instruction, epc 0x0000000000000346
core   0:           tval 0x0000000010500073
```

Tracer log
```log
rv rf w 0 0 32 03a4000000000000
rv commit 0 0000000000000340
rv rf w 0 0 32 00000000000001d2
rv commit 0 0000000000000344
rv commit 0 0000000000000346
```

Dolu1990, 2024-07-03:
> The execution of the WFI instruction causes a "trap_illegal_instruction" on the spike, but not on the DUT ?

Ahhh likely this is because of the mstatus.TW field.

On NaxRiscv it is implemented as "doing nothing" I will take a look how hard it would be to implement it as the spec says

Dolu1990, 2024-07-03:
implemented it as spike does, should be good now. WFI will always trap in user mode, and trap in supervisor mode if mstatus tw is 1

Bill94l, 2024-07-03:
In the last update, why did you remove `|| priv.User()`? In this case, there will be no trap in the user mode?

Dolu1990, 2024-07-03:
WFI && !(priv.isMachine() || !priv.logic.machine.mstatus.tw && (Bool(!priv.p.withSupervisor) || priv.isSupervisor()))

=> 

Dont trap if : 
- priv.isMachine()
- !tw && (Supervisor not implemented || inSupervisor)

which mean that when in user mode, it will trap when : 
- tw
- !tw && supervisor not implemented

right ?

Bill94l, 2024-07-03:
In any case, with both modifications I still have the same execution error. I will give you everything you need to be able to reproduce the execution of the two codes that fail

[basic_test_1_wfi_missedTrap.zip](https://github.com/user-attachments/files/16087689/basic_test_1_wfi_missedTrap.zip)
[basic_test_2_wfi_missedTrap.zip](https://github.com/user-attachments/files/16087693/basic_test_2_wfi_missedTrap.zip)

In these files you will have:  `test.S`  `test.bin`  `test.dump`  `test.o`  `spike.log`  `tracer.log`  `wave.fst`

Dolu1990, 2024-07-04:
There is something weird in the basic_test_2 fst file.
I can only see the WFI signal in the fetch_0 stage, which mean it isn't used in the execute pipeline.

When i generate the hardware myself, i get other signals as "EU0_ExecutionUnitBase_pipeline_execute_0_EnvCallPlugin_WFI"

Are you sure you got all the upstream changes (main branch) ?

Bill94l, 2024-07-04:
I am up to date with the main branch and have added your latest modification to `NaxRiscv/src/main/scala/naxriscv/execute/EnvCallPlugin.scala`

```scala
WFI && !(priv.isMachine() || !priv.logic.machine.mstatus.tw && (Bool(!priv.p.withSupervisor) || priv.isSupervisor()))
```
I might have shared an old `wave.fst` file for the `basic_test_2` previously. Please find attached the new `wave.fst` file that I have just generated
[wave.zip](https://github.com/user-attachments/files/16096246/wave.zip)
 
Does the execution of the basic_test_1 and basic_test_2 programs work correctly for you?

Dolu1990, 2024-07-04:
> Does the execution of the basic_test_1 and basic_test_2 programs work correctly for you?

I didn't tried, as i don't know exactly what configuration you use. Also i try to avoid trying to reproduce the problem localy if it is possible for me to get the wave. Reasons is things can easily get out of sync and be hard to reproduce.

For some reasons, 
    if (priv.p.withUser) when(WFI) {
      trap := True
    }
Isn't triggered in your wave, even when WFI is high.

Can you send me your verilog ?

Bill94l, 2024-07-04:
Hi, 

Sorry, there was an error in the update on my part. I forgot to include this piece of code:

```Scala
if (priv.p.withUser) when(WFI) {
  trap := True
}
```

Now the codes (basic test 1 & basic test_2) work correctly on my side with SocSim

Thanks :D

closed 2024-07-04 by Bill94l

Dolu1990, 2024-07-04:
Cool ^^

commit 31e9db6e (2024-07-03): Fix #112 mstatus.tw
commit 6f736929 (2024-07-03): Fix #112 mstatus.tw
commit bc18f667 (2024-07-03): Fix #112 mstatus.tw