SpinalHDL/NaxRiscv #113: How to integrate my own IP

phillippZZ, 2024-07-02:
Hi, Dolu.

I'm using litex with your naxriscv and trying to figure out how I can integrate my IP with your code. My ip uses axi4 and I don't know how litex integrates tilelink with axi4. Could you please give me a simple example?

Plus, I saw some code defined in "https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/naxriscv/core.py" as following:
> class NaxRiscv(CPU):
    category             = "softcore"
    family               = "riscv"
    name                 = "naxriscv"
    ...
    **scala_files      = ["gen.scala"]**

Does it mean when litex generates bitstream, it downloads the source code of "https://github.com/SpinalHDL/NaxRiscv" and uses "gen.scala"? If so, where and how can I add my IP?

Dolu1990, 2024-07-03:
Hi,

I would say it realy depend what SoC you want to start with, if you want to use litex based SoC or full SpinalHDL one ?

phillippZZ, 2024-07-03:
I want to use litex :)

enjoy-digital, 2024-07-03:
For the integration with LiteX, you can have a look at: https://github.com/enjoy-digital/litex/wiki/Reuse-a-(System)Verilog,-VHDL,-Amaranth,-Spinal-HDL,-Chisel-core

And also this repository that integrate different AXI cores: https://github.com/enjoy-digital/litex_verilog_axi_test

This should allow you to connect your AXI components to the main LiteX bus and communicate with it from NaxRiscv CPU.

closed 2024-07-03 by phillippZZ