SpinalHDL/NaxRiscv #118: Need help with the infrastructure regarding interconnection

phillippZZ, 2024-07-16:
Hi, Dolu.

I'm looking for the infrastructure of naxriscv regarding interconnection but can't find any on "https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/introduction/index.html". Could you tell me how multiple cpus, l1 cache, l2 cache, main memory, IOs are connected to each other? For example, what protocal do they use, is it tilelink or Axi4? Are both l1 and l2 cache coherent? And any other information regarding interconnection would help.

Thank you!

atkarim, 2024-07-17:
Hi @phillippZZ 

As far as I know, there are 2 architectures, 1 for simulation with SocDemo.scala, 1 for fpga emulation that connect to Litex in NaxSoc.scala.

The 2 architectures are almost similar, but there are some bridges in NaxSoc to connect to Litex, as it uses AXI.

### **SocDemo architecture**
[image]

### **NaxSoc architecture** 
[image]

L1 and L2 are coherent, that's why they use Tilelink Coherent bus to communicate.
Tilelink-C is only present between L1 and L2. Between Fetch cache and L2, there is a simple Tilelink bus.

[image]

phillippZZ, 2024-07-17:
@atkarim Wow!!!
That's a lot of work and thank you so much!

@Dolu1990 @atkarim BTW, is it possobile that I add some kind of bridge provided by litex or naxriscv so that an IP which uses AXI can be itegrated and impact the cache or memory. My friend has such an IP and sorry I don't know the details yet. 

All I know is that it's based on an algorithm that could improve the speed of reading from cache/memory with multiple cores.

Dolu1990, 2024-07-17:
The Nax SoC for litex does this already i think : 
 ```scala
    val dma = c.withDma generate new Area {
      val bus = slave(
        Axi4(
          Axi4Config(
            addressWidth = 32,
            dataWidth = mainDataWidth,
            idWidth = 4
          )
        )
      )

      val bridge = new Axi4ToTilelinkFiber(64, 4)
      bridge.up load bus.pipelined(ar = StreamPipe.HALF, aw = StreamPipe.HALF, w = StreamPipe.FULL, b = StreamPipe.HALF, r = StreamPipe.FULL)
      bridge.down.setDownConnection(a = StreamPipe.FULL)
      memFilter.up << bridge.down

      //As litex reset will release before our one, we need to ensure that we don't eat a transaction
      Fiber build {
        bridge.read.get
        bridge.write.get
        when(ClockDomain.current.isResetActive){
          bus.ar.ready := False
          bus.aw.ready := False
          bus.w.ready := False
        }
      }
    }
```

That dma bus will be coherent with all caches

Dolu1990, 2024-07-17:
If you want to get good perforance on that AXI to tilelink bridge : 
- Do not sent request which use the same AXI ID than another request which is still inflight
- Avoid AXI burst that are more than 64 bytes
- Avoid if possible splitting memory bursts into less than 64 bytes

64 byte for everything is you can ^^

closed 2024-07-17 by phillippZZ