SpinalHDL/NaxRiscv #123: Is there any implementation of FENCE instructions?

zyn810039594, 2024-09-23:
Hi!

I'm testing somethine about memory access, and I want to keep the order between two store/load instruction. I used "fence rw,rw" to do that, but nothing happened.

After checking the file, FENCE instruction has been found in EnvCallPlugin, 

https://github.com/SpinalHDL/NaxRiscv/blob/main/src/main/scala/naxriscv/execute/EnvCallPlugin.scala#60

but I've not found any implementation of it. Is this feature been supported? Or it's just done in another module?

Thanks for your help!

Dolu1990, 2024-09-23:
Hi,

Fence are only "properly" implemented if memory coherency is enabled. It is done in the Lsu2Plugin : 
https://github.com/SpinalHDL/NaxRiscv/blob/1c50e84d9a6f7ea93d7153f12906c25552267b9d/src/main/scala/naxriscv/lsu2/Lsu2Plugin.scala#L1857

It will add some logic to hold instruction just before the issue queue (dispatcher)

zyn810039594, 2024-09-26:
Ok, i got it! thanks!

closed 2024-09-26 by zyn810039594