SpinalHDL/NaxRiscv #126: Improvements: Updating Spike and RVLS Submodules

Bill94l, 2024-11-15:
Hi @Dolu1990 ,

Would you like to update the `riscv-isa-sim` and `rvls` submodules to the latest commit hash?

I tested this update on my side, and it works well. Could you please let me know the steps to follow to create a pull request for your review and validation?

Looking forward to your feedback, thank you!

Dolu1990, 2024-11-15:
Hi,
Sure ^^
Hmm so to get it through smoothly, the way to do is : 
- Create the PR in NaxRiscv with all the required changes
  - If a submodule need to merge a PR first, indicate it.
- Wait that the github action success
- Then on my side i will run some additional regression with randomized NaxRiscv configurations
- Then it should be good for merge

The one thing which is important, is that that NaxRiscv PR has all the info, so that in some way, the maintener (me) can be next to amnesic and still be able to get the context. (Mainteners have to do a lot of switching context all the day, it break brains  cells XDXD)

Dolu1990, 2024-11-15:
Also note, you have to poke mainteners when things get lost.

Bill94l, 2024-11-29:
Hi @Dolu1990,

Thank you for the detailed steps.

To proceed smoothly :

1. We first need to validate the pull requests for the submodules:
   - **SPIKE** https://github.com/SpinalHDL/riscv-isa-sim/pull/2#issuecomment-2506607012
   - **RVLS** https://github.com/SpinalHDL/rvls/pull/1#issue-2588700654

2. After the submodule updates are validated, we can move on to the `NaxRiscv` pull request.
For this step, `NaxRiscv` will need to point to the updated submodules hashe. Could you clarify if this update should be handled on my side or yours? how can i change it ?
    - **NAXRISCV** https://github.com/SpinalHDL/NaxRiscv/pull/125#issue-2588582034

3. Commit updates for FPU testing in `RVLS` and `NaxRiscv`:
After validation, I will make the necessary changes to properly support FPU testing using `socSim`.

4. Complete the `addr_to_mem` function in `RVLS`:
This function is essential for enabling the `lrsc` test and Linux boot functionality with `socSim` and the updated `riscv-isa-sim`. Here is the current implementation I tried:
    ```CPP
    // should return NULL for MMIO addresses
    char* SpikeIf::addr_to_mem(reg_t addr)  {
        auto r = getRegion(addr);
        if (r == NULL || r->type != RegionType::mem) {
            return NULL;
        }
        return (char*)memory->memory.get(addr);
    }
    ```
    Unfortunately, it doesn't work. If you have any suggestions for improvement, I’d appreciate it.
    For reference:
    - Definition in `RVLS`: https://github.com/Bill94l/rvls/blob/b5c15e129f8168c2317b2c129aef91adf935bfeb/src/hart.cpp#L41C1-L46C2

    - Usage in `riscv-isa-sim`: https://github.com/Bill94l/spinal-riscv-isa-sim/blob/cbc35ae812cdca97380417575df0cb16716ec96d/riscv/mmu.h#L299C3-L311C4

    - Definition in `riscv-isa-sim`: https://github.com/Bill94l/spinal-riscv-isa-sim/blob/cbc35ae812cdca97380417575df0cb16716ec96d/riscv/simif.h#L17C1-L19C70

Looking forward to your feedback and validation to move forward! 😊

Thanks :D

Dolu1990, 2024-11-29:
Hi ^^

> 1) We first need to validate the pull requests for the submodules:

What's about having NaxRiscv branchs in rvls / riscv-isa-sim ? This way it won't interfer too much with other projects ?

> 2) After the submodule updates are validated, we can move on to the NaxRiscv pull request.
For this step, NaxRiscv will need to point to the updated submodules hashe. Could you clarify if this update should be handled on my side or yours? how can i change it ?

 I use smartgit to handle git submodule, not sure exactly how to handle that in command line.
I can do it if you want to ^^

> Complete the addr_to_mem function in RVLS:

Ahhh there the idea is to specify more precisely which region of the memory i capable of what, right ?
I think addr_to_mem should always return null, to force spike checking every address.
Instead, checks should be added in mmio_xxx. (?)

Bill94l, 2024-11-29:
Hi
> What's about having NaxRiscv branchs in rvls / riscv-isa-sim ? This way it won't interfer too much with other projects ?

That's a great idea! However, in that case, I would need to create a new pull request for the `NaxRiscv` branch that you plan to create. Could you confirm if this is what you'd like me to do?

> I use smartgit to handle git submodule, not sure exactly how to handle that in command line.

If you want to create a NaxRiscv branch in the riscv-isa-sim and rvls submodules. For each submodule after cloning the repository, we need to use:
```bash
git checkout NaxRiscv
```

>  I think addr_to_mem should always return null, to force spike checking every address.

If `addr_to_mem` always returns `NULL`, here’s the relevant implementation for `RVLS` in `ext/rvls/src/hart.cpp`:

**ext/rvls/src/hart.cpp**
  ```CPP
  // should return NULL for MMIO addresses
  char* SpikeIf::addr_to_mem(reg_t addr)  {
      //if((addr & 0xE0000000) == 0x00000000) return NULL;
      //printf("addr_to_mem %lx ", addr);
      //return (char*) memory->memory.get(addr);
      return NULL;
  }
  ```
To make the `lrsc` test and Linux boot work correctly with `RVLS`, I needed to modify the `riscv-isa-sim` source code as follows:

```diff
--- 'a/riscv/mmu.h'
+++ b/riscv/mmu.h
@@ -304,7 +304,7 @@ public:
     }
 
     reg_t paddr = translate(generate_access_info(vaddr, STORE, {}), 1);
-    if (sim->reservable(paddr))
+    if (!sim->reservable(paddr))
       return load_reservation_address == paddr;
     else
       throw trap_store_access_fault((proc) ? proc->state.v : false, vaddr, 0, 0);
```

However, when I apply this modification to `riscv-isa-sim`, the `lrsc` test and Linux boot fail to work properly with the regression test simulator, as its `addr_to_mem` does not always return `NULL`. Here’s the regression test simulator's `addr_to_mem` implementation in `src/test/cpp/naxriscv/src/main.cpp`:
**src/test/cpp/naxriscv/src/main.cpp**
```CPP
// should return NULL for MMIO addresses
    virtual char* addr_to_mem(reg_t addr)  {
        if((addr & 0xE0000000) == 0x00000000) return NULL;
        return (char*)memory.get(addr);
    }
```
I’d like to find a solution that works both for RVLS and the regression test simulator.
Let me know if you have any suggestions or if further clarification is needed! 😊

Dolu1990, 2024-11-29:
I created the rvls / riscv-isa-sim branches : 
- https://github.com/SpinalHDL/riscv-isa-sim/tree/naxriscv
- https://github.com/SpinalHDL/rvls/tree/naxriscv

They should be on the proper git hash for you to open a PR :D

Bill94l, 2024-12-02:
Hi @Dolu1990 

Thank you for setting up the branches!

    - For `riscv-isa-sim`, I would prefer the branch to be created from commit `5f4eca2c0a5ee7808ad235f5230f3bd2c14a6ffa` (the second-to-last commit).
    For `RVLS`, the branch should be created from the latest commit `b5c15e129f8168c2317b2c129aef91adf935bfeb`.

Thanks :D

Bill94l, 2024-12-02:
The `--enable-commitlog` flag is no longer required when building the updated `riscv-isa-sim` :
```bash
cd $NAXRISCV/ext/riscv-isa-sim
mkdir build
cd build
(---) ../configure --prefix=$RISCV --enable-commitlog  --without-boost --without-boost-asio --without-boost-regex
(+++) ../configure --prefix=$RISCV --without-boost --without-boost-asio --without-boost-regex
make -j$(nproc)
```
**Proposal:**
Would it make sense to create a new branch named `naxriscv` in this repository?
The idea is for this branch to point to the `naxriscv` branches of both submodules, `riscv-isa-sim` and `rvls`. 

What do you think?

Dolu1990, 2024-12-02:
> Hi @Dolu1990
>
> Thank you for setting up the branches!
>
> * For `riscv-isa-sim`, I would prefer the branch to be created from commit `5f4eca2c0a5ee7808ad235f5230f3bd2c14a6ffa` (the second-to-last [...]

Ahhh damned, i have seen the PR before this message, the PR already got merged :/

Is it ok ?

> Proposal:
Would it make sense to create a new branch named naxriscv in this repository?
The idea is for this branch to point to the naxriscv branches of both submodules, riscv-isa-sim and rvls.

Hmmm, let's name it "rvls-update" instead ?

closed 2024-12-09 by Bill94l