SpinalHDL/NaxRiscv #133: #include <cstdint> in my device.h

Rishik-Y, 2025-02-26:
Hello!
I am trying to be able to run this on my NixOS
And i dont seem to understand something and i would like to ask on this if you dont mind
I seem to be needing to keep 
#include <cstdint> in my device.h for the make to work
and it doesnt compile unless we keep that
so why havent we pushed this pull request yet?
https://github.com/SpinalHDL/riscv-isa-sim/pull/1

I mean i havent glanced alot of the code yet + i am still very much of a newbie so i may be missing something here.

Edit: Just checked the [main](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/fesvr/device.h) repository, even it has `#include <cstdint>` in it

Dolu1990, 2025-02-27:
Hi,

> so why havent we pushed this pull request yet?

I lost it in my emails XD

I merged the PR and update the NaxRiscv submodule.
Let's me know if the problem persist.

Regards
Charles

Rishik-Y, 2025-02-27:
Hi,
I actually am quite surprised and its a first for me,
To hear that not only there was no actual reason but also that nobody actually realised since past 2 years!! XD

Welp the problem is fixed thx
Closing this issue as its completed! :)

closed 2025-02-27 by Rishik-Y