SpinalHDL/NaxRiscv #134: Make failing due to git checkout v4.216 from Readme.md

Rishik-Y, 2025-02-26:
There is an Issue in being able to run the program where keeping 
git checkout [v4.216](https://github.com/SpinalHDL/NaxRiscv/blob/main/src/test/cpp/naxriscv/README.md) will result in giving an error during make
which can simply be fixed by keeping `#include <memory> ` in `verilator/src/V3Const.cpp`
Looking at the verilator master branch, it has already been fixed.
It would be better if someone either changes the [Readme.md](https://github.com/SpinalHDL/NaxRiscv/blob/main/src/test/cpp/naxriscv/README.md) keeping another checkout version
or assuming some conflicts arise changing the branch, we can simply fork it and add `#include <memory> ` in `verilator/src/V3Const.cpp`

Dolu1990, 2025-02-27:
Hi,

Hmmm is the fix contained in verilator [v4.228](https://github.com/verilator/verilator/releases/tag/v4.228) ?

Rishik-Y, 2025-02-27:
Hello, 👋
The fix seems to be there since [v4.222](https://github.com/verilator/verilator/releases/tag/v4.222) where they included `#include <memory>` to fix the error as well.
However cant say whether it causes some other error on not.

I just tested v4.228 and till this step 
`sbt "runMain naxriscv.Gen"`
And no issues have occured by any sense including fixing the error

But please dont take it as final say,
since i am testing this on NixOS, 
where i cant seem to go beyond last step
`make compile`
and still trying to figure things out on how to package it.

Though I do think think it will not cause any issue. 
So i may create a pull request, leaving up to you to decide.

Dolu1990, 2025-02-28:
All merged, thanks :)

closed 2025-02-28 by Dolu1990