SpinalHDL/NaxRiscv #140: Error while doing sbt "runMain naxriscv.Gen"

Rishik-Y, 2025-03-12:
Hi @Dolu1990,
i tried testing the program after the new merge,
and it seems i am getting error directly here
[sbt "runMain naxriscv.Gen"](https://github.com/SpinalHDL/NaxRiscv/blob/main/src/test/cpp/naxriscv/README.md)

Where i get this error:

```
sbt "runMain naxriscv.Gen"
[info] welcome to sbt 1.6.0 (Oracle Corporation Java 11.0.25)
[info] loading settings for project naxriscv-build from plugins.sbt ...
[info] loading project definition from /home/Kihsir/Downloads/Git/Scala/NaxRiscv/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project spinalhdl-build from plugin.sbt ...
[info] loading project definition from /home/Kihsir/Downloads/Git/Scala/NaxRiscv/ext/SpinalHDL/project
[info] loading settings for project all from build.sbt ...
[info] set current project to NaxRiscv (in build file:/home/Kihsir/Downloads/Git/Scala/NaxRiscv/)
[info] compiling 127 Scala sources and 3 Java sources to /home/Kihsir/Downloads/Git/Scala/NaxRiscv/target/scala-2.12/classes ...
[error] /home/Kihsir/Downloads/Git/Scala/NaxRiscv/src/main/scala/naxriscv/platform/Tracer.scala:175:62: too many arguments (2) for method newContext: (workspace: String)Long
[error]   val handle = Frontend.newContext(workspace.getAbsolutePath,spikeLogFileOut)
[error]                                                              ^
[error] /home/Kihsir/Downloads/Git/Scala/NaxRiscv/src/main/scala/naxriscv/platform/Tracer.scala:193:67: value loadU32 is not a member of object rvls.jni.Frontend
[error]   override def loadU32(address: Long, data: Int): Unit = Frontend.loadU32(handle, address, data)
[error]                                                                   ^
[error] /home/Kihsir/Downloads/Git/Scala/NaxRiscv/src/main/scala/naxriscv/platform/Tracer.scala:198:142: too many arguments (5) for method trap: (handle: Long, hartId: Int, interrupt: Boolean, code: Int)Boolean
[error]   override def trap(hartId: Int, interrupt: Boolean, code: Int, fault_addr: Long): Unit = if(!Frontend.trap(handle, hartId, interrupt, code, fault_addr)) throw new Exception()
[error]                                                                                                                                              ^
[error] /home/Kihsir/Downloads/Git/Scala/NaxRiscv/src/main/scala/naxriscv/platform/tilelinkdemo/SocSim.scala:133:6: value wavePath is not a member of spinal.core.sim.SpinalSimConfig
[error]   sc.wavePath(s"waves/${workspaceOutputSubDir}")
[error]      ^
[error] /home/Kihsir/Downloads/Git/Scala/NaxRiscv/src/main/scala/naxriscv/platform/tilelinkdemo/SocSim.scala:201:34: overloaded method value withCb with alternatives:
[error]   [T <: spinal.core.Component](compiled: spinal.core.sim.SimCompiled[T], window: Long, seed: Int)(testbench: (T, (=> Unit) => Unit) => Unit)Unit <and>
[error]   [T <: spinal.core.Component](compiled: spinal.core.sim.SimCompiled[T], window: Long, seed: Int, dualSimEnable: Boolean)(testbench: (T, (=> Unit) => Unit) => Unit)Unit
[error]  cannot be applied to (spinal.core.sim.SimCompiled[naxriscv.platform.tilelinkdemo.SocSim.SocDemoSim], window: Int, seed: Int, String, runBuildroot: Boolean)
[error]       case true => DualSimTracer.withCb(compiled, window = 50000 * 10, seed = 2, file, runBuildroot = false)(testIt)
[error]                                  ^
[error] /home/Kihsir/Downloads/Git/Scala/NaxRiscv/src/main/scala/naxriscv/platform/tilelinkdemo/SocSim.scala:201:110: missing argument list for method testIt in object SocSim
[error] Unapplied methods are only converted to functions when a function type is expected.
[error] You can make this conversion explicit by writing `testIt _` or `testIt(_,_,_,_)` instead of `testIt`.
[error]       case true => DualSimTracer.withCb(compiled, window = 50000 * 10, seed = 2, file, runBuildroot = false)(testIt)
[error]                                                                                                              ^
[error] /home/Kihsir/Downloads/Git/Scala/NaxRiscv/src/main/scala/naxriscv/platform/tilelinkdemo/SocSim.scala:211:34: overloaded method value withCb with alternatives:
[error]   [T <: spinal.core.Component](compiled: spinal.core.sim.SimCompiled[T], window: Long, seed: Int)(testbench: (T, (=> Unit) => Unit) => Unit)Unit <and>
[error]   [T <: spinal.core.Component](compiled: spinal.core.sim.SimCompiled[T], window: Long, seed: Int, dualSimEnable: Boolean)(testbench: (T, (=> Unit) => Unit) => Unit)Unit
[error]  cannot be applied to (spinal.core.sim.SimCompiled[naxriscv.platform.tilelinkdemo.SocSim.SocDemoSim], window: Int, seed: Int, String, runBuildroot: Boolean)
[error]       case true => DualSimTracer.withCb(compiled, window = 50000 * 10, seed = 2, file, runBuildroot = true)(testIt)
[error]                                  ^
[error] /home/Kihsir/Downloads/Git/Scala/NaxRiscv/src/main/scala/naxriscv/platform/tilelinkdemo/SocSim.scala:211:109: missing argument list for method testIt in object SocSim
[error] Unapplied methods are only converted to functions when a function type is expected.
[error] You can make this conversion explicit by writing `testIt _` or `testIt(_,_,_,_)` instead of `testIt`.
[error]       case true => DualSimTracer.withCb(compiled, window = 50000 * 10, seed = 2, file, runBuildroot = true)(testIt)
[error]                                                                                                             ^
[error] 8 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 14 s, completed Mar 11, 2025, 12:51:30 PM
```

It could very well be a me issue but looking at the error seems possibly some code issue?

Dolu1990, 2025-03-14:
@Bill94l Hi,

I don't know how, but apparently, the automated regressions didn't cached issues ?
Seems like the RVLS being used when cloning the master branch isn't in sync ?

I have the same issue when cloning a fresh NaxRiscv main branch.

Bill94l, 2025-03-14:
Hi @Dolu1990, @Rishik-Y,

The error you are encountering is due to the fact that you haven’t run the **make install** command. Without this step, some essential patches are not applied:
- **RVLS patch** (files *rvls.diff* and *rvls.patch*)

- **Patch for the SpinalHDL library** (file adding_wavePath_simConfig.patch)

Follow the instructions available in the **src/test/README** file, to properly install and avoid this error :

```bash
# Clone repository
git clone https://github.com/SpinalHDL/NaxRiscv.git
cd NaxRiscv
export NAXRISCV=$(pwd)  # Set environment variable

# Full installation (toolchain + core + simulators)
make install
sbt "runMain naxriscv.Gen"
```

@Dolu1990 If you confirm that the **RVLS and SpinalHDL patches** are validated on your side, I can then submit a pull-request to integrate these changes and remove patches. I will also update the README and the RVLS hash, although that is not directly related to the error you are experiencing.

Have a nice weekend ^^

Dolu1990, 2025-03-14:
Ahhhhh, hmmm, we could merge those rvls patch into a proper rvls branch ?

Bill94l, 2025-03-14:
Hello,

The patches are already available in your current **main** branch, so there is no need to merge them.

[image]

If you prefer that we remove the patches, I can directly integrate the patch code into RVLS and SpinalHDL, if that works for you.

Rishik-Y, 2025-03-14:
> Follow the instructions available in the **src/test/README** file, to properly install and avoid this error :

Hi @Bill94l,
Thanks so much for your answer.
I actually didnt knew about this 😓
I was following the readme from here **src/test/cpp/naxriscv/README.md**

Which was written over [here](https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/introduction/index.html#how-to-use)
In its documentation page.

But thanks for correcting me,
will make sure to check it out by running `make install` later.

Rishik-Y, 2025-03-14:
Hi @Dolu1990 @Bill94l
FIrst of Thanks very much for your answers!

I am bit confused? 
Sorry about it
But why is there two different readme over here?

> Follow the instructions available in the src/test/README file, to properly install and avoid this error :

I understand i probably would have to follow this code

But what is this [readme](https://github.com/SpinalHDL/NaxRiscv/blob/main/src/test/cpp/naxriscv/README.md) exactly for?

As in the documentation redirects to [this](https://spinalhdl.github.io/NaxRiscv-Rtd/main/NaxRiscv/introduction/index.html#how-to-use)

Also as @Bill94l, it is indeed most likely to be make install issue..
I am still trying to figure out how exactly i could run it in NixOS

Dolu1990, 2025-03-14:
@Bill94l I mean to merge those files into the real repository ^^
having patch files may increase maintenance overhead in the future, as it is very lose.