SpinalHDL/NaxRiscv #142: Question on the CLINT

bantierr, 2025-07-10:
Hi again, 
Sorry raise a new issue so soon. I am having trouble getting the CLINT to work as expected. I can generate IPIs with no problem, but I somehow cannot clear the MIP register. Here is the behavior I am observing:
Core 1 send an IPI to core 0:
```
rv io 1 1 0000000010010000 8004567580045675 0f 4 0
rv commit 1 00000000800403a0   <----------------------------------------- IPI sent here
rv store com 1 15 1 0000000080180018 0000000000000000
rv commit 1 00000000800403a4
rv trap 0 1 3 0000000001400063   <----------------------------------------- received here
```
A few instructions are executed on each core (core 0 disabled mstatus.MIE), then core 0 tries to clear the interrupt:
```
rv io 0 1 0000000010010000 0000000000000000 01 1 1    <- Attempt to write 0 to MIP
rv load flu 0
rv rf w 1 0 32 ffffffff80050d2c
rv commit 1 0000000080040428
rv load exe 1 6 1 000000008000bfa8 0000000000000000
rv trap 0 0 7 0000000010010000 <-----------------------------------Fails here
```
The addresses are identical, so I do not really see any reason why a store fault should happen on the second write.
Am I missing something, any help would be appreciated :)
Thank you for your time,
Quentin

bantierr, 2025-07-10:
Ok, so I digged around a bit, and it seems like the width of the store is the issue. 
Store to the CLINT using a store byte or half-word instruction does not work. but using a store word or double-word is fine.

I feel like that is maybe a functional bug ? I think an IO region should be accessible no matter the instruction access width.

Sorry for finding this in the first place ^^'

Thanks for the help

Dolu1990, 2025-07-11:
Hi, 

> Store to the CLINT using a store byte or half-word instruction does not work.

Right ^^

Always use 32 bits access to peripherals that expect it.

> I feel like that is maybe a functional bug ? I think an IO region should be accessible no matter the instruction access width.

It is by design.
Supporting multiple access width cost design time + hardware, isn't worth it, especialy in FPGA.

bantierr, 2025-07-11:
Alright, I see ! 
Thanks for the answer and have a nice day !