SpinalHDL/NaxRiscv #144: Potential decode error

bantierr, 2025-07-30:
Hi,

this is not exactly what I am trying to test, but I think I stumbled across a decoder issue I think. In a program I was running, the instruction `0xfb23c2e3 (@ pc=0x800a528c)` should be decoded as a `blt     t2, s2, pc - 92` instruction, as in spike, but it triggers a trap.

Here is a screenshot of the waveform isolating the bug:

[image]

 I am not quite sure what the root cause it, but I can try to fix it if you comfirm it is a bug.
Waveform:

[wave.zip](https://github.com/user-attachments/files/21513887/wave.zip)

Dolu1990, 2025-07-30:
Hi,

The fst file doesn't open, did you also had another weird wave.fst.?? in the same folder ?

bantierr, 2025-07-31:
Wait, I made a mistake I am so sorry, it is not a decode issue.