SpinalHDL/NaxRiscv #146: Interrupt handle problem when debug

HaroldZ32, 2025-08-15:
Hi,

There's a problem when I try to debug with nax. 

If an interrupt occurs while cpu is halted, it should be holded. 

[image]

And if there's an EXECUTE request from Debug Module, the pipeline will run for one step, but it will trigger the isFireing signal of decoded stage of Frontend pipeline. And the interrupt will be processed by mistake.

[image]

In risc-v debug spec, all the interrupt should be masked.

1.0 Version
[image]

0.13 Version
[image]

I think it's a bug that should be fixed. I'll make a PR if you think it's definitely a problem.

Thank for you help!

Dolu1990, 2025-08-15:
Hi,

Looking at the code, seems like adding : 
``` scala
valid clearWhen(setup.debugMode)
```
one line after https://github.com/SpinalHDL/NaxRiscv/blob/fb12adc9d4dcd7c3d92bc31903d7cd66c77f30aa/src/main/scala/naxriscv/misc/PrivilegedPlugin.scala#L661
would fix it.

But i didn't tested it.
Let me know it if fixes it for you.

> . I'll make a PR if you think it's definitely a problem.

Thanks :D, PR welcome :)

Regards
Charles

HaroldZ32, 2025-09-29:
> Hoo, how big is it of a change ?

I need to extract all the ram and place it on the toplevel to compare the ram signal. It's not much code change but huge change of the architecture. Or is there a better way to do it?